ucb-bar/fpga-zynq #4: Hello World not working after bitfile regeneration

crolfes, 2015-07-23:
Hi,

the prebuild images for the zedboard run fine.
```
root@zynq:~# ./fesvr-zynq pk hello
hello!
```
But when I create my own bitfile with Vivado (2014 or 2015, doesn't matter) the Hello World test progam produces no output any more. Seems like it stucks.
I execute the commands, like in your tutorial "2) Pushing Your Rocket Modifications to the FPGA".
Then I replace the file boot.bin on the SD card. But I cannot communicate with the RISCV any more.

sbeamer, 2015-07-23:
You will also need to recompile the parts of the stack ([riscv-fesvr](https://github.com/riscv/riscv-fesvr) and [riscv-pk](https://github.com/riscv/riscv-pk)) since the newest Rocket Chip implements a newer version of the privileged spec. Instructions for building riscv-fesvr for the FPGA can be found in the FPGA [README](https://github.com/ucb-bar/fpga-zynq#fesvr). Building pk for the FPGA is no different than any other platform, you can just follow its normal directions. Technically, you will not need to recompile hello since the user ISA hasn't changed.

We will be releasing new pre-built images in the next few weeks as the current round of revisions to Rocket Chip wrap up in time for Hot Chips.

crolfes, 2015-07-24:
I have recompiled riscv-fesvr riscv-pk and copied them into the zedboard ramdisk image.
But now I have an issue with an incompatible GLIBCXX library. I also rebuilt the uImage (make arm-linux), but the same error message is still there.

```
PetaLinux v2013.10 (Yocto 1.4) zynq ttyPS0
...
root@zynq:~# cp libfesvr.so /usr/lib/
root@zynq:~# ./fesvr-zedboard pk hello
./fesvr-zedboard: /usr/lib/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /usr/lib/libfesvr.so)
```

sbeamer, 2015-07-24:
This sounds like the arm-gcc used to make the disk image is different than the one used to compile the applications. The distributed image was made with Vivado 2014.4 and its corresponding arm compiler in the SDK.

Building a new disk image from scratch with the same compiler should work. You might need to copy a few libraries from the SDK into your image. Did you copy libstdc++?

sbeamer, 2015-08-21:
The prebuilt images have all been rebuilt including the ramdisk and the corresponding repos have been updated. You can also grab the ramdisk straight from [s3](https://s3-us-west-1.amazonaws.com/riscv.org/fpga-zynq-files/uramdisk.image.gz).

sbeamer, 2015-08-26:
With no responses in over a month, I'm closing this. If the above suggestions did not fix your problem, please comment again and we will reopen this issue.

closed 2015-08-26 by sbeamer