ucb-bar/fpga-zynq #6: How to increase BTB entries for normal operation on FPGA

max250, 2015-09-11:
Hi. I have a question about increasing BTB entries.
I want to increase the BTB entries of rocket-chip from 62 to 254 (considering 2 RAS entries).
For this configuation, I change the value of a parameter, NBTBEntries, written in rocket-chip/src/main/scala/PublicConfigs.scala.

And then, I generate verilog codes & a bitstream for zc706 FPGA.
I'm trying to run lua benchmark using a lua interpreter on this image, but it shows me a error message, illegal instruction.

Is there something more I did't know to change BTB entry size?

sbeamer, 2015-09-11:
Does Vivado successfully place and route the design with that many BTB entries? If it doesn't make timing, it will issue a critical warning but still produce a bitstream. In our experience, the BTB entries are extremely resource intensive when mapped to an FPGA.

Do you experience the same problem when generating your own bitstream but not changing the number of BTB entries?

Do you experience the same problem when you decrease the number of BTB entries?

max250, 2015-09-13:
When I generate & check both the original rocket and a rocket of a version decreased BTB entries, these images are normally running the lua interpreter on zc706 FPGA.
In this situation, your indication is reasonable & worth checking timing issue. I will check the timing of the P&R image when BTB entries are increased.

sbeamer, 2015-09-22:
Have you had any success? Other issues?

max250, 2015-09-24:
When I generate bitstream with 254 BTB entries, there are no errors related to timing.
But I don't know why it didn't work.

ccelio, 2015-09-25:
Do other programs work? Does hello world? Do the riscv-tests run on the FPGA? Is lua interpreter the only failing program you've thrown at it? 

Did you try 256 BTB entries? 128 BTB Entries? By setting it to a non-power of 2, you might be exposing a bug that is otherwise hidden.  Also, the RAS and BTB are completely orthogonal, and there is no meaning between the default's of 62 and 2. I think that's what was chosen because it ran well on a particular benchmark. =)

sbeamer, 2016-01-06:
I just ran this again (254 BTB entries) and it now appears to work. I don't know which commits could have fixed it, but I can't get it to hang now. Do you still have problems with the newest rocket chip?

sbeamer, 2016-01-14:
If this is still problematic, please comment again to re-open this issue.

closed 2016-01-14 by sbeamer