ucb-bar/fpga-zynq #9: Changing the Processor's Clockrate

metbosch, 2016-04-19:
When I try to modify the `RC_CLK_MULT` and `RC_CLK_DIVIDE` variables in the file `src/verilog/clocking.vh` to increase the processor frequency, Vivado seems to ignore the `RC_CLK_DIVIDE` value and use 1 (the value for `RC_CLK_MULT` seems right) causing a constraint violation and aborting the implementation.
The default values are `10.0 / 40.0` and I'm trying to use `27.0 / 100.0` which is near to the maximum allowed frequency in my board before violating the timing reports. Also, I experimented with other values for `RC_CLK_DIVIDE` and the same error appeared.

sbeamer, 2016-04-20:
Could you be more specific about the error you are getting? How do you know it is using 1 for RC_CLK_DIVIDE?

Vivado will complain if the values for MULT or DIVIDE fall out of certain ranges, so you will have to adjust both in order to approximate the desired clockrate. If you exceed the range, the error message will typically tell you what the correct range is. We did not include these ranges in the README since they depend on the board. As we warn in the README, changing ZYNQ_CLK_PERIOD is more involved.

metbosch, 2016-04-20:
I modify the file `src/verilog/clocking.vh` and set the content to:

> \`ifndef _clocking_vh_
\`define _clocking_vh_
\`define ZYNQ_CLK_PERIOD 10.0
\`define RC_CLK_MULT     27.0
\`define RC_CLK_DIVIDE   100.0
\`endif // _clocking_vh_

I open Vivado using the command `make vivado`, run the synthesis without problems and finally run the implementation that fails because a critical route design error. The message is the following:

> [DRC 23-20] Rule violation (PDRC-34) MMCM_adv_ClkFrequency_div_no_dclk - The computed value 2700.000 MHz (CLKIN1_PERIOD, net gclk_i) for the VCO operating frequency of the MMCME2_ADV site MMCME2_ADV_X0Y0 (cell MMCME2_BASE_inst) falls outside the operating range of the MMCM VCO frequency for this device (600.00002122251317 - 1200.0000424450263 MHz). The computed value is (CLKFBOUT_MULT_F * 1000 / (CLKINx_PERIOD * DIVCLK_DIVIDE)). Please run update_timing to update the MMCM settings. If that does not work, adjust either the input period CLKINx_PERIOD (10.000000), multiplication factor CLKFBOUT_MULT_F (27.000000) or the division factor DIVCLK_DIVIDE (1), in order to achieve a VCO frequency within the rated operating range for this device.
[Vivado_Tcl 4-16] Error(s) found during DRC. Router not run.

sbeamer, 2016-04-20:
This error does seem odd. What happens for other MULTs or DIVIDEs? In particular, DIVIDE < 40?

metbosch, 2016-04-21:
I tried with `5.0 / 20.0`, that should be the same as `10.0 / 40.0` (default values), and I get the same error:

>[DRC 23-20] Rule violation (PDRC-34) MMCM_adv_ClkFrequency_div_no_dclk - The computed value 500.000 MHz (CLKIN1_PERIOD, net gclk_i) for the VCO operating frequency of the MMCME2_ADV site MMCME2_ADV_X0Y0 (cell MMCME2_BASE_inst) falls outside the operating range of the MMCM VCO frequency for this device (600.00002122251317 - 1200.0000424450263 MHz). The computed value is (CLKFBOUT_MULT_F * 1000 / (CLKINx_PERIOD * DIVCLK_DIVIDE)). Please run update_timing to update the MMCM settings. If that does not work, adjust either the input period CLKINx_PERIOD (10.000000), multiplication factor CLKFBOUT_MULT_F (5.000000) or the division factor DIVCLK_DIVIDE (1), in order to achieve a VCO frequency within the rated operating range for this device.
[Vivado_Tcl 4-16] Error(s) found during DRC. Router not run.

Also, I tried `5.0 / 40.0` and `20.0 / 80.0` obtaining the same results:

>[DRC 23-20] Rule violation (PDRC-34) MMCM_adv_ClkFrequency_div_no_dclk - The computed value 500.000 MHz (CLKIN1_PERIOD, net gclk_i) for the VCO operating frequency of the MMCME2_ADV site MMCME2_ADV_X0Y0 (cell MMCME2_BASE_inst) falls outside the operating range of the MMCM VCO frequency for this device (600.00002122251317 - 1200.0000424450263 MHz). The computed value is (CLKFBOUT_MULT_F * 1000 / (CLKINx_PERIOD * DIVCLK_DIVIDE)). Please run update_timing to update the MMCM settings. If that does not work, adjust either the input period CLKINx_PERIOD (10.000000), multiplication factor CLKFBOUT_MULT_F (5.000000) or the division factor DIVCLK_DIVIDE (1), in order to achieve a VCO frequency within the rated operating range for this device.
[Vivado_Tcl 4-16] Error(s) found during DRC. Router not run.

>[DRC 23-20] Rule violation (PDRC-34) MMCM_adv_ClkFrequency_div_no_dclk - The computed value 2000.000 MHz (CLKIN1_PERIOD, net gclk_i) for the VCO operating frequency of the MMCME2_ADV site MMCME2_ADV_X0Y0 (cell MMCME2_BASE_inst) falls outside the operating range of the MMCM VCO frequency for this device (600.00002122251317 - 1200.0000424450263 MHz). The computed value is (CLKFBOUT_MULT_F * 1000 / (CLKINx_PERIOD * DIVCLK_DIVIDE)). Please run update_timing to update the MMCM settings. If that does not work, adjust either the input period CLKINx_PERIOD (10.000000), multiplication factor CLKFBOUT_MULT_F (20.000000) or the division factor DIVCLK_DIVIDE (1), in order to achieve a VCO frequency within the rated operating range for this device.
[Vivado_Tcl 4-16] Error(s) found during DRC. Router not run.

In contrast, I used `11.0 / 40.0` and the implementation fails because of timing problems. Maybe the timing checks are done before the another one. Here comes the error log:

> [Timing 38-282] The design failed to meet the timing requirements. Please see the timing summary report for details on the timing violations.

After all of these attempts, I tried again with `10.0 / 40.0` and the implementation succeed.

sbeamer, 2016-04-21:
I just did a fresh clone. I was able to reproduce your error for 27/100. When I used 11/40, it was successful. Are you using the included Verilog, or recently generated Verilog? It is possible the new design has worse timing. If you find anything, please let us know.

metbosch, 2016-04-22:
I'm not sure, but I think that I'm using the included Verilog. I followed the instructions of [https://github.com/ucb-bar/fpga-zynq#fromscratch](README > From scratch).
Today, I tried to change the ZYNQ_CLK_PERIOD instead of the divider values and the same error appeared. I modified the value in the 3 required places (clocking.vh, constrs/base.xdc and the visual block in Vivado).

sbeamer, 2016-04-22:
Not sure what to say since it did work with 11/40 on this end, but Vivado can be a bit shaky at times. If you aren't using the included verilog, it is possible rocket chip's critical path has worsened this Spring.

For the 4th RISC-V workshop, we will be updating this repo for Vivado 2016.2 and updating the pre-generated content. In the mean time, I would recommend using 10/40 since that works for you, but if you find a solution to your problem, please let us know. Those clocking parameters come into `MMCME2_BASE` in `src/verilog/rocketchip_wrapper.v`.

davidbiancolin, 2016-08-24:
To follow up on this, the PLLs in FPGAs have pretty stringent rules as to how their dividers can be configured; it can be pretty difficult to get legal frequencies by simply passing divider constants, which is what we effectively do.  

In the future I'll change the project to have vivado select MCMM configurations that, to a best effort, get close to the desired frequency, if no exact and legal configuration exists.

closed 2016-08-24 by davidbiancolin