ucb-bar/fpga-zynq #10: Quick instructions not working on Zybo

roman3017, 2016-05-07:
I have followed 1) Quick Instructions for Zybo and board boots to ARM-Linux. But neither Hello nor RISCV-Linux works for me. There is no output but fesvr-zynq can be interrupted with ^C.

---BUILD---
make fetch-images
make fetch-riscv-linux
make load-sd SD=/media/user/C4CA-9924
---LOG---
...
PetaLinux v2013.10 (Yocto 1.4) zynq ttyPS0                                      
                                                                                
zynq login: root                                                                
Password:                                                                       
login[812]: root login  on `ttyPS0'                                             
                                                                                
root@zynq:~# ./fesvr-zynq pk hello
^C
root@zynq:~# ls                                                                 
bbl         fesvr-zynq  hello       pk                                            
---END---

xfguo, 2016-05-10:
:+1: I have the exactly same issue. I will try to rebuild the whole zybo image...

metbosch, 2016-05-10:
Same issue on the Zedboard. However, it worked fine some weeks ago.
I have made a clean clone and I'm trying to rebuild all the stuff again,

sbeamer, 2016-05-10:
We will hopefully be able to look into the Zybo-specific issues later this week. In general when using the Rocket infrastructure, you should be careful to make sure the RTL and toolchain are compatible versions. Although the RISC-V user ISA is fixed, there continues to be slight tweaks to both the privileged spec and the platform spec.

xfguo, 2016-05-11:
After some debug with arm-zynq-gdb and gdbserver. I think the reason is the `pk` program is compiled with hard-float support but for ZYBO rocket configration, there is no FPU.

I got this message when I debug with gdb. 

```
../machine/minit.c:56: FPU not found; recompile pk with -msoft-float
```

So I recompile the `pk` with `-msoft-float`, then at least the hello program works well for me.

roman3017, 2016-05-12:
Thank you very much @xfguo for finding the problem. I will try it out.

sbeamer, 2016-05-12:
Yes, pk for Zybo needs to be compiled with soft-float.

closed 2016-05-12 by sbeamer

roman3017, 2016-05-15:
@xfguo can you please tell me how did you build with soft-float? I have tried this script:

-----
#! /bin/bash
. build.common
if [ ! `which riscv64-unknown-elf-gcc` ]
then
  echo "riscv64-unknown-elf-gcc doesn't appear to be installed; use the full-on build.sh"
  exit 1
fi
CFLAGS=-msoft-float CC=riscv64-unknown-elf-gcc build_project riscv-pk --prefix=$RISCV/riscv64-unknown-elf --host=riscv
-----

and got this error:

-----
configure:2364: checking whether the C compiler works
configure:2386: riscv64-unknown-elf-gcc -msoft-float   conftest.c  >&5
/.../riscv/lib/gcc/riscv64-unknown-elf/6.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccp2dWAo.o: can't link hard-float modules with soft-float modules
/.../riscv/lib/gcc/riscv64-unknown-elf/6.1.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/ccp2dWAo.o
collect2: error: ld returned 1 exit status
configure:2390: $? = 1
configure:2428: result: no
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "RISC-V Proxy Kernel"
| #define PACKAGE_TARNAME "riscv-pk"
| #define PACKAGE_VERSION "?"
| #define PACKAGE_STRING "RISC-V Proxy Kernel ?"
| #define PACKAGE_BUGREPORT "Andrew Waterman"
| #define PACKAGE_URL ""
| /* end confdefs.h.  */
| 
| int
| main ()
| {
| 
|   ;
|   return 0;
| }
configure:2433: error: in `/.../rocket-chip/riscv-tools/riscv-pk/build':
configure:2435: error: C compiler cannot create executables
-----

xfguo, 2016-05-16:
hi @roman3017, I just modified the generated Makefile and add -msoft-float to CFLAGS . I got same error when I try to `configure ...`.

adecold, 2016-05-27:
I also ran into this issue while testing it out on the zybo

roman3017, 2016-09-12:
The quick instructions work now. Thank you very much.
```
 _____       _           _      _
| ___ \     | |         | |    (_)
| |_/ / ___ | |_   __ _ | |     _  _ __   _   _ __  __
|  __/ / _ \| __| / _` || |    | || '_ \ | | | |\ \/ /
| |   |  __/| |_ | (_| || |____| || | | || |_| | >  <
\_|    \___| \__| \__,_|\_____/|_||_| |_| \__,_|/_/\_\

PetaLinux v2013.10 (Yocto 1.4) zynq ttyPS0

zynq login: root
Password: 
login[815]: root login  on `ttyPS0'

root@zynq:~# ./fesvr-zynq pk hello 
hello
root@zynq:~# ./fesvr-zynq bbl      
              vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
                  vvvvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvv  
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvv    
rr                vvvvvvvvvvvvvvvvvvvvvv      
rr            vvvvvvvvvvvvvvvvvvvvvvvv      rr
rrrr      vvvvvvvvvvvvvvvvvvvvvvvvvv      rrrr
rrrrrr      vvvvvvvvvvvvvvvvvvvvvv      rrrrrr
rrrrrrrr      vvvvvvvvvvvvvvvvvv      rrrrrrrr
rrrrrrrrrr      vvvvvvvvvvvvvv      rrrrrrrrrr
rrrrrrrrrrrr      vvvvvvvvvv      rrrrrrrrrrrr
rrrrrrrrrrrrrr      vvvvvv      rrrrrrrrrrrrrr
rrrrrrrrrrrrrrrr      vv      rrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrr          rrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrr      rrrrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrrrr  rrrrrrrrrrrrrrrrrrrrrr

       INSTRUCTION SETS WANT TO BE FREE
[    0.000000] Linux version 4.6.2-g8205b66-dirty (juander@schnipp) (gcc version 6.1.0 (GCC) ) #12 Fri Sep6
[    0.000000] Available physical memory: 252MB
[    0.000000] Initial ramdisk at: 0xffffffff80012d48 (422621 bytes)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080400000-0x000000008fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080400000-0x000000008fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080400000-0x000000008fffffff]
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 63630
[    0.000000] Kernel command line: 
[    0.000000] PID hash table entries: 1024 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 5, 131072 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 251160K/258048K available (1738K kernel code, 97K rwdata, 356K rodata, 492K init, 2)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:0 nr_irqs:0 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112s
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS ()
[    0.000000] pid_max: default: 32768 minimum: 301
[    0.000000] Mount-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.000000] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.000000] devtmpfs: initialized
[    0.000000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 191126044627500s
[    0.000000] NET: Registered protocol family 16
[    0.000000] clocksource: Switched to clocksource riscv_clocksource
[    0.000000] NET: Registered protocol family 2
[    0.000000] TCP established hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000000] TCP bind hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000000] TCP: Hash tables configured (established 2048 bind 2048)
[    0.000000] UDP hash table entries: 256 (order: 1, 8192 bytes)
[    0.000000] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes)
[    0.000000] NET: Registered protocol family 1
[    0.050000] Unpacking initramfs...
[    0.200000] console [sbi_console0] enabled
[    0.200000] futex hash table entries: 256 (order: 0, 6144 bytes)
[    0.210000] workingset: timestamp_bits=61 max_order=16 bucket_order=0
[    0.270000] devtmpfs: mounted
[    0.270000] Freeing unused kernel memory: 492K (ffffffff80000000 - ffffffff8007b000)
[    0.270000] This architecture does not have kernel memory protection.
# 
```