ucb-bar/fpga-zynq #12: No output from fesvr

adecold, 2016-05-28:
Hello,
I'm fairly new to the RISC/Rocketchip architecture and I'm trying to get the test program running on a couple of Zynq boards. I'm able to go through all of the steps in the guide expect getting a response from fesvr. It doesn't seem to hang the device and I'm able to Ctrl+C it.

- I initially tried the zybo but I suspect it fails due to an unresolved issue described here #10 
- I also have a [Zynq Mini Module Plus](http://picozed.org/product/zynq-mmp) board and I've attempted to add support for it by modifiying some existing files. See repo [here](https://github.com/adecold/zmmp7100). I get the same behaviour from fesvr. I do see a concerning boot up message:
> uio_pdrv_genirq 43c00000.htif: failed to get IRQ
> uio_pdrv_genirq: probe of 43c00000.htif failed with error -22
I quick google search shows it in the boot up log of someone who was able to still run fesvr.

Any tips on what I'm doing wrong or how I can debug this issue?

Thanks

davidbiancolin, 2016-08-24:
Did you end up resolving this?

tajayi, 2016-08-25:
I don't believe I was. I have since abandoned the effort..

closed 2016-08-25 by tajayi