ucb-bar/fpga-zynq #14: Port fpga-zync to myr z-turn board

rbrlibero, 2016-07-25:
Dear you all, i'm trying to get a vivado project for porting  to z-turn board , with no luck 

There is no way to add to git repositories a vivado project for zedboard and/or zybo to syntetize ?

rbrlibero, 2016-07-25:
my problem was already opened as issue 

 rocket-chip issues 174

sagark, 2016-07-25:
Your board looks very similar to the Zybo. As a starting point, I would follow the instructions in this repository for generating a vivado project for the Zybo that uses the verilog included in this repo.

rbrlibero, 2016-07-26:
i've tried to build vivado project for zedboard or zybo but i  got

[error] (rocketchip/compile:run) Nonzero exit code: 1
[error] Total time: 3 s, completed Jul 21, 2016 2:30:25 PM
make[1]: Leaving directory `/home/test/Program/rocket-chip/fsim'
cp: cannot stat ‘generated-src/{Top,ZynqAdapter}.DefaultFPGAConfig.v’: No such file or directory
make: *** [rocket] Error 1

in fpga-zybo images  cannot be added reference vivado project to let me to try a porting ?

sagark, 2016-07-27:
You can jump directly to the "Generating Project for Configuration" section of the README, no need to generate verilog from rocket-chip (this repo includes verilog from a known-compatible version of rocket-chip).

closed 2016-08-24 by davidbiancolin