ucb-bar/fpga-zynq #16: No ZYBO linux image provided

davidbiancolin, 2016-08-24:
For the time being bbl is not provided with the zybo images; Need to build a new bbl instance with a specific payload for zybo (RV64IMA).

bkoppelmann, 2016-09-08:
Hey David,
what is exactly needed as a payload? I looked [here](http://stackoverflow.com/questions/37849574/how-to-rebuild-bbl-with-payload-option) but just adding vmlinux to the --with-payload option of bbl's configure did not work. 
Cheers,
Bastian

davidbiancolin, 2016-09-08:
Hey Bastian, 
Can you describe me what you mean by does not work? Did bbl not absorb the payload?

bkoppelmann, 2016-09-08:
Thanks for the quick response. 

Here's how I produced bbl:
- I downloaded all images using 'make fetch-riscv-linux' in the zybo folder. 
- I installed the riscv-tools from the rocket submodule of this repo, using build.sh with the RV64IMA addition to the gnu-toolchain
- I rebuild riscv-pk using 'configure --prefix=/home/juander/coding/riscv/tools --host=riscv64-unknown-elf --with-payload=fpga-zynq/zybo/fpga-images-zybo/riscv/vmlinux' 
- I loaded the resulting bbl onto the zybo board into /root/home, mounted the sdcard and ran './fesvr-zynq +disk=/sdcard/riscv/root.bin bbl /sdcard/riscv/vmlinux' 

The result is that the RISCV splash appears and nothing more happens.
I tried running that exactly this bbl + vmlinux in spike and got the nice message: "core 0: exception trap_instruction_access_fault, epc 0x0000000000000000" so I guess linux traps somewhere. I suspect this is related to [this](https://github.com/riscv/riscv-pk/issues/18) issue.

Cheers,
Bastian

davidbiancolin, 2016-09-08:
So as described in that issue, we no longer have +disk support (read Howie's comment specifically). You'll be running `./fesvr-zynq bbl` and that's it. 

More to the point, i've duplicated your error. I'll let you know when i resolve it.

bkoppelmann, 2016-09-08:
Hey David,

thanks again for your help :). I solved the exception problem at least in spike by rebuilding Linux using the priv-1.9 branch. This boots Linux fine until it ultimately fails mounting the rootfs, as suspected after reading Howie's comment. I'll try now to add an initramfs to the kernel :)

Cheers,
Bastian

davidbiancolin, 2016-09-08:
Right, i was about to go about doing that. The old linux image is still on 1.7. Thanks for looking into this.

bkoppelmann, 2016-09-10:
Thanks for doing the bump here. Forgot that part :)

closed 2016-09-12 by davidbiancolin