ucb-bar/fpga-zynq #21: cannot generate new bitstream for 2 core Rocketchip

harriss95, 2016-09-19:
Im using ZEDBOARD for FPGA.

The problem raise when try to generate verilog code from fpga-zynq 

im using this version https://github.com/ucb-bar/rocket-chip/commit/255ef05e21b583f32870c1656861908da613d8de

ccelio, 2016-09-19:
Does 1 core rocket-chip work? Does it run on the FPGA? What is the error you get when doing 2 core?

harriss95, 2016-09-21:
The 2 core rocket-chip works fine. 

this is the error output for command `make fpga-images-zedboard/boot.bin`

> source rocketchip_wrapper.tcl -notrace
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/home/hars/154/Vivado/2015.4/data/ip'.
Command: synth_design -top rocketchip_wrapper -part xc7z020clg484-1
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xc7z020'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7z020'
WARNING: [Synth 8-992] S_AXI_GP0_ACLK_temp is already implicitly declared earlier [/home/hars/rocket-chip/fpga-zynq/zedboard/zedboard_rocketchip_DefaultFPGAConfig/zedboard_rocketchip_DefaultFPGAConfig.srcs/sources_1/bd/system/ip/system_processing_system7_0_0/hdl/verilog/processing_system7_v5_5_processing_system7.v:2701]
WARNING: [Synth 8-992] S_AXI_GP1_ACLK_temp is already implicitly declared earlier [/home/hars/rocket-chip/fpga-zynq/zedboard/zedboard_rocketchip_DefaultFPGAConfig/zedboard_rocketchip_DefaultFPGAConfig.srcs/sources_1/bd/system/ip/system_processing_system7_0_0/hdl/verilog/processing_system7_v5_5_processing_system7.v:2702]
ERROR: [Synth 8-2543] port connections cannot be mixed ordered and named [/home/hars/rocket-chip/fpga-zynq/zedboard/src/verilog/rocketchip_wrapper.v:333]
INFO: [Synth 8-2350] module rocketchip_wrapper ignored due to previous errors [/home/hars/rocket-chip/fpga-zynq/zedboard/src/verilog/rocketchip_wrapper.v:4]
Failed to read verilog '/home/hars/rocket-chip/fpga-zynq/zedboard/src/verilog/rocketchip_wrapper.v'
INFO: [Common 17-83] Releasing license: Synthesis
6 Infos, 2 Warnings, 0 Critical Warnings and 2 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
INFO: [Common 17-206] Exiting Vivado at Wed Sep 21 00:20:56 2016...
[Wed Sep 21 00:20:57 2016] synth_1 finished
wait_on_run: Time (s): cpu = 00:00:00.75 ; elapsed = 00:00:38 . Memory (MB): peak = 1282.805 ; gain = 0.000 ; free physical = 395 ; free virtual = 6471
# launch_runs impl_1 -to_step write_bitstream
ERROR: [Common 17-70] Application Exception: Failed to launch run 'impl_1' due to failures in the following run(s):
synth_1
These failed run(s) need to be reset prior to launching 'impl_1' again.

zhemao, 2016-11-01:
It seems to be working for me with the latest version of fpga-zynq and associated rocket-chip submodule. Could you try again after updating? You won't be able to place the default core design, but you should be able to place two small cores.

closed 2016-11-29 by davidbiancolin