ucb-bar/fpga-zynq #22: How to put files into riscv linux

proyar, 2016-10-03:
1.We tried to install the rocket chip with risc-v linux on Zedboard, and we had successfully entered riscv linux. However, we couldn't put the files compiled by assembler(using the instruction : riscv64-unknown-linux-gnu-g++) into riscv linux. Would someone please tell me how to put it in? (p.s we tried to mount the files into the file called root.bin,but it seems that risc-v linux doesn't use this file to boot.)
2. If we use the linux instrustion [make CONFIG=RoccExampleConfig] on our workshop, would it be RoCC there? If not, how to enable RoCC?

davidbiancolin, 2016-10-07:
1. Since we no longer have support for emulating a block device, you'll need to drop the files you care about into a custom initramfs. In the future we'll support a block device through a tether. 

2. Yes, you can set the CONFIG to point to one that correctly instantiates a coprocessor.

zhemao, 2016-11-01:
We should update the README to cover this.

davidbiancolin, 2016-11-29:
This has been updated.

closed 2016-11-29 by davidbiancolin

muhammadali201, 2018-10-28:
any info about how to create initramfs / or add your custom files in initramfs?