ucb-bar/fpga-zynq #24: Error running 'hello'

jsren, 2016-10-07:
I've (hopefully!) followed the excellent readme and have downloaded the pre-built binaries for my Zybo board. When I ssh into it and attempt to run `./fesvr-zynq pk hello` however, I'm presented with:

```../machine/minit.c:26: assertion failed: EXTRACT_FIELD(ms, MSTATUS_VM) == VM_CHOICE```

I've searched through GitHub - looks like the latest source for _riscv-pk_ has an assert 
https://github.com/riscv/riscv-pk/blob/8006d004a59618ce455c4365c09e138b06ca7b70/machine/minit.c#L26

whereas the source linked `The Zybo build was last tested with [this version ](https://github.com/ucb-bar/rocket-chip/commit/2f71a3da5a7d41b4aa2c7a617902f2aee8f2cbe1)of the toolchain.` instead just tests and continues? 
https://github.com/riscv/riscv-pk/blob/d37b50edf5faee2fe41a2586479ddab7c8a2ec9b/pk/minit.c#L24

Not sure if that's relevant, whether the assert is actually meant to _pass_ or whether this actually relates to bbl?

Cheers

davidbiancolin, 2016-10-07:
@zhemao Can you address this, since you updated the images last.

zhemao, 2016-10-07:
Oh crap, it looks like the DefaultFPGASmallConfig has virtual memory turned off now. So of course it's not going to work with the new pk, which requires virtual memory.

I can fix this in my outstanding PR for the serial tether change. It might take a while for that to make it in.

@jsren If you don't want to wait, you can go to rocket-chip/src/main/scala/coreplex/Configs.scala and edit the WithSmallCores config. If you change "UseVM" and "UseUser" to true and recompile the boot image, it should work.

davidbiancolin, 2016-10-07:
@jsren You could also try the previous commit for the time being: 86d2cf1de7a7268141c653364c9114837d0e1b69

jsren, 2016-10-07:
@zhemao @davidbiancolin Cheers!

aroelke, 2016-10-24:
I've been following the tutorial for pushing rocket modifications to the FPGA on my Zedboard using the default ZynqConfig configuration, and when I go to run the example hello program I get a similar error to what @jsren found:

`fesvr-zynq: ../fesvr/dtm.cc:67: uint32_t dtm_t::run_program(const uint32_t*, size_t, size_t): Assertion `n <= ram_words()' failed.`

Somehow it seems to think that the amount of memory the program wants is greater than available, but this also happens when I increase memory size (and I don't see how hello could need more than 256MB of memory anyway).  This doesn't happen with the image that comes with the repository, but it does happen if I rebuild boot.bin according to the tutorial, or fetch it with make fetch-images.

zhemao, 2016-11-01:
This has been fixed by 9a0b82d0c47911e56d4755cb2e8f4aa15713bae4.

closed 2016-11-01 by zhemao