ucb-bar/fpga-zynq #25: priv-1.9-rc0-like branch for fpga-zynq and rocket-chip

heshamelmatary, 2016-10-17:
Hi,

riscv-tools and spike have priv-1.9-rc0 branches, which I use as I consider them stable (no changes). I've been developing assuming these branches on spike, fesvr and pk/bbl. Now, given that I got a new zybo board, I want to run my programs on fpga-zynq and/or rocket-chip priv-1.9-rc0-compatible branch/tag. 

I am able to run pk, hello and linux-riscv on the latest fpga-zynq/zybo (after the UseVM fix https://github.com/ucb-bar/fpga-zynq/issues/24). However, my binary (seL4-based system), doesn't run on this latest RISC-V/fpga-zynq/rocket-chip revision. 

The error/warning that I get is:

> root@zynq:~# ./fesvr-zynq /sdcard/riscv/sel4test-driver-image-riscv-spike 
warning: tohost and fromhost symbols not in ELF; can't communicate with target

The same binary runs on spike git:(priv-1.9) and riscv-fesvr git:(priv-1.9) fine.

Would appreciate any help how to get priv-1.9(-rc0) implementation of rocket-chip/fpga-zybq on zybo.

Cheers,
Hesham

zhemao, 2016-11-01:
The priv-1.9-rc0 branches of spike and riscv-tools seem to have been deleted. We've made a lot of privileged ISA changes since them, so I'm not sure what to tell you. The priv-1.9 ISA will be in draft mode for a while. Maybe @aswaterman can comment on when we expect a stable ISA release.

heshamelmatary, 2016-11-06:
I used the priv-1.9-rc0 from riscv-tools, and it checks out some hardcoded revisions rather than branches. Seems like a lot has changed (entry address, ABI and PTE format) diff priv-1.9. It's becoming a hurdle (especially for supervisor-level SW like OSes/bootloaders) to keep up with this such changes.

I would suggest to have "stable" branches and/or revisions including the rocket chip, spike, riscv-fesvr (both for ARM and native) and pk as  standard/reference implementations **for each priv release**.

zhemao, 2016-11-07:
We will have stable releases of riscv-tools for stable privileged ISA releases. Priv-1.9 is not meant to be a stable release. There is a reason all of the ISA documents say "draft" on them. Priv-1.7 is the current stable privileged ISA, and the master branch of riscv-tools conforms to that spec.

closed 2016-11-29 by davidbiancolin