ucb-bar/fpga-zynq #26: testchipip submodule not being initialised

jsren, 2016-11-01:
Running the instructions in _Pushing Your Rocket Modifications to the FPGA_ doesn't seem to initialise the _testchipip_ submodule, leading to build errors.

I'm not sure if I'm doing something wrong, but I didn't see anything about cloning in the readme. My build script is this (bash shell):
```
    git clone https://github.com/ucb-bar/fpga-zynq
    cd fpga-zynq
    git submodule update --init testchipip  # <-----
    cd zybo
    make fetch-images
    make fetch-riskv-linux
    make init-rocket-submodule
    patch_config
    make pack-rocket
    make rocket
    make project
    make fpga-images-zybo/boot.bin
```
(where `patch_config` is a temporary workaround for #24)

Omitting the indicated line results in `make rocket` failing.

davidbiancolin, 2016-11-01:
@zhemao Can you update the README to reflect your changes to the repo over the past month? We can't let the README quality regress. 

We probably want to update the make recipe "init-rocket-submodule" to initialize all submodules required for a fresh build appropriately, as the process is fairly nuanced now. Or remove the recipe all together and explain it in the README.

zhemao, 2016-11-01:
I'm just going to have all of the submodules get updated as a dependency of the rocketchip.stamp rule. This way, all you have to do is run `make`. I'll update README accordingly.

ccelio, 2016-11-01:
How does that work if I want to use my own submodule branch? Will that undo me switching to a different commit?

zhemao, 2016-11-01:
Good point. In that case I'll just have a separate init-submodules rule that initializes everything.

zhemao, 2016-11-01:
Okay, I've updated this.

But there's a remaining issue that the instructions on building riscv-linux are out of date. I'll create a separate issue for that.

closed 2016-11-01 by zhemao