ucb-bar/fpga-zynq #27: How to build riscv-linux from source for zybo?

colin4124, 2016-11-22:
`make fetch-riscv-linux-deliver` would download the latest version of Pre-Built Binary and Root FS. But I used the old version of rocket chip (ba96ad2b383a97a15b2d95b1acfd551f576c8faa) , fesvr and bbl. It is too old to use the latest version of pre-built Binary and Root FS. So where to get a modified version of the kernel that ignores FPU instructions?

davidbiancolin, 2016-11-22:
You could try [these](https://github.com/ucb-bar/fpga-images-zybo/tree/396e68f5ba5db672c49972be915a02a2242044bb) images first.

davidbiancolin, 2016-11-29:
Did this work out for you?

colin4124, 2016-11-30:
Yes, pre-built version works. But I want to get a modified version of the kernel that ignores FPU instructions.

closed 2016-11-30 by colin4124

davidbiancolin, 2016-11-30:
Essentially, you'll need to build the kernel with a RV64IMA version of the compiler. I think the README on [riscv-tools](https://github.com/riscv/riscv-tools) gives a pretty good overview of how to build everything from scratch. We plan on putting the flow we use to build our kernel + ramdisk for the FPGA images out in the open at some point.