ucb-bar/fpga-zynq #29: Running rocc on zybo

colin4124, 2016-11-23:
I want to run rocc (https://github.com/ucb-bar/rocc-template) on zybo, I follow the instructions from **2) Pushing Your Rocket Modifications to the FPGA:**

```
make rocket CONFIG=Sha3CPPConfig
make project CONFIG=Sha3CPPConfig
make fpga-images-zybo/boot.bin CONFIG=Sha3CPPConfig
```

In `make fpga-images-zybo/boot.bin CONFIG=Sha3CPPConfig`, it reports the error:

> ERROR: [Place 30-640] Place Check : This design requires more LUT as Logic cells than are available in the target device. This design requires 38836 of such cell types but only 17600 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.

So, does it means `set_param drc.disableLUTOverUtilError 1` ? If so, which file should I write to ?

seldridge, 2016-11-23:
That error is that you ran out of space on the Zybo's programmable logic. If I remember correctly, the "small" configuration that is supposed to be built for a Zybo (nothing in a RoCC socket, no hard FPU, no L2) uses around 15k LUTs, so there's just not a lot of extra space for doing hardware designs with RoCCs.

Unfortunately, the advised option is likely to purchase a Zedboard which has enough space to fit a full-sized Rocket and something in a RoCC socket.

colin4124, 2016-11-23:
@seldridge I have try RoCC config extend from "small" configuration, it is still ran out of space. Can I disable some config in PrivateConfigs.scala would fit in Zybo with RoCC socket?

seldridge, 2016-11-23:
You can try, but unfortunately [SmallConfig](https://github.com/ucb-bar/rocket-chip/blob/94cc1efadc43a10ad0bdd14b0f672da89a729ffa/src/main/scala/coreplex/Configs.scala#L245) already makes most of the reasonable optimizations to reduce size:
  * No FPU
  * No L2
  * L1 cache reductions
  * BTB with no entries

There _may_ be something else that you can throw out here or reduce in size, but there's just not a lot of non-essential components left.

colin4124, 2016-11-23:
Here is RoCC template Config:

```
class Sha3Config extends SmallConfig {
  override val topDefinitions:World.TopDefs = {
    (pname,site,here) => pname match {
      case WidthP => 64
      case Stages => Knob("stages")
      case FastMem => Knob("fast_mem")
      case BufferSram => Dump(Knob("buffer_sram"))
      case RoccMaxTaggedMemXacts => 32
      case BuildRoCC => Seq( 
                          RoccParameters(    
                            opcodes = OpcodeSet.custom0,
                            generator = (p: Parameters) => (Module(new Sha3Accel()(p.alterPartial({ case CoreName => "Rocket" })))) ))
    }
  }
 
  override val topConstraints:List[ViewSym=>Ex[Boolean]] = List(
    ex => ex(WidthP) === 64,
    ex => ex(Stages) >= 1 && ex(Stages) <= 4 && (ex(Stages)%2 === 0 || ex(Stages) === 1),
    ex => ex(FastMem) === ex(FastMem),
    ex => ex(BufferSram) === ex(BufferSram)
    //ex => ex[Boolean]("multi_vt") === ex[Boolean]("multi_vt")
  )
  override val knobValues:Any=>Any = {
    case "stages" => 1
    case "fast_mem" => true
    case "buffer_sram" => false
    case "multi_vt" => true
  }
}
```
Is there any non-essential components left in Sha3Config ?

colin4124, 2016-11-28:
rocc (https://github.com/ucb-bar/rocc-template) itself is too big to run on zybo, so I write simple one it works.

closed 2016-11-28 by colin4124