ucb-bar/fpga-zynq #30: Vivado (still, 2016.2) cannot retime FPU.

davidbiancolin, 2016-11-23:
This is a known issue, but there have been some recent developments in the retiming flow in Vivado so i took the time to re-investigate why this continues to be case. 

As of 2015.3, synthesis supports retiming with either a custom synthesis strategy enabling it, or by passing -retime on to synth_design. (See [UG901](https://www.xilinx.com/support/documentation/sw_manuals/xilinx/2016_2/ug901-vivado-synthesis.pdf))AFAICT, Vivado can now retime in both directions. In any case, i've tried moving the pipeline registers in the FPU to the front of the sub units to no effect. 

Pipeline registers behind:
```
INFO: [Synth 8-5816] Retiming module `FPU__GB0_tempName`                            

    Effective logic levels on critical path before retiming is: 43                                                                                                                                                                            
    Total number of crtical paths = 252                                             
                                                                                    
    Optimizing at the module level                                                  
    Optimizing locally to improve critical paths(may not reduce worst delay)    
                                                                                    
    Effective logic levels on critical path after retiming is: 43                   
    Total number of crtical paths = 684                                             
    Numbers of forward move = 0, and backward move = 65 
```       

Pipeline registers in front: 
```
INFO: [Synth 8-5816] Retiming module `FPU__GB0_tempName`                        
    Effective logic levels on critical path before retiming is: 43              
    Total number of crtical paths = 875                                         
                                                                                
    Optimizing at the module level                                              
    Cannot find a feasible solution:                                            
        Effective logic levels from fpuOpti_0/top/target/coreplex/tiles_0/fpuOpt/FPUFMAPipe_1_1/fma/T_7__2(fixed:BRAM/DSP/SRL) to fpuOpti_0/top/target/coreplex/tiles_0/fpuOpt/FPUFMAPipe_1_1/T_370_data_reg[0](fixed:OUTPUT) is: 43
        Effective logic levels found across for latency (=1) is: 43             
    Optimizing locally to improve critical paths(may not reduce worst delay)    
                                                                                
    Effective logic levels on critical path after retiming is: 43               
    Total number of crtical paths = 250                                         
    Numbers of forward move = 8, and backward move = 5                          
                                                         
```
                                                  
I will continue to update this issue as I find the time to investigate further.

ccelio, 2016-11-23:
Good idea on posting this as an issue!

ben-k, 2017-02-16:
Of possible interest from a related hw-dev thread is that Altera's tools struggle too:

Rocket w/ FPU, no retiming: 30MHz
Rocket w/ FPU, with retiming: 35MHz
Rocket w/ FPU, "aggressive" retiming: 40MHz
Rocket w/out FPU: 70MHz

davidbiancolin, 2017-02-16:
That's actually rather surprising. I was expecting to work better than that. At some point i'll just bite the bullet and install Quartus on the millennium machines. 

The right approach is to use the newest version of the tools, target both Stratix10 and some other V series family, and try building both configurations (registers in front and behind). 

It would not surprise me if Altera selectively decided to improve retiming for Stratix10 targets.

davidbiancolin, 2017-08-09:
In 2016.2, i've tried disabling DSP inference, which reports tell me allows the single-precision FPU to be retimed. Vivado still can't retime the double-precision FMA though. Will try again with a newer version when i get around to it.