ucb-bar/fpga-zynq #31: Quick instructions not working on Zedboard

gon1332, 2016-11-25:
After:
```
$ make fetch-images
$ make fetch-riscv-linux
$ make load-sd SD=path_to_mounted_sdcard
```

and trying to boot from the SD card nothing happens. I suppose that the case is in the format of the SD.
It has two partitions. The first one is formatted as `fat` and it will be used for the bootloader while the second one is formatted as `ext4` to hold the file system.

As long as _Quick Instructions_ are based on RAMDisk the `ext4` filesystem is not used. So the `path_to_mounted_sdcard` in our case is the `fat` formatted partition.

Other petalinux projects boot successfully. I use `minicom` to connect to the board. In the case of this project's prebuilt data, nothing appears on `minicom`'s screen.

yanjiahuang, 2017-04-02:
It seems that `$ make fetch-riscv-linux` does not work anymore.

davidbiancolin, 2017-06-20:
make fetch-riscv-linux isn't supported anymore. There should be a bbl instance that includes a linux image and initramfs in the provided ARM filesystem. 

I'm closing this because we let it fester -- reopen if you still want help.

closed 2017-06-20 by davidbiancolin