ucb-bar/fpga-zynq #32: testchipip build error - Adapter.scala:14: not found: object cde

mward15, 2016-12-14:
I am trying to follow the instructions from README.md at "Working with Vivado" chapter to build a vivado project. When I enter **`make project`** command, I run into many build issues. The first issue is **`Adapter.scala:14: not found: object cde`**, and I am not sure how and where to get this "object cde" ! Below is a full error log:
```
[mward@hobby-pc zybo]$ pwd
/home/mward/fpga-zynq/zybo
[mward@hobby-pc zybo]$ make project
rm -f /home/mward/fpga-zynq/common/../testchipip/lib
ln -s /home/mward/fpga-zynq/common/lib /home/mward/fpga-zynq/common/../testchipip/lib
cd /home/mward/fpga-zynq/common/../testchipip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/mward/fpga-zynq/rocket-chip/sbt-launch.jar package
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Set current project to testchipip (in build file:/home/mward/fpga-zynq/testchipip/)
[info] Compiling 12 Scala sources to /home/mward/fpga-zynq/testchipip/target/scala-2.11/classes...
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:14: not found: object cde
[error] import cde.{Parameters, Field}
[error]        ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:16: not found: type Field
[error] case object SerialInterfaceWidth extends Field[Int]
[error]                                          ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:19: not found: type Parameters
[error]   def apply(p: Parameters) = p.alterPartial({
[error]                ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:28: not found: type Parameters
[error] class SerialAdapter(implicit p: Parameters) extends TLModule()(p) {
[error]                                 ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:29: overloaded method value apply with alternatives:
[error]   [T](pname: config.Field[T], site: config.View)T <and>
[error]   [T](pname: config.Field[T])T
[error]  cannot be applied to (testchipip.SerialInterfaceWidth.type)
[error]   val w = p(SerialInterfaceWidth)
[error]           ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:32: could not find implicit value for parameter p: config.Parameters
[error]     val mem = new ClientUncachedTileLinkIO
[error]               ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:75: could not find implicit value for parameter p: config.Parameters
[error]   val put_acquire = Put(
[error]                        ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:82: could not find implicit value for parameter p: config.Parameters
[error]   val get_acquire = Get(
[error]                        ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:183: not found: type Parameters
[error]   implicit val p: Parameters
[error]                   ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:192: not found: type Parameters
[error]   implicit val p: Parameters
[error]                   ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:198: not found: type Parameters
[error]   implicit val p: Parameters
[error]                   ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:202: class BaseCoreplexBundle takes type parameters
[error]   val coreplexIO: BaseCoreplexBundle
[error]                   ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Adapter.scala:214: class BaseCoreplexBundle takes type parameters
[error]   val coreplexIO: BaseCoreplexBundle
[error]                   ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/ChannelMapper.scala:5: not found: object cde
[error] import cde.Parameters
[error]        ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/ChannelMapper.scala:12: not found: type Parameters
[error] class ChannelAddressMapper(n: Int)(implicit p: Parameters) extends Module {
[error]                                                ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/ChannelMapper.scala:14: could not find implicit value for parameter p: config.Parameters
[error]     val in  = Vec(n, new ClientUncachedTileLinkIO).flip
[error]                      ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/ChannelMapper.scala:15: could not find implicit value for parameter p: config.Parameters
[error]     val out = Vec(n, new ClientUncachedTileLinkIO)
[error]                      ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/ChannelMapper.scala:38: not found: type Parameters
[error] class ChannelAddressUnmapper(n: Int, c: Clock = null, r: Bool = null)(implicit p: Parameters)
[error]                                                                                   ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/ChannelMapper.scala:41: could not find implicit value for parameter p: config.Parameters
[error]     val in =  Vec(n, new ClientUncachedTileLinkIO).flip
[... 215 lines cut ...]
[error]        ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Util.scala:13: type mismatch;
[error]  found   : chisel3.core.Direction.Input.type
[error]  required: Boolean
[error]     val reset = Bool(INPUT)
[error]                      ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Util.scala:14: type mismatch;
[error]  found   : chisel3.core.Direction.Output.type
[error]  required: Boolean
[error]     val reset_sync = Bool(OUTPUT)
[error]                           ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Util.scala:32: recursive value value needs type
[error]     val value = UInt(OUTPUT, width = w)
[error]                              ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Util.scala:52: not found: type Parameters
[error] class PutSeqDriver(val s: Seq[Tuple2[BigInt,Int]])(implicit p: Parameters) extends Driver()(p) {
[error]                                                                ^
[error] /home/mward/fpga-zynq/testchipip/src/main/scala/Util.scala:62: could not find implicit value for parameter p: config.Parameters
[error]     Put(UInt(0), blockAddr, beatAddr, UInt(d))
[error]        ^
[error] 94 errors found
[error] (compile:compileIncremental) Compilation failed
[error] Total time: 8 s, completed Dec 14, 2016 4:34:48 PM
make: *** [/home/mward/fpga-zynq/common/lib/testchipip.stamp] Error 1
```

davidbiancolin, 2016-12-15:
It's seems like you may have initialized the submodules incorrectly or you tried bumping/using a different version of rocketchip. Did "make init-submodules", or did you try to initialize them in a different manner?

mward15, 2016-12-16:
@davidbiancolin I tried to make init-submodules again, but it didn't help to me - got the same problem in the end. Please tell, where do you obtain this object cde, so that I could link to it manually? I tried searching the internet for object cde, also looked through your whole repository, and still couldn't find this object cde anywhere... In what file the object cde is defined?

davidbiancolin, 2016-12-16:
CDE is a git submodule of rocketchip. The head of master of fpga-zynq points at [this](https://github.com/ucb-bar/rocket-chip/tree/5392219d86bd8ff3b241122bd38852e89fa06ac2) revision of rocket-chip that includes CDE.

Can you tell me what SHA/commit of fpga-zynq and rocketchip you are using?

mward15, 2016-12-19:
@davidbiancolin I am using the latest version (HEAD) of fpga-zynq, and almost the latest version of rocket-chip (can't determine exact version, was updated 1 week ago). David, thank you very much for focusing my attention on rocket-chip repo, I think I found the source of a problem - someone has decided to make chisel "more flexible", removed CDE (context-dependent-environments) from rocket-chip default submodules - and this change has been merged 3 weeks ago: https://github.com/ucb-bar/rocket-chip/pull/444/commits/e8e95d4bcf025f266070819c266c7d7ddae8676f

davidbiancolin, 2016-12-19:
No problem. So make sure to use the version of rocketchip that the submodule pointer defaults to: 5392219d86bd8ff3b241122bd38852e89fa06ac2. Newer versions of rocketchip will not function with this repository. 

Yes, rocketchip's parameterization system is being overhauled. The functionality provided by CDE has mostly been moved into the config package.

closed 2016-12-19 by mward15