ucb-bar/fpga-zynq #36: Extending Rocket on Zybo

jsren, 2017-01-11:
Hi, so I'm trying to get Rocket to run via some external memory and interface with other peripherals and in so doing have it run programs entirely independently of the ARM APU. I'm assuming that'll take specific handling of the RISC binaries and not using system calls etc, but I've a couple of questions about the structure of Rocket to which I wondered if you wouldn't mind giving brief answers?

I've spent quite a while trying to figure out how everything fits together, and I've made a diagram of how I _think_ things are working with the hope of avoiding pestering you too much!

1) Is this diagram accurate (w.r.t the default config)?
[image]

2) Are the changes (in red) proposed in the second diagram feasible?
[image]

3) The L1 dcache source is very dense. I'm trying to make it write-through, and I didn't see anything existing that supports that. Is there, and do you think it would be a big task to make it so? 

4) For memory-mapping the LEDs, switches, PMID and other peripherals, to prevent caching, would setting a TileLink `RegionType` of `UNCACHED` do it? Do I need to avoid `Acquire` support?

Apologies if this is the wrong place to ask! If so, any direction on where to point questions would be really appreciated.

davidbiancolin, 2017-01-11:
1. It's correct though, as i'm sure you are aware you are leaving out the tether from the ARM core to the rocket-chip target.
2. Yes it is quite feasible and encouraged!
3. Best posed at the rocket-chip repository. Though i'm fairly certain that write through is not a supported configuration. 
4. Again best posted at rocket-chip. There are examples there. (Though SiFive has the best examples of IP integration (like GPIO) ([Freedom SoC](https://github.com/sifive/freedom), [Periphery IP](https://github.com/sifive/sifive-blocks): https://github.com/sifive/freedom

In general, questions and issues raised about this repository should relate directly to how rocketchip is being hosted on the ZYNQ platforms this repository supports.

davidbiancolin, 2017-01-11:
Sorry, on a re-read of the question, i need to update my first answer with another question. When you say: 

> have it run programs entirely independently of the ARM APU.

Do you mean that you also want the core to boot independently of the ARM core?

jsren, 2017-01-11:
Thanks for the reply - and the links!

> Do you mean that you also want the core to boot independently of the ARM core?

That would be the ideal, yes.

davidbiancolin, 2017-01-11:
In that case you'll need to configure rocketchip to boot standalone from some additional boot device (like Flash, in fabric ROM, etc...).

jsren, 2017-01-11:
@davidbiancolin Thanks for all the help! - I'll go re-post the questions in the Rocket repo.

closed 2017-01-11 by jsren

josecm, 2020-02-10:
@jsren I am trying to do exactly the same thing. Running Rocket standalone (no pk, no fesrv) with direct access to peripherals and DRAM on a zynq board. Were you successful? If so, can you give me some pointers on how to achieve this?