ucb-bar/fpga-zynq #39: make rocket fails with java.lang.reflect.InvocationTargetException

OYTIS, 2017-02-25:
Commands to reproduce:

    git clone git@github.com:ucb-bar/fpga-zynq.git
    cd zybo
    make init-submodules
    make rocket

Error:

    [error] (run-main-0) java.lang.reflect.InvocationTargetException
    java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at
    sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:44)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:91)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:91)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:218)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:216)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:216)
	at chisel3.Driver$.elaborate(Driver.scala:91)
    Caused by: chisel3.core.Binding$BindingException: t (chisel3.core.UInt@4ae48) must be unbound Type. Try using cloneType?: Already bound to OpBinding(zynq.NastiFIFO@4ac1e)
	at chisel3.core.Binding$.checkUnbound(Binding.scala:115)
	at chisel3.core.Reg$.makeType(Reg.scala:15)
	at chisel3.core.Reg$.do_apply(Reg.scala:63)
	at chisel3.core.Reg$.apply(Reg.scala:46)
	at chisel3.core.Reg$.apply(Reg.scala:53)
	at zynq.NastiFIFO.<init>(Top.scala:156)
	at zynq.ZynqAXISlave$$anonfun$14.apply(Top.scala:57)
	at zynq.ZynqAXISlave$$anonfun$14.apply(Top.scala:57)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at zynq.ZynqAXISlave.<init>(Top.scala:57)
	at zynq.Top$$anonfun$12.apply(Top.scala:28)
	at zynq.Top$$anonfun$12.apply(Top.scala:28)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at zynq.Top.<init>(Top.scala:28)
    java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:2081)
	at sbt.Defaults$.toError(Defaults.scala:39)
	at  sbt.Defaults$$anonfun$runTask$1$$anonfun$apply$38$$anonfun$apply$39.apply(Defaults.scala:750)
	at sbt.Defaults$$anonfun$runTask$1$$anonfun$apply$38$$anonfun$apply$39.apply(Defaults.scala:748)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)

rocket-chip commit: f3299ae91d3f01d0349eb4746886e303e8fb1b41

milapowel, 2017-02-27:
Hi, I have same problem on zedboard, I tried with java:
java-1.7.0-openjdk-1.7.0.101-2.6.6.1.el7_2.x86_64
java-1.8.0-openjdk-1.8.0.101-3.b13.el7_2.x86_64
jdk1.8.0_60

but always same problem

davidbiancolin, 2017-02-27:
I'll have a look at this today.

davidbiancolin, 2017-02-27:
That should do it. @zhemao *wrist slap*