ucb-bar/fpga-zynq #40: Build from fresh clone broken

jsren, 2017-03-07:
Building from a fresh clone using the instructions in the readme gives the following error

(Reproduced on two different platforms)

```
[info] Set current project to rocketchip (in build file:/home/jsren/fpga-zynq2/rocket-chip/)
[info] Updating {file:/home/jsren/fpga-zynq2/rocket-chip/}cde...
[info] Resolving jline#jline;2.12.1 ...
[info] Done updating.
[warn] Scala version was updated by one of library dependencies:
[warn] 	* org.scala-lang:scala-library:2.11.6 -> 2.11.7
[warn] To force scalaVersion, add the following:
[warn] 	ivyScala := ivyScala.value map { _.copy(overrideScalaVersion = true) }
[warn] Run 'evicted' to see detailed eviction warnings
[info] Updating {file:/home/jsren/fpga-zynq2/rocket-chip/}coreMacros...
[info] Resolving edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT ...
[warn] 	module not found: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT
[warn] ==== local: tried
[warn]   /home/jsren/.ivy2/local/edu.berkeley.cs/firrtl_2.11/1.1-SNAPSHOT/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/edu/berkeley/cs/firrtl_2.11/1.1-SNAPSHOT/firrtl_2.11-1.1-SNAPSHOT.pom
[info] Resolving jline#jline;2.12.1 ...
[info] downloading https://repo1.maven.org/maven2/org/scalamacros/paradise_2.11.7/2.1.0/paradise_2.11.7-2.1.0.jar ...
[info] 	[SUCCESSFUL ] org.scalamacros#paradise_2.11.7;2.1.0!paradise_2.11.7.jar (209ms)
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT: not found
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn] 	Note: Unresolved dependencies path:
[warn] 		edu.berkeley.cs:firrtl_2.11:1.1-SNAPSHOT
[warn] 		  +- edu.berkeley.cs:coremacros_2.11:3.1-SNAPSHOT
[info] Compiling 4 Scala sources to /home/jsren/fpga-zynq2/rocket-chip/context-dependent-environments/target/scala-2.11/classes...
[warn] there were 6 feature warnings; re-run with -feature for details
[warn] one warning found
[info] Packaging /home/jsren/fpga-zynq2/rocket-chip/context-dependent-environments/target/scala-2.11/cde_2.11-1.0.jar ...
[info] Done packaging.
sbt.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT: not found
	at sbt.IvyActions$.sbt$IvyActions$$resolve(IvyActions.scala:313)
	at sbt.IvyActions$$anonfun$updateEither$1.apply(IvyActions.scala:191)
	at sbt.IvyActions$$anonfun$updateEither$1.apply(IvyActions.scala:168)
	at sbt.IvySbt$Module$$anonfun$withModule$1.apply(Ivy.scala:156)
	at sbt.IvySbt$Module$$anonfun$withModule$1.apply(Ivy.scala:156)
	at sbt.IvySbt$$anonfun$withIvy$1.apply(Ivy.scala:133)
	at sbt.IvySbt.sbt$IvySbt$$action$1(Ivy.scala:57)
	at sbt.IvySbt$$anon$4.call(Ivy.scala:65)
	at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:93)
	at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:78)
	at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:97)
	at xsbt.boot.Using$.withResource(Using.scala:10)
	at xsbt.boot.Using$.apply(Using.scala:9)
	at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:58)
[error] (coreMacros/*:update) sbt.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT: not found
```

smosanu, 2017-03-10:
I also encountered this error and fixed it by rebuilding firrtl and adding firrtl/utils/bin/ to PATH

jsren, 2017-03-10:
@smosanu Thanks! It doesn't seem to work for me, though. Am I missing anything?

In firrtl: `make clean`, `make build`, export PATH="$PATH:/abs/path/to/firrtl/utils/bin/"
In emulator: `make`

```[error] (coreMacros/*:update) sbt.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT: not found```

I've confirmed that `firrtl` is definitely in the path.

ccelio, 2017-03-10:
Does following the instructions in the FIRRTL readme address this issue? (not being snarky).

https://github.com/ucb-bar/firrtl

jsren, 2017-03-10:
@ccelio Thanks a lot! - `sbt publish-local` worked like a charm.

davidbiancolin, 2017-03-10:
Yeah, sorry guys, you'll either need to publish-local or manage the firrtl dependency manually. 

When publishing locally, you just need to be wary of blowing away a FIRRTL of the same version, that you might be using elsewhere locally.

PR #41 (2017-03-10): Manage firrtl dependency manually
Fixes #40. 

This mainly serves to make it easier for new users who are unfamiliar with firrtl dependency problem. Existing users just need to be wary that it's managed manually in this repository.
closed 2017-03-10 by davidbiancolin