ucb-bar/fpga-zynq #42: "make rocket" fails with a 32 bit architecture

adrien-lessard, 2017-03-12:
## Problem

Building the rocket-chip on a 32-bit architecture causes a failure in
> testchipip/src/main/scala/Adapter.scala:99

## Steps to reproduce

1. Start with a fresh repository
2. Use this [configuration file](https://github.com/ucb-bar/fpga-zynq/files/836213/Configs.scala.txt) instead in \<repo\>/common/src/main/scala/
3. `$ cd <repo>/zedboard`
`$ make rocket CONFIG=Config32bitsFailure`
4. Observe the same output as here: [output_log.txt](https://github.com/ucb-bar/fpga-zynq/files/836212/output_log.txt)
5. You may observe that changing back to 64 bits in the provided configuration file fixes the issue

## Additional info

At the point of failure, here's the value of some relevant variables:

64 bits:
> w = 32
tlBeatAddrBits = 3
tlByteAddrBits = 3
tlDataBits = 64
pAddrBits = 32
tlDataBeats = 8
tlWriteMaskBits = 8
p(TLId) = L1toL2

32 bits:
> w = 32
tlBeatAddrBits = 4
tlByteAddrBits = 2
tlDataBits = 32
pAddrBits = 32
tlDataBeats = 16
tlWriteMaskBits = 4
p(TLId) = L1toL2

The line `addr(tlByteAddrBits - 1, log2Up(w/8))` in a 32-bit architecture causes to try accessing `addr` in the range `(1, 2)`, which is invalid

milapowel, 2017-03-15:
Hi, I have same problem with my RV32IMA configuration.

Generated Configuration String
plic {
  priority 0xc000000;
  pending 0xc001000;
  ndevs 0;
};
rtc {
  addr 0x200bff8;
};
ram {
  0 {
    addr 0x80000000;
    size 0x10000000;
  };
};
core {
  0 {
    0 {
      isa rv32imas;
      timecmp 0x2004000;
      ipi 0x2000000;
      plic {
        m {
         ie 0xc002000;
         thresh 0xc200000;
         claim 0xc200004;
        };
        s {
         ie 0xc002080;
         thresh 0xc201000;
         claim 0xc201004;
        };
      };
    };
  };
};
bootrom {
  addr 0x1000;
  size 0x1000;
}
clint {
  addr 0x2000000;
  size 0x10000;
}

Ended with same error as adrien-lessard.

adrien-lessard, 2017-04-04:
Any news on this issue?

davidbiancolin, 2017-06-20:
Hey Howie can you deal with this?

davidbiancolin, 2017-08-09:
I'm sorry, but we don't really have the cycles right now to support non-default configurations right now.

closed 2017-08-09 by davidbiancolin