ucb-bar/fpga-zynq #43: Linux on rocket-chip zybo : disk option has no effect

JishinMaster, 2017-03-17:
I have a Zybo board loaded with a rocket-chip core (riscv64ima) and I used the linux images obtained with "make fetch-riscv-linux".
I have cross-compiled busybox 1.26.2 and made a new root.bin (removed the old one).
When I use "./fesvr-zynq +disk=/sdcard/riscv/root.bin bbl /sdcard/riscv/vmlinux" or "./fesvr-zynq bbl /sdcard/riscv/vmlinux" it gives me the same result : linux boots and somehow has another filesystem than the one in my root.bin, and contains busybox 1.21.1.

The optional disk option does not seem to have any effect. Since I don't have another root.bin on my sdcard, where does it load busybox from?

seldridge, 2017-03-17:
I assume that this is coming from an initramfs. The `+disk` option has not been usable for some time. You can specify your own initramfs when you compile your kernel and then use that vmlinux as the payload when you build bbl. Some discussion here which may help: [1](https://github.com/riscv/riscv-pk/issues/18), [2](https://github.com/ucb-bar/fpga-zynq/issues/16).

Chris Celio has a base project that may help with this: https://github.com/ccelio/initramfs_linux_flow. 

However, you only need to run `./fesvr-zynq bbl` to boot linux as everything is packaged inside bbl.

davidbiancolin, 2017-03-17:
Yes, as @seldridge has suggested, if you are using the provided images, then the bbl instance you are loading onto the target contains an linux image with an initramfs baked in, that will be unpacked and mounted at boot.  @ccelio's base project above is quite useful. 

+disk is not longer supported.  I'm sorry for leaving the README out of date for so long.

closed 2017-03-17 by davidbiancolin

JishinMaster, 2017-03-18:
Thanks for the hints! I didn't manage to get the kernel build use initramfs.txt so I populated the filesystem myself. I used riscv-linux master and soft float seems to work okay on the Zybo.
Thanks again for the support!

Following is what I got : 

root@zynq:~# ./fesvr-zynq bblvmlinux 
              vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
                  vvvvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvv  
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvv    
rr                vvvvvvvvvvvvvvvvvvvvvv      
rr            vvvvvvvvvvvvvvvvvvvvvvvv      rr
rrrr      vvvvvvvvvvvvvvvvvvvvvvvvvv      rrrr
rrrrrr      vvvvvvvvvvvvvvvvvvvvvv      rrrrrr
rrrrrrrr      vvvvvvvvvvvvvvvvvv      rrrrrrrr
rrrrrrrrrr      vvvvvvvvvvvvvv      rrrrrrrrrr
rrrrrrrrrrrr      vvvvvvvvvv      rrrrrrrrrrrr
rrrrrrrrrrrrrr      vvvvvv      rrrrrrrrrrrrrr
rrrrrrrrrrrrrrrr      vv      rrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrr          rrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrr      rrrrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrrrr  rrrrrrrrrrrrrrrrrrrrrr

       INSTRUCTION SETS WANT TO BE FREE
[    0.000000] Linux version 4.6.2 (fou@foubuntu) (gcc version 6.1.0 (GCC) ) #36 Sat Mar 18 16:10:52 CET 2017
[    0.000000] Available physical memory: 248MB
[    0.000000] Initial ramdisk at: 0xffffffff80014c48 (2381461 bytes)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080800000-0x000000008fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080800000-0x000000008fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080800000-0x000000008fffffff]
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 62620
[    0.000000] Kernel command line: 
[    0.000000] PID hash table entries: 1024 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 5, 131072 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 244564K/253952K available (2149K kernel code, 107K rwdata, 564K rodata, 2412K init, 221K bss, 9388K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:0 nr_irqs:0 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 191126044627 ns
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=100000)
[    0.000000] pid_max: default: 32768 minimum: 301
[    0.000000] Mount-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.000000] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.000000] devtmpfs: initialized
[    0.000000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.000000] NET: Registered protocol family 16
[    0.000000] vgaarb: loaded
[    0.000000] clocksource: Switched to clocksource riscv_clocksource
[    0.000000] NET: Registered protocol family 2
[    0.000000] TCP established hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000000] TCP bind hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000000] TCP: Hash tables configured (established 2048 bind 2048)
[    0.000000] UDP hash table entries: 256 (order: 1, 8192 bytes)
[    0.000000] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes)
[    0.000000] NET: Registered protocol family 1
[    0.260000] Unpacking initramfs...
[    0.530000] console [sbi_console0] enabled
[    0.530000] futex hash table entries: 256 (order: 0, 6144 bytes)
[    0.540000] io scheduler noop registered
[    0.540000] io scheduler cfq registered (default)
[    0.600000] Freeing unused kernel memory: 2412K (ffffffff80000000 - ffffffff8025b000)
[... 60 lines cut ...]
	unix2dos, unlink, unlzma, unlzop, unshare, unxz, unzip, uptime, users,
	usleep, uudecode, uuencode, vconfig, vi, vlock, volname, wall, watch,
	watchdog, wc, wget, which, who, whoami, whois, xargs, xz, xzcat, yes,
	zcat, zcip
/ # ls /bin
ash            dumpkmap       kbd_mode       mv             setarch
base64         echo           kill           netstat        setserial
busybox        ed             linux32        nice           sh
cat            egrep          linux64        pidof          sleep
catv           false          ln             ping           stat
chattr         fatattr        login          ping6          stty
chgrp          fdflush        ls             pipe_progress  su
chmod          fgrep          lsattr         printenv       sync
chown          fsync          lzop           ps             tar
conspy         getopt         makemime       pwd            touch
cp             grep           mkdir          reformime      true
cpio           gunzip         mknod          rev            umount
cttyhack       gzip           mktemp         rm             uname
date           hostname       more           rmdir          usleep
dd             hush           mount          rpm            vi
df             ionice         mountpoint     run-parts      watch
dmesg          iostat         mpstat         scriptreplay   zcat
dnsdomainname  ipcalc         mt             sed

`

PR #44 (2017-03-17): Linux boot documentation fix
Fixes #43. Finally brings documentation regarding linux boot up to date with repository.