ucb-bar/fpga-zynq #47: How to customize the number of tiles/cores ?

yargo11, 2017-06-14:
I want to increase the number of cores/tiles in my project. So, what i need to do, and how i can check/confirm it`s working? 

I change the parameter NTILES for 4(fpga-zynq/rocket-chip/src/main/scala/uncore/builder.scala) and after this i executed the command $ make fpga-images-zezboard/boot.bin on zedboard directory.

Copy the files inside /fpga-zynq/zedboard/ for SD, insert SD on the Zedboard and turn on the board.
When i execute "grep -c ^processor /proc/cpuinfo", i got 2 as response for the command.
If a execute ./fesvr-zynq bbl and try execute "grep -c ^processor /proc/cpuinfo" again, i got 0 as response.

ben-k, 2017-06-16:
There are already Chisel hooks to change the number of cores in your design, so you can build a four-core design by modifying the ZynqConfig that is built by default.  In `common/src/main/scala/Configs.scala`, first add the line `import coreplex._` to get access to the appropriate config.  Then you can modify ZynqConfig like so:

```
class ZynqConfig extends Config(new WithNCores(4) ++ new WithZynqAdapter ++ new DefaultFPGAConfig)
```

Then building the boot.bin as normal should synthesize an FPGA bitfile with this configuration change (assuming your FPGA has the resources for four cores).

Also note that in order for Linux to recognize multiple cores, you'll need to make sure that SMP support is enabled (see e.g. riscv/riscv-tools/issues/87).

davidbiancolin, 2017-06-20:
FYI, four cores probably won't fit (I haven't tried); you should try two instead. Closing this issue.

closed 2017-06-20 by davidbiancolin