ucb-bar/fpga-zynq #48: can't connect zedboard by command " screen /dev/tty.usbmodem1411 115200,cs8,-parenb,-cstopb"

Pengnian, 2017-06-18:
Hi, All

I  can't connect the zedboard by command " screen /dev/tty.usbmodem1411 115200,cs8,-parenb,-cstopb ", please help me ,Thanks~

davidbiancolin, 2017-06-20:
Hey!

Does the device appear? 

and

Have you tried ethernet?

Pengnian, 2017-06-20:
Just difference device number(my usburat port device number is ttyACM0 not tty.usbmodem1411).Thank you very much. I want to ask another question,can i run program in risc-v linux  not ARM linux use pk, can you tell me how to operation ? Thank you very much again.

ninode, 2017-06-20:
Hello, 

I'm having the same issue with connecting the zedboard.

@Pengnian ould you please tell how you figured out that your port device is ttyACM0. I've tried it and it didn't respond. 

@davidbiancolin could you please provide further instruction to connecting the zedboard for beginners ? 

Thank you

Pengnian, 2017-06-21:
You should check zedboard usbuart device number  in your computer.  You can use command " ls  /dev",
when usb connecting your computer, and ejecting usb use command again.you can find your device number.

ninode, 2017-06-21:
It doesn't show actually. w do you configure the board's IP address as 192.168.5.1 ? Isn't 192.168.1.10 by default ?

davidbiancolin, 2017-06-21:
Our provided images configure the ARM to 192.168.1.5 as its default. I don't know why it was statically assigned that. 

Are you using Ethernet or USBUART?

ninode, 2017-06-21:
I'm using Ethernet and following the quick instructions to push the rocket to the zedboard.

Silly question, would the host recognize the board as 192.168.1.5 if the SD card is empty ? 

Thank you

davidbiancolin, 2017-06-21:
It would not. The ramdisk that includes all of that configuration information lives on the SD card.

ninode, 2017-06-22:
Thank you so much @davidbiancolin. I had my address set on 192.168.1.10 from a previous bring up. I've formatted the SD card and pushed the new files and everything works perfectly.

ben-k, 2017-06-22:
To answer the outstanding question from @Pengnian, to run Linux on the RISC-V core, you need to use the Berkeley Bootloader (BBL), not pk.  As described in the README, running the following on the ARM core should do it:

```
./fesvr-zynq bbl
```

I'm closing this as it seems all of the issues have been addressed.

closed 2017-06-22 by ben-k