ucb-bar/fpga-zynq #49: How to add my programs into uarmdisk.image.gz

Pengnian, 2017-06-21:
Hi all,

I have implemented loading rocket to my zedboard , want add my compiled programs into RAM linux (ur amdisk.image.gz) to running,just like hello(./fesvr-zynq pk hello).But I don't know what to do, plaese help me.Thanks~

davidbiancolin, 2017-06-21:
If you want to add programs that you can run on top of PK, you just need to include the compiled programs in the ARM's ramdisk. 

see Appendix B, and  make ```ramdisk_open``` and ```make ramdisk_close```.  Navigate into the home directory within the ramdisk and you should find the default bbl, pk, and hello we provide.

Pengnian, 2017-06-22:
@David Biancolin  Thanks very much! 😁

Pengnian, 2017-06-22:
@davidbiancolin hello, you can tell me the details about  how to use command "make ramdisk_open "  " make ramdisk_close", I just entering the directory of "uramdisk.image.gz",then  used commands. If you can tell me the details of operation,I will appreciate you very much.

ben-k, 2017-06-22:
You need to run these commands from the directory corresponding to your FPGA.  For example, from the root of the repo:

```
cd zedboard
make ramdisk-open
```

Note that the command uses a dash, not an underscore (this is correct in the README but not in the above comment).  Otherwise, just follow the instructions in the README.

davidbiancolin, 2017-06-22:
Oh dear. Thanks ben.

Pengnian, 2017-06-23:
@davidbiancolin @ben-k  oh,thanks two guys

I'm a beginner,your guides important for me,thanks for you patience !

davidbiancolin, 2017-06-26:
I'm closing this; reopen if you need a bit more help.

closed 2017-06-26 by davidbiancolin

ninode, 2017-08-01:
Hi @Pengnian, did you manage to add programs and run them with the command ? 
Have you tired to run any programs for GPIO ? 
Thanks