ucb-bar/fpga-zynq #50: Building a riscv rtos

ninode, 2017-06-22:
Hello, 

Is it possible to push a small rtos with rocket cores to the fpga instead of riscv-linux, using the build from scratch steps ? The rtos in question has its own gcc.

If this is possible, could you please point out which files are necessary to do and which steps could be followed in your "build from scratch" steps ? 

Regards

ben-k, 2017-06-22:
I think @grebe may be thinking about this question, but I don't know what the status is.

davidbiancolin, 2017-06-23:
That's a nontrivial undertaking, where you'd have to port the RTOS to run on top of BBL (our system is tethered).   

To my knowledge, [zephyr](https://github.com/fractalclone/zephyr-riscv), has been ported to both pulpino and SiFive platforms. Perhaps looking at those projects is a better place to start.

ninode, 2017-06-26:
@davidbiancolin Thank you, I will take a look at Zephyr porting projects. 
Could you please elaborate on the running RTOS on top of BBL part ? 

Thank you very much

davidbiancolin, 2017-06-26:
The "target" riscv processor we host in this project doesn't have a conventional console device, nor is there a block device. Console I/O is proxied over the tether to the ARM A9 host, and a host program (fesvr-zynq), running on the A9, loads program from which the core will eventually boot. You just need to be cognizant of the fact that the machine hosted in this repository, differs from the Freedom platform, despite the fact they are both derived from rocket-chip.

ninode, 2017-06-28:
Thank you @davidbiancolin, I understand.

So, the files required to run Rocket on the PL are BOOT.bin which is generated by Vivado.

But uImage, devicetree.dtb, and ramdisk depend on the OS. 

Assuming the OS is rewritten to communicate with fesvr, could I somehow change these steps or scripts ($ make arm-linux $ make arm-dtb) to fit my rtos port ?

davidbiancolin, 2017-07-12:
In principle, you shouldn't have to change anything in the bitstream if you port the RTOS right. You definitely shouldn't have to touch uImage, devicetree.dtb. You would need to change the ramdisk, just as you would to add new binaries to run on the RISCV core.

ninode, 2017-07-12:
Perfect. Thank you guys

closed 2017-07-12 by ninode

ninode, 2017-08-09:
Hi @davidbiancolin I want to run a LED blinking program on the rocket core. My program is in VHDL. Do you have any pointers on how I could make it work ?