ucb-bar/fpga-zynq #52: Bare-metal Rocket Chip on Zynq

ninode, 2017-07-05:
Hello, 

Is it possible to have a bare-metal config in build from scratch by omitting some steps ? 

Thank you !

ben-k, 2017-07-07:
This is a vague question - can you clarify?  The Rocket core in the default image can already run bare-metal programs in machine mode.

ninode, 2017-07-08:
Sorry for the silly question.

The pre-built image of ramdisk contains a linux image, so how does it run bare-metal ? 

Is it correct to say that up to step 3.7) Building/Obtaining riscv-linux, Rocket runs bare-metal ? 

What if I have a bootloader with another image other than linux, will the configuration up to 3.7 run it ? (I'm confused because the uImage and the dtb have linux peripherals for the ARM and that could be in the way of running another os). 

Again, thank you for your patience with my confusion. 

Thank you

ben-k, 2017-07-10:
I think the confusion on my part (and possibly on yours) is in distinguishing between the two different ways in which we "run Linux" in this system.  There are two different processor systems in play here - the dual-core ARM on the Zynq part, and the RISC-V Rocket softcore.

Step 3.7 is concerned with building a version of Linux to run on the RISC-V softcore.  This is one of several options for running software on Rocket.  If you've written and compiled them correctly, you can run programs truly bare-metal on Rocket:

```
./fesvr-zynq my-baremetal-program
```

You can also use the proxy kernel to handle some syscalls:

```
./fesvr-zynq pk my-c-program
```

Or you can boot Linux and then run programs inside of the OS in user mode:

```
./fesvr-zynq bbl
...  // Linux boots to command line
./my-user-program
```

But all of the above is concerned with how programs are run on Rocket, not on the ARM cores.  The links you provided are instead concerned with different ways of running software on the ARM cores themselves.  In the setup we provide, a single OS (Linux with SMP enabled) is running across both ARM cores.  It should certainly be possible to run different programs on each ARM core as you suggest, but I have no idea of how to go about it.  Probably following those app notes would be a good start.

ninode, 2017-07-10:
Thank you so much Ben ! This is perfect

closed 2017-07-10 by ninode

ninode, 2017-07-11:
Just one more question about the Rocket cores. The OS which is ported to RISC-V should be run as a bare-metal program (./fesvr-zynq my-baremetal-program) or a c program (./fesvr-zynq my-c-program)?

ben-k, 2017-07-11:
In this repo, the Linux OS is loaded into the memory of the RISC-V core and run by the Berkeley Boot Loader (BBL).  This program initializes peripherals, sets up the page table and virtual memory, loads the Linux kernel to virtual memory, and boots the kernel.  It basically acts as a hypervisor for the kernel (which runs in supervisor mode).

I don't know how you could boot Linux in a tethered system without something like BBL - something needs to set up the environment properly before Linux starts running.

ninode, 2017-07-11:
Actually not Linux, a static rtos for small embedded systems which is ported to the RISC-V (PULPino MCU).  The rtos image needs to be loaded on the Zedboard (rtos.elf).

In order to boot this rtos on the Zedboard development board, can we run this rtos.elf program with ./fesvr-zynq my-baremetal program (or ./fesvr-zynq my-c-program) ?