ucb-bar/fpga-zynq #53: Make rocket error

kllbrd, 2017-07-07:
`[root@localhost zedboard]# make rocket
rm -f /usr/local/fpga-zynq/common/../testchipip/lib
ln -s /usr/local/fpga-zynq/common/lib /usr/local/fpga-zynq/common/../testchipip/lib
cd /usr/local/fpga-zynq/common/../testchipip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /usr/local/rocket-chip/sbt-launch.jar package
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Set current project to testchipip (in build file:/usr/local/fpga-zynq/testchipip/)
[success] Total time: 0 s, completed Jul 7, 2017 7:04:53 PM
cp /usr/local/fpga-zynq/common/../testchipip/target/scala-2.11/*.jar /usr/local/fpga-zynq/common/lib
cp: cannot stat '/usr/local/fpga-zynq/common/../testchipip/target/scala-2.11/*.jar': No such file or directory
/usr/local/fpga-zynq/common/Makefrag.pkgs:2: recipe for target '/usr/local/fpga-zynq/common/lib/testchipip.stamp' failed
make: *** [/usr/local/fpga-zynq/common/lib/testchipip.stamp] Error 1
[root@localhost zedboard]# make rocket
rm -f /usr/local/fpga-zynq/common/../testchipip/lib
ln -s /usr/local/fpga-zynq/common/lib /usr/local/fpga-zynq/common/../testchipip/lib
cd /usr/local/fpga-zynq/common/../testchipip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /usr/local/rocket-chip/sbt-launch.jar package
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Set current project to testchipip (in build file:/usr/local/fpga-zynq/testchipip/)
[success] Total time: 0 s, completed Jul 7, 2017 7:06:47 PM
cp /usr/local/fpga-zynq/common/../testchipip/target/scala-2.11/*.jar /usr/local/fpga-zynq/common/lib
cp: cannot stat '/usr/local/fpga-zynq/common/../testchipip/target/scala-2.11/*.jar': No such file or directory
/usr/local/fpga-zynq/common/Makefrag.pkgs:2: recipe for target '/usr/local/fpga-zynq/common/lib/testchipip.stamp' failed
make: *** [/usr/local/fpga-zynq/common/lib/testchipip.stamp] Error 1
`
That happens because in this: /fpga-zynq/testchipip/target there is this: scala-2.10 and not 2.11.
I changed it from common/Makefrag.pkgs to 2.10, the error gone but I got a bunch of other errors situated in:
1.   /usr/local/fpga-zynq/common/src/main/scala/Configs.scala
2.   /usr/local/fpga-zynq/common/src/main/scala/SerialDriver.scala
3.   /usr/local/fpga-zynq/common/src/main/scala/Top.scala
4.   /usr/local/fpga-zynq/common/src/main/scala/TestHarness.scala
(sum of 72 errors)
What I'm doint wrong?
Any kind of help would be appreciated.
Thanks!

kllbrd, 2017-07-10:
solved

closed 2017-07-10 by kllbrd

ninode, 2017-07-12:
Hi @kllbrd, How did you solve it ? Did you initialize testchipip ?
Also, did make bitstream work ?

kllbrd, 2017-07-12:
Hello @ninode,
Initially I tried to give the path of my rocket-chip directory in fpga-zynq/common/Makefrag. 
Then I decide to have my fpga-zynq directory fully updated so I initialized both rocket-chip dir and testchipip.
So, yes, in essence that was the solution to this problem.
All the tools now are working properly, I have my zedboard running on rocket.
Greetings.