ucb-bar/fpga-zynq #55: make arm-uboot error

kllbrd, 2017-07-10:
Hello, when I try to make arm-uboot from my zedboard directory:

[root@localhost zedboard]# make arm-uboot
# compile board-compatible u-boot
git submodule update --init /run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx
# copy relevant configuration files
if [ -a soft_config/boards.cfg ] ; \
then \
    cp soft_config/boards.cfg /run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx/ ; \
fi;
cp soft_config/zynq_zed.h /run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx/include/configs/
# actually build
cd /run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx && make CROSS_COMPILE=arm-xilinx-linux-gnueabi- zynq_zed_config
make[1]: Entering directory '/run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx'
Configuring for zynq_zed board...
make[1]: Leaving directory '/run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx'
cd /run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx && make CROSS_COMPILE=arm-xilinx-linux-gnueabi- -j16
make[1]: Entering directory '/run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx'
make[1]: arm-xilinx-linux-gnueabi-gcc: Command not found
/bin/sh: arm-xilinx-linux-gnueabi-gcc: command not found
dirname: missing operand
Try 'dirname --help' for more information.
  GEN     include/autoconf.mk.dep
/bin/sh: arm-xilinx-linux-gnueabi-gcc: command not found
  GEN     include/autoconf.mk
/bin/sh: arm-xilinx-linux-gnueabi-gcc: command not found
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  CHK     include/generated/version_autogenerated.h
/bin/sh: arm-xilinx-linux-gnueabi-gcc: command not found
/bin/sh: arm-xilinx-linux-gnueabi-ld: command not found
  CC      lib/asm-offsets.s
/bin/sh: arm-xilinx-linux-gnueabi-gcc: command not found
/run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx/./Kbuild:34: recipe for target 'lib/asm-offsets.s' failed
make[2]: *** [lib/asm-offsets.s] Error 127
make[2]: *** Waiting for unfinished jobs....
  CC      arch/arm/lib/asm-offsets.s
/bin/sh: arm-xilinx-linux-gnueabi-gcc: command not found
/run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx/./Kbuild:83: recipe for target 'arch/arm/lib/asm-offsets.s' failed
make[2]: *** [arch/arm/lib/asm-offsets.s] Error 127
Makefile:1075: recipe for target 'prepare0' failed
make[1]: *** [prepare0] Error 2
make[1]: Leaving directory '/run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx'
../common/Makefrag:133: recipe for target 'arm-uboot' failed
make: *** [arm-uboot] Error 2

Any ideas? 
Thank you.

kllbrd, 2017-07-10:
update ---
solved 
new errors:

# compile board-compatible u-boot
git submodule update --init /run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx
# copy relevant configuration files
if [ -a soft_config/boards.cfg ] ; \
then \
    cp soft_config/boards.cfg /run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx/ ; \
fi;
cp soft_config/zynq_zed.h /run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx/include/configs/
# actually build
cd /run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx && make CROSS_COMPILE=arm-xilinx-linux-gnueabi- zynq_zed_config
make[1]: Entering directory '/run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx'
awk: symbol lookup error: awk: undefined symbol: mpfr_z_sub
Makefile:468: recipe for target 'zynq_zed_config' failed
make[1]: *** [zynq_zed_config] Error 127
make[1]: Leaving directory '/run/media/kllbrd/work/fpga-zynq/common/u-boot-xlnx'
../common/Makefrag:133: recipe for target 'arm-uboot' failed
make: *** [arm-uboot] Error 2

I can't find this -> zynq_zed_config.
Same problem for the Zybo.

Thanks.

kllbrd, 2017-07-11:
solved

closed 2017-07-11 by kllbrd

ninode, 2017-07-31:
Hello, how did you solve please ? Thank you

kllbrd, 2017-07-31:
Hello @ninode, you have to open a new terminal, source the settings64.sh of Vivado and have super user permissions.

ninode, 2017-07-31:
Thank you @kllbrd. I did source the settings of Vivado and SDK and still got the error. 

```
alpha@alpha-VirtualBox:~$ cd Xilinx/Vivado/2016.2/
alpha@alpha-VirtualBox:~/Xilinx/Vivado/2016.2$ source settings64.sh
alpha@alpha-VirtualBox:~/Xilinx/Vivado/2016.2$ cd ../../SDK/2016.2/
alpha@alpha-VirtualBox:~/Xilinx/SDK/2016.2$ source settings64.sh
alpha@alpha-VirtualBox:~/Xilinx/SDK/2016.2$ export XILINX_VIVADO=/home/alpha/Xilinx/Vivado
alpha@alpha-VirtualBox:~/Xilinx/SDK/2016.2$ cd
alpha@alpha-VirtualBox:~$ cd fpga-zynq/zedboard/
alpha@alpha-VirtualBox:~/fpga-zynq/zedboard$ make arm-uboot
# compile board-compatible u-boot
git submodule update --init /home/alpha/fpga-zynq/common/u-boot-xlnx
# copy relevant configuration files
if [ -a soft_config/boards.cfg ] ; \
	then \
	    cp soft_config/boards.cfg /home/alpha/fpga-zynq/common/u-boot-xlnx/ ; \
	fi;
cp soft_config/zynq_zed.h /home/alpha/fpga-zynq/common/u-boot-xlnx/include/configs/
# actually build
cd /home/alpha/fpga-zynq/common/u-boot-xlnx && make CROSS_COMPILE=arm-xilinx-linux-gnueabi- zynq_zed_config
make[1]: Entering directory `/home/alpha/fpga-zynq/common/u-boot-xlnx'
Configuring for zynq_zed board...
make[1]: Leaving directory `/home/alpha/fpga-zynq/common/u-boot-xlnx'
cd /home/alpha/fpga-zynq/common/u-boot-xlnx && make CROSS_COMPILE=arm-xilinx-linux-gnueabi- -j16
make[1]: arm-xilinx-linux-gnueabi-gcc: Command not found
make[1]: Entering directory `/home/alpha/fpga-zynq/common/u-boot-xlnx'
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
dirname: missing operand
Try 'dirname --help' for more information.
  GEN     include/autoconf.mk.dep
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
  GEN     include/autoconf.mk
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  CHK     include/generated/version_autogenerated.h
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
/bin/sh: 1: arm-xilinx-linux-gnueabi-ld: not found
  CC      lib/asm-offsets.s
  CC      arch/arm/lib/asm-offsets.s
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
make[2]: *** [lib/asm-offsets.s] Error 127
make[2]: *** Waiting for unfinished jobs....
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
make[2]: *** [arch/arm/lib/asm-offsets.s] Error 127
make[1]: *** [prepare0] Error 2
make[1]: Leaving directory `/home/alpha/fpga-zynq/common/u-boot-xlnx'
make: *** [arm-uboot] Error 2
```

Also, the arm-xilinx-linux-gnueabi is in the PATH and these parameters are set.
```
export CROSS_COMPILE=arm-xilinx-linux-gnueabi-
export ARCH=arm
```
and the location of arm-xilinx-linux-gnueabi-gcc:
```
alpha@alpha-VirtualBox:~/Xilinx/SDK/2016.2/gnu/arm/lin/bin$ which arm-xilinx-linux-gnueabi-gcc
/home/alpha/Xilinx/SDK/2016.2/gnu/arm/lin/bin/arm-xilinx-linux-gnueabi-gcc
```
I don't understand why it says not found.

kllbrd, 2017-07-31:
Try to do this:
Open a new terminal (not as root)
source the settings of Vivado: source ~/Vivado/2016.2/settings64.sh
then make arm-uboot
if it asks for root permissions try to su and re-make it.
@ninode

ninode, 2017-08-09:
Hi @kllbrd, thank you for helping. I tried this and it still gave the same error.