ucb-bar/fpga-zynq #56: make bitstream ERROR: [Common 17-70] Application Exception: Failed to launch run 'impl_1' due to failures in the following run(s):

ninode, 2017-07-10:
Hello, 

When I run make bistream for zed I get the following error. Even in GUI.

Does anyone know where this is coming from ? 

PS: I've checked my licence and it's working.

Thank you

`---------------------------------------------------------------------------------
Start Technology Mapping
---------------------------------------------------------------------------------
INFO: [Synth 8-4480] The timing for the instance top/FPGAZynqTop/DefaultCoreplex/tiles_0/HellaCache/mshrs/sdq_reg (implemented as a block RAM) might be sub-optimal as no optional output register could be merged into the block ram. Providing additional output register may help in improving timing.
INFO: [Synth 8-4480] The timing for the instance top/FPGAZynqTop/DefaultCoreplex/tiles_0/HellaCache/meta/_T_2352_0_reg (implemented as a block RAM) might be sub-optimal as no optional output register could be merged into the block ram. Providing additional output register may help in improving timing.
INFO: [Synth 8-4480] The timing for the instance top/FPGAZynqTop/DefaultCoreplex/tiles_0/HellaCache/meta/_T_2352_1_reg (implemented as a block RAM) might be sub-optimal as no optional output register could be merged into the block ram. Providing additional output register may help in improving timing.
INFO: [Synth 8-4480] The timing for the instance top/FPGAZynqTop/DefaultCoreplex/tiles_0/HellaCache/meta/_T_2352_2_reg (implemented as a block RAM) might be sub-optimal as no optional output register could be merged into the block ram. Providing additional output register may help in improving timing.
INFO: [Synth 8-4480] The timing for the instance top/FPGAZynqTop/DefaultCoreplex/tiles_0/HellaCache/meta/_T_2352_3_reg (implemented as a block RAM) might be sub-optimal as no optional output register could be merged into the block ram. Providing additional output register may help in improving timing.
src/central_freelist.cc:322] tcmalloc: allocation failed 16384 
/home/alpha/Xilinx/Vivado/2016.2/bin/loader: line 164:  3414 Killed                  "$RDI_PROG" "$@"
Parent process (pid 3414) has died. This helper process will now exit
[Mon Jul 10 00:40:50 2017] synth_1 finished
wait_on_run: Time (s): cpu = 00:00:01 ; elapsed = 00:51:03 . Memory (MB): peak = 1270.480 ; gain = 8.000 ; free physical = 1430 ; free virtual = 3060
# launch_runs impl_1 -to_step write_bitstream
ERROR: [Common 17-70] Application Exception: Failed to launch run 'impl_1' due to failures in the following run(s):
synth_1
These failed run(s) need to be reset prior to launching 'impl_1' again.

INFO: [Common 17-206] Exiting Vivado at Mon Jul 10 01:11:17 2017...`

davidbiancolin, 2017-07-12:
Do you know why the parent process died?

ninode, 2017-07-12:
Because of the make bitstream command (make_bitstream_ZynqConfig.tcl).
Make rocket and make project succeeded. And this happens right after the area optimization.

Also after the synthesis begins, this shows:
`
source src/tcl/make_bitstream_ZynqConfig.tcl
# open_project zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.xpr
Scanning sources...
Finished scanning sources
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/home/alpha/Xilinx/Vivado/2016.2/data/ip'.
# reset_run synth_1
# reset_run impl_1
# launch_runs synth_1
Adding cell -- xilinx.com:ip:proc_sys_reset:5.0 - proc_sys_reset_0
# launch_runs synth_1
Adding cell -- xilinx.com:ip:proc_sys_reset:5.0 - proc_sys_reset_0
INFO: [Ipptcl 7-1463] No Compatible Board Interface found. Board Tab not created in customize GUI`

Is this relevant or problematic ?

Here's more of log: 
`Report RTL Partitions: 
+------+------------------------+------------+----------+
|      |RTL Partition           |Replication |Instances |
+------+------------------------+------------+----------+
|1     |FPU__GB0                |           1|     27150|
|2     |FPU__GB1                |           1|     22422|
|3     |RocketTile__GCB0        |           1|     17734|
|4     |HellaCache              |           1|      9703|
|5     |RocketTile__GCB2        |           1|     11300|
|6     |L2BroadcastHub          |           1|     29520|
|7     |DefaultCoreplex__GCB1   |           1|      3335|
|8     |FPGAZynqTop__GC0        |           1|      8331|
|9     |ZynqAXISlave            |           1|       491|
|10    |rocketchip_wrapper__GC0 |           1|      1741|
---------------------------------------------------------------------------------
Start Timing Optimization
---------------------------------------------------------------------------------
---------------------------------------------------------------------------------
Start Applying XDC Timing Constraints
---------------------------------------------------------------------------------
---------------------------------------------------------------------------------
Finished Applying XDC Timing Constraints : Time (s): cpu = 00:02:51 ; elapsed = 00:10:51 . Memory (MB): peak = 2724.598 ; gain = 1841.562 ; free physical = 80 ; free virtual = 264`

I could provide all the log content if you want.

ninode, 2017-07-12:
Also I don't know if it was because I did it wrong. So here's what I did. 

-I have Vivado 2016.2 and its settings64.sh and a JVM that runs scala
-make init-submodules yileds *no rule to make target*
-in testchipip: git submodule update --init
-in rocket-chip! git submodule update --init
-in rocket-chip/riscv-tools: git submodule update --init --recursive and ./build.sh successfully
-in rocket-chip/emulator: make, make run and make debug (same in vsim).
-in fpga-zynq/zedboard: make rocket and make project successfully

davidbiancolin, 2017-07-12:
There is no Makefile in the root of this repository. You need to cd into zedboard/ first, before calling make init-submodules. If you do this you won't need to init the submodule manually. Also, you don't need to make anything in rocket-chip, aside from the toolchain if you don't already have one and need to compile new binaries. 

This should be sufficient to produce an updated boot.bin with the newly compiled bitstream.

```
cd zedboard
make init-submodules
make
```

Can you just attach the entire log?

ninode, 2017-07-12:
Ok, I'll retry. Thank you.

Here's the entire log file:
```
#-----------------------------------------------------------
# Vivado v2016.2 (64-bit)
# SW Build 1577090 on Thu Jun  2 16:32:35 MDT 2016
# IP Build 1577682 on Fri Jun  3 12:00:54 MDT 2016
# Start of session at: Wed Jul 12 06:40:36 2017
# Process ID: 14353
# Current directory: /home/alpha/fpga-zynq/zedboard
# Command line: vivado -mode tcl -source src/tcl/make_bitstream_ZynqConfig.tcl
# Log file: /home/alpha/fpga-zynq/zedboard/vivado.log
# Journal file: /home/alpha/fpga-zynq/zedboard/vivado.jou
#-----------------------------------------------------------
source src/tcl/make_bitstream_ZynqConfig.tcl
# open_project zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.xpr
Scanning sources...
Finished scanning sources
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/home/alpha/Xilinx/Vivado/2016.2/data/ip'.
# reset_run synth_1
# reset_run impl_1
# launch_runs synth_1
Adding cell -- xilinx.com:ip:proc_sys_reset:5.0 - proc_sys_reset_0
INFO: [Ipptcl 7-1463] No Compatible Board Interface found. Board Tab not created in customize GUI
Adding cell -- xilinx.com:ip:processing_system7:5.5 - processing_system7_0
Successfully read diagram <system> from BD file </home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/system.bd>
Verilog Output written to : /home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/hdl/system.v
Verilog Output written to : /home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/hdl/system_wrapper.v
Wrote  : </home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/system.bd> 
INFO: [BD 41-1029] Generation completed for the IP Integrator block proc_sys_reset_0 .
WARNING: [xilinx.com:ip:processing_system7:5.5-1] system_processing_system7_0_0: The Zynq BFM requires an AXI BFM license to run. Please ensure that you have purchased and setup the AXI BFM license prior to running simulation with this block. Please contact your Xilinx sales office for more information on purchasing this license
INFO: [BD 41-1029] Generation completed for the IP Integrator block processing_system7_0 .
INFO: [BD 41-1029] Generation completed for the IP Integrator block axi_interconnect_0/s00_couplers/auto_pc .
INFO: [BD 41-1029] Generation completed for the IP Integrator block axi_interconnect_1/s00_couplers/auto_pc .
Exporting to file /home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/hw_handoff/system.hwh
Generated Block Design Tcl file /home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/hw_handoff/system_bd.tcl
Generated Hardware Definition File /home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/hdl/system.hwdef
INFO: [HDL 9-2216] Analyzing Verilog file "/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v" into library work [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:1]
INFO: [HDL 9-2216] Analyzing Verilog file "/home/alpha/fpga-zynq/zedboard/src/verilog/rocketchip_wrapper.v" into library work [/home/alpha/fpga-zynq/zedboard/src/verilog/rocketchip_wrapper.v:1]
INFO: [HDL 9-1065] Parsing verilog file "/home/alpha/fpga-zynq/zedboard/src/verilog/clocking.vh" included at line 2. [/home/alpha/fpga-zynq/zedboard/src/verilog/rocketchip_wrapper.v:2]
INFO: [HDL 9-2216] Analyzing Verilog file "/home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/hdl/system.v" into library work [/home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/hdl/system.v:1]
[Wed Jul 12 06:41:22 2017] Launched synth_1...
Run output will be captured here: /home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.runs/synth_1/runme.log
launch_runs: Time (s): cpu = 00:00:24 ; elapsed = 00:00:34 . Memory (MB): peak = 1261.348 ; gain = 259.125 ; free physical = 142 ; free virtual = 5079
# wait_on_run synth_1
[Wed Jul 12 06:41:22 2017] Waiting for synth_1 to finish...

*** Running vivado
    with args -log rocketchip_wrapper.vds -m64 -mode batch -messageDb vivado.pb -notrace -source rocketchip_wrapper.tcl

****** Vivado v2016.2 (64-bit)
  **** SW Build 1577090 on Thu Jun  2 16:32:35 MDT 2016
  **** IP Build 1577682 on Fri Jun  3 12:00:54 MDT 2016
    ** Copyright 1986-2016 Xilinx, Inc. All Rights Reserved.

source rocketchip_wrapper.tcl -notrace
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/home/alpha/Xilinx/Vivado/2016.2/data/ip'.
Command: synth_design -top rocketchip_wrapper -part xc7z020clg484-1
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xc7z020'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7z020'
INFO: Launching helper process for spawning children vivado processes
INFO: Helper process launched with PID 14502 
[... cut ...]
	Parameter C_USE_S_AXI_HP0 bound to: 1 - type: integer 
	Parameter C_USE_S_AXI_HP1 bound to: 0 - type: integer 
	Parameter C_USE_S_AXI_HP2 bound to: 0 - type: integer 
	Parameter C_USE_S_AXI_HP3 bound to: 0 - type: integer 
	Parameter C_USE_M_AXI_GP0 bound to: 1 - type: integer 
	Parameter C_USE_M_AXI_GP1 bound to: 0 - type: integer 
	Parameter C_USE_S_AXI_GP0 bound to: 0 - type: integer 
	Parameter C_USE_S_AXI_GP1 bound to: 0 - type: integer 
	Parameter C_USE_S_AXI_ACP bound to: 0 - type: integer 

```

ninode, 2017-07-12:
log 2 

```
INFO: [Synth 8-5534] Detected attribute (* keep = "true" *) [/home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/ip/system_processing_system7_0_0/hdl/verilog/processing_system7_v5_5_processing_system7.v:1338]
INFO: [Synth 8-5534] Detected attribute (* keep = "true" *) [/home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/ip/system_processing_system7_0_0/hdl/verilog/processing_system7_v5_5_processing_system7.v:1339]
INFO: [Synth 8-638] synthesizing module 'BUFG' [/home/alpha/Xilinx/Vivado/2016.2/scripts/rt/data/unisim_comp.v:607]
INFO: [Synth 8-256] done synthesizing module 'BUFG' (39#1) [/home/alpha/Xilinx/Vivado/2016.2/scripts/rt/data/unisim_comp.v:607]
INFO: [Synth 8-638] synthesizing module 'BIBUF' [/home/alpha/Xilinx/Vivado/2016.2/scripts/rt/data/unisim_comp.v:268]
INFO: [Synth 8-256] done synthesizing module 'BIBUF' (40#1) [/home/alpha/Xilinx/Vivado/2016.2/scripts/rt/data/unisim_comp.v:268]
INFO: [Synth 8-638] synthesizing module 'PS7' [/home/alpha/Xilinx/Vivado/2016.2/scripts/rt/data/unisim_comp.v:33164]
INFO: [Synth 8-256] done synthesizing module 'PS7' (41#1) [/home/alpha/Xilinx/Vivado/2016.2/scripts/rt/data/unisim_comp.v:33164]
INFO: [Synth 8-256] done synthesizing module 'processing_system7_v5_5_processing_system7' (42#1) [/home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/ip/system_processing_system7_0_0/hdl/verilog/processing_system7_v5_5_processing_system7.v:156]
WARNING: [Synth 8-350] instance 'inst' of module 'processing_system7_v5_5_processing_system7' requires 685 connections, but only 672 given [/home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/ip/system_processing_system7_0_0/synth/system_processing_system7_0_0.v:453]
INFO: [Synth 8-256] done synthesizing module 'system_processing_system7_0_0' (43#1) [/home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/ip/system_processing_system7_0_0/synth/system_processing_system7_0_0.v:59]
WARNING: [Synth 8-350] instance 'processing_system7_0' of module 'system_processing_system7_0_0' requires 113 connections, but only 103 given [/home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/hdl/system.v:1463]
INFO: [Synth 8-256] done synthesizing module 'system' (44#1) [/home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.srcs/sources_1/bd/system/hdl/system.v:823]
INFO: [Synth 8-638] synthesizing module 'Top' [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:186516]
INFO: [Synth 8-638] synthesizing module 'FPGAZynqTop' [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:175498]
INFO: [Synth 8-638] synthesizing module 'TLXbar_socBus' [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:14]
WARNING: [Synth 8-3848] Net _GEN_2 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:293]
WARNING: [Synth 8-3848] Net _GEN_3 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:295]
WARNING: [Synth 8-3848] Net _GEN_4 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:297]
WARNING: [Synth 8-3848] Net _GEN_5 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:299]
WARNING: [Synth 8-3848] Net _GEN_6 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:301]
WARNING: [Synth 8-3848] Net _GEN_7 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:303]
WARNING: [Synth 8-3848] Net _GEN_8 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:305]
WARNING: [Synth 8-3848] Net _GEN_18 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:325]
WARNING: [Synth 8-3848] Net _GEN_19 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:327]
WARNING: [Synth 8-3848] Net _GEN_17 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:323]
WARNING: [Synth 8-3848] Net _GEN_9 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:307]
WARNING: [Synth 8-3848] Net _GEN_10 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:309]
WARNING: [Synth 8-3848] Net _GEN_11 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:311]
WARNING: [Synth 8-3848] Net _GEN_12 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:313]
WARNING: [Synth 8-3848] Net _GEN_13 in module/entity TLXbar_socBus does not have driver. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:315]
[... cut ...]
INFO: [Common 17-14] Message 'Synth 8-5544' appears 100 times and further instances of the messages will be disabled. Use the Tcl command set_msg_config to change the current settings.
INFO: [Synth 8-4471] merging register 'CONTROLReg_serial_reg[2:0]' into 'CONTROLReg_buserror_reg[2:0]' [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:166406]
WARNING: [Synth 8-3936] Found unconnected internal register 'sbAcqReg_addr_block_reg' and it is trimmed from '26' to '6' bits. [/home/alpha/fpga-zynq/zedboard/src/verilog/Top.ZynqConfig.v:166804]
---------------------------------------------------------------------------------
Finished 
```

ninode, 2017-07-12:
log3: 

```
Finished RTL Optimization Phase 2 : Time (s): cpu = 00:01:46 ; elapsed = 00:02:27 . Memory (MB): peak = 1864.160 ; gain = 981.125 ; free physical = 73 ; free virtual = 2893
---------------------------------------------------------------------------------

Report RTL Partitions: 
+------+------------------------+------------+----------+
|      |RTL Partition           |Replication |Instances |
+------+------------------------+------------+----------+
|1     |FPU__GB0                |           1|     34460|
|2     |FPU__GB1                |           1|     34686|
|3     |RocketTile__GCB0        |           1|     29182|
|4     |HellaCache              |           1|     13892|
|5     |RocketTile__GCB2        |           1|     13988|
|6     |L2BroadcastHub          |           1|     43943|
|7     |DefaultCoreplex__GCB1   |           1|      6878|
|8     |FPGAZynqTop__GC0        |           1|     13352|
|9     |ZynqAXISlave            |           1|       857|
|10    |rocketchip_wrapper__GC0 |           1|      2309|
+------+------------------------+------------+----------+
---------------------------------------------------------------------------------
Start RTL Component Statistics 
---------------------------------------------------------------------------------
Detailed RTL Component Info : 
+---Adders : 
	   2 Input    107 Bit       Adders := 1     
	   2 Input    105 Bit       Adders := 1     
	   2 Input     73 Bit       Adders := 1     
	   3 Input     65 Bit       Adders := 1     
	   2 Input     64 Bit       Adders := 8     
	   3 Input     64 Bit       Adders := 2     
	   2 Input     58 Bit       Adders := 2     
	   2 Input     55 Bit       Adders := 2     
	   2 Input     54 Bit       Adders := 3     
	   2 Input     49 Bit       Adders := 1     
	   2 Input     40 Bit       Adders := 3     
	   2 Input     39 Bit       Adders := 1     
	   2 Input     32 Bit       Adders := 3     
	   2 Input     29 Bit       Adders := 1     
	   2 Input     26 Bit       Adders := 4     
	   2 Input     25 Bit       Adders := 1     
	   2 Input     21 Bit       Adders := 1     
	   2 Input     20 Bit       Adders := 1     
	   2 Input     14 Bit       Adders := 3     
	   4 Input     14 Bit       Adders := 1     
	   3 Input     14 Bit       Adders := 2     
	   3 Input     13 Bit       Adders := 2     
	   2 Input     13 Bit       Adders := 2     
	   2 Input     12 Bit       Adders := 5     
	   4 Input     11 Bit       Adders := 1     
	   3 Input     11 Bit       Adders := 2     
	   2 Input     11 Bit       Adders := 3     
	   3 Input     10 Bit       Adders := 2     
	   2 Input      9 Bit       Adders := 2     
	   2 Input      8 Bit       Adders := 5     
	  14 Input      8 Bit       Adders := 1     
	   2 Input      7 Bit       Adders := 9     
	   2 Input      6 Bit       Adders := 7     
	   3 Input      6 Bit       Adders := 1     
	   2 Input      5 Bit       Adders := 12    
	  16 Input      5 Bit       Adders := 2     
	   2 Input      4 Bit       Adders := 21    
	   3 Input      4 Bit       Adders := 6     
	   8 Input      4 Bit       Adders := 2     
	   2 Input      3 Bit       Adders := 94    
	   3 Input      3 Bit       Adders := 3     
	   4 Input      3 Bit       Adders := 2     
	   2 Input      2 Bit       Adders := 11    
	   3 Input      2 Bit       Adders := 3     
[... 3129 lines cut ...]
|7     |DefaultCoreplex__GCB1   |           1|      3335|
|8     |FPGAZynqTop__GC0        |           1|      8331|
|9     |ZynqAXISlave            |           1|       491|
|10    |rocketchip_wrapper__GC0 |           1|      1741|
+------+------------------------+------------+----------+
---------------------------------------------------------------------------------
Start Timing Optimization
---------------------------------------------------------------------------------
---------------------------------------------------------------------------------
Start Applying XDC Timing Constraints
---------------------------------------------------------------------------------
---------------------------------------------------------------------------------
Finished Applying XDC Timing Constraints : Time (s): cpu = 00:02:51 ; elapsed = 00:10:51 . Memory (MB): peak = 2724.598 ; gain = 1841.562 ; free physical = 80 ; free virtual = 264
---------------------------------------------------------------------------------
Parent process (pid /home/alpha/Xilinx/Vivado/2016.2/bin/loader: line 164: 14499 Killed                  "$RDI_PROG" "$@"
14499) has died. This helper process will now exit
[Wed Jul 12 06:57:40 2017] synth_1 finished
wait_on_run: Time (s): cpu = 00:00:00.94 ; elapsed = 00:17:21 . Memory (MB): peak = 1269.254 ; gain = 7.906 ; free physical = 1406 ; free virtual = 2157
# launch_runs impl_1 -to_step write_bitstream
ERROR: [Common 17-70] Application Exception: Failed to launch run 'impl_1' due to failures in the following run(s):
synth_1
These failed run(s) need to be reset prior to launching 'impl_1' again.

INFO: [Common 17-206] Exiting Vivado at Wed Jul 12 07:00:14 
```

davidbiancolin, 2017-07-12:
Consult this [vivado forum entry](https://forums.xilinx.com/t5/Synthesis/Synthesis-failed-without-reporting-any-error/td-p/686000).

ninode, 2017-07-12:
Thank you, I'll take a look and close the issue.

closed 2017-07-12 by ninode