ucb-bar/fpga-zynq #58: programming zedboard led through ARM

ninode, 2017-07-18:
Hello,

I  want to run a program for LEDs with ./fesvr-zynq pk program-led (elf), but this requires modifying the block design of zedboard_rocketchip.
For a program targeting the IO of the zedboard (LED), I'm thinking of adding a DigiLED IP block into the design generated by make project, then export it to SDK, and then make a baremetal application which targets the LEDs. Is this the only way to work with programmable logic through linux (having the PS handle everything) ?

Thank you

ninode, 2017-08-09:
@davidbiancolin I'm working with a Zedboard in /fpga-zynq/zedboard and I want to run a LED blinking program ON THE ROCKET CORE and not on the arm core. Could you please provide any pointers ?

davidbiancolin, 2017-08-09:
You'll need to include a GPIO in the rocket-chip configuration you're building. In the top-level chisel project you'll need to wire out that signal to a bool you'll need to add the IO of the "Top" module. This will expose an I/O at the top level of the generated verilog. You'll then need to add this I/O to top-level module in rocketchip_wrapper.v, wiring out the signal from the module "top". Finally, you'll need to add a pin constraint to the project to ensure you're driving the right pin.

From the software side, the GPIO will be memory mapped so you should be able to write a small baremetal program to interact with it, which you'll load with PK. 

I'm going to suggest you have a look at: https://dev.sifive.com/freedom-soc/evaluate/fpga/

Which is probably better suited for the sorts of tinkering you are trying to do. 

- D

ninode, 2017-08-10:
>  In the top-level chisel project you'll need to wire out that signal to a bool you'll need to add the IO of the "Top" module.
Are you referring to this file: /fpga-zynq/rocket-chip/chisel3/src/main/resources/top.cpp ? 
It has timing parameters, where does the boolean go ? (Sorry for beginner questions)
After the bool is exposed (say IO), is this the correct wiring (attached) ?
[image]
[image]

davidbiancolin, 2017-08-10:
No, you need a GPIO to rocket-chip, not the vivado system block diagram. You will not have to touch that. 

You're going to need to write *scala*, and mix in one of [these](https://github.com/sifive/sifive-blocks/tree/master/src/main/scala/devices/gpio) into [this](https://github.com/ucb-bar/fpga-zynq/blob/f03982e6a155d333b3b3708185898745df5d4004/common/src/main/scala/Top.scala#L227) part of the code. Which means you'll need to add the sifive blocks as a submodule. 

This is not going to be a trivial endeavour, so i will reiterate, i *highly, highly, highly* recommend looking at the SiFive freedom prototyping setup that i linked to previously.

ninode, 2017-08-10:
Okay, thank you so much for your help. I sincerely appreciate it.

closed 2017-08-10 by ninode