ucb-bar/fpga-zynq #59: Instructions not working on Zedboard

kllbrd, 2017-07-25:
I followed all the steps, with the Zed booting properly when I login and run
`./fesvr-zynq pk hello`
or
`./fesvr-zynq bbl`
the system stucks without the possibility to interrupt with Ctrl+C.

My SD's directory environment:
SD/
-bbl
-BOOT.bin
-uImage
-devicetree.dtb
-uramdisk.image.gz

Any kind of help would be appreciated.
Thanks.

davidbiancolin, 2017-08-09:
Are you using the provided images, including the BBL and PK provided there?

kllbrd, 2017-08-10:
No that wasn't a problem. This is solved... 
Thanks anyway!

closed 2017-08-10 by kllbrd

clx7913, 2018-04-03:
@kllbrd Could you tell me how do you make it work?
I have the same problem.My board is also Zedboard and "./fesvr-zynq pk hello"  also doesn't work.