ucb-bar/fpga-zynq #60: ERROR make arm-uboot: arm-xilinx-linux-gnueabi-gcc: not found

ninode, 2017-07-31:
Hello, 
I have the following error in zedboard while running make arm-uboot. Could you please tell me how to solve it?

Thank you !
```
alpha@alpha-VirtualBox:~/fpga-zynq/zedboard$ make arm-uboot
# compile board-compatible u-boot
git submodule update --init /home/alpha/fpga-zynq/common/u-boot-xlnx
# copy relevant configuration files
if [ -a soft_config/boards.cfg ] ; \
	then \
	    cp soft_config/boards.cfg /home/alpha/fpga-zynq/common/u-boot-xlnx/ ; \
	fi;
cp soft_config/zynq_zed.h /home/alpha/fpga-zynq/common/u-boot-xlnx/include/configs/
# actually build
cd /home/alpha/fpga-zynq/common/u-boot-xlnx && make CROSS_COMPILE=arm-xilinx-linux-gnueabi- zynq_zed_config
make[1]: Entering directory `/home/alpha/fpga-zynq/common/u-boot-xlnx'
Configuring for zynq_zed board...
make[1]: Leaving directory `/home/alpha/fpga-zynq/common/u-boot-xlnx'
cd /home/alpha/fpga-zynq/common/u-boot-xlnx && make CROSS_COMPILE=arm-xilinx-linux-gnueabi- -j16
make[1]: execvp: arm-xilinx-linux-gnueabi-gcc: Not a directory
make[1]: Entering directory `/home/alpha/fpga-zynq/common/u-boot-xlnx'
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
dirname: missing operand
Try 'dirname --help' for more information.
  GEN     include/autoconf.mk.dep
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
  GEN     include/autoconf.mk
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  CHK     include/generated/version_autogenerated.h
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
/bin/sh: 1: arm-xilinx-linux-gnueabi-ld: not found
  CC      lib/asm-offsets.s
  CC      arch/arm/lib/asm-offsets.s
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
make[2]: *** [lib/asm-offsets.s] Error 127
make[2]: *** Waiting for unfinished jobs....
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
make[2]: *** [arch/arm/lib/asm-offsets.s] Error 127
make[1]: *** [prepare0] Error 2
make[1]: Leaving directory `/home/alpha/fpga-zynq/common/u-boot-xlnx'
make: *** [arm-uboot] Error 2

```

davidbiancolin, 2017-08-09:
You need to install the arm-xilinx-linux-gnueabi toolchain. You're going to be cross compiling code on your host machine for the zynq's arm core (arm-v7).

ninode, 2017-08-09:
@davidbiancolin 
This directory in SDK shows arm-xilinx-linux-gnueabi-gcc (which is not found in error)
```
alpha@alpha-VirtualBox:~/Xilinx/SDK/2016.2/gnu/arm/lin/bin$ ls
arm-xilinx-eabi-addr2line   arm-xilinx-linux-gnueabi-addr2line
arm-xilinx-eabi-ar          arm-xilinx-linux-gnueabi-ar
arm-xilinx-eabi-as          arm-xilinx-linux-gnueabi-as
arm-xilinx-eabi-c++         arm-xilinx-linux-gnueabi-c++
arm-xilinx-eabi-c++filt     arm-xilinx-linux-gnueabi-c++filt
arm-xilinx-eabi-cpp         arm-xilinx-linux-gnueabi-cpp
arm-xilinx-eabi-elfedit     arm-xilinx-linux-gnueabi-elfedit
arm-xilinx-eabi-g++         arm-xilinx-linux-gnueabi-g++
arm-xilinx-eabi-gcc         **arm-xilinx-linux-gnueabi-gcc**
arm-xilinx-eabi-gcc-4.9.2   arm-xilinx-linux-gnueabi-gcc-4.9.2
arm-xilinx-eabi-gcc-ar      arm-xilinx-linux-gnueabi-gcc-ar
arm-xilinx-eabi-gcc-nm      arm-xilinx-linux-gnueabi-gcc-nm
arm-xilinx-eabi-gcc-ranlib  arm-xilinx-linux-gnueabi-gcc-ranlib
arm-xilinx-eabi-gcov        arm-xilinx-linux-gnueabi-gcov
arm-xilinx-eabi-gdb         arm-xilinx-linux-gnueabi-gdb
arm-xilinx-eabi-gprof       arm-xilinx-linux-gnueabi-gprof
arm-xilinx-eabi-ld          arm-xilinx-linux-gnueabi-ld
arm-xilinx-eabi-nm          arm-xilinx-linux-gnueabi-nm
arm-xilinx-eabi-objcopy     arm-xilinx-linux-gnueabi-objcopy
arm-xilinx-eabi-objdump     arm-xilinx-linux-gnueabi-objdump
arm-xilinx-eabi-ranlib      arm-xilinx-linux-gnueabi-ranlib
arm-xilinx-eabi-readelf     arm-xilinx-linux-gnueabi-readelf
arm-xilinx-eabi-size        arm-xilinx-linux-gnueabi-size
arm-xilinx-eabi-sprite      arm-xilinx-linux-gnueabi-sprite
arm-xilinx-eabi-strings     arm-xilinx-linux-gnueabi-strings
arm-xilinx-eabi-strip       arm-xilinx-linux-gnueabi-strip
alpha@alpha-VirtualBox:~/Xilinx/SDK/2016.2/gnu/arm/lin/bin$ arm-xilinx-linux-gnueabi-gcc --version
bash: /home/alpha/Xilinx/SDK/2016.2/gnu/arm/lin/bin/arm-xilinx-linux-gnueabi-gcc: No such file or directory

```
Is this not the arm-xilinx-linux-gnueabi toolchain ?

Thank you

davidbiancolin, 2017-08-09:
It is... That last error is weird. It's on your PATH right?

ninode, 2017-08-09:
After source settings.sh in Vivado and SDK:
echo $PATH
`/home/alpha/Xilinx/Vivado/2016.2/bin:/home/alpha/Xilinx/Vivado_HLS/2016.2/bin:/home/alpha/Xilinx/DocNav:/home/alpha/Xilinx/SDK/2016.2/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/microblaze/lin/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/arm/lin/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/microblaze/linux_toolchain/lin64_be/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/microblaze/linux_toolchain/lin64_le/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/aarch64/lin/aarch64-linux/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/aarch64/lin/aarch64-none/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/home/alpha/Xilinx/SDK/2016.2/tps/lnx64/cmake-3.3.2/bin:/home/alpha/Xilinx/Vivado/2016.2/bin:/home/alpha/Xilinx/Vivado_HLS/2016.2/bin:/home/alpha/Xilinx/DocNav:/home/alpha/Xilinx/SDK/2016.2/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/microblaze/lin/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/arm/lin/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/microblaze/linux_toolchain/lin64_be/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/microblaze/linux_toolchain/lin64_le/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/aarch64/lin/aarch64-linux/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/aarch64/lin/aarch64-none/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/home/alpha/Xilinx/SDK/2016.2/tps/lnx64/cmake-3.3.2/bin:/bin:/usr/bin
`
arm-xilinx-linux-gnueabi-gcc doesn't show in PATH.

I added export PATH=/home/alpha/Xilinx/SDK/2016.2/gnu/arm/lin/bin:$PATH. Same error

I added export PATH=/home/alpha/Xilinx/SDK/2016.2/gnu/arm/lin/bin/gcc-arm-xilinx-linux-gnueabi:$PATH. Same error. 

I have found some answers with Codesourcery bin in PATH. Should that work ?

ninode, 2017-08-09:
Even in Codesourcery:

```
alpha@alpha-VirtualBox:~/CodeSourcery/bin$ ls
arm-linux-ar                        arm-xilinx-linux-gnueabi-as       arm-xilinx-linux-gnueabi-gcc-4.7.2   arm-xilinx-linux-gnueabi-ld       arm-xilinx-linux-gnueabi-sprite
arm-linux-gcc                       arm-xilinx-linux-gnueabi-c++      arm-xilinx-linux-gnueabi-gcc-ar      arm-xilinx-linux-gnueabi-nm       arm-xilinx-linux-gnueabi-strings
arm-linux-ld                        arm-xilinx-linux-gnueabi-c++filt  arm-xilinx-linux-gnueabi-gcc-nm      arm-xilinx-linux-gnueabi-objcopy  arm-xilinx-linux-gnueabi-strip
arm-linux-objcopy                   arm-xilinx-linux-gnueabi-cpp      arm-xilinx-linux-gnueabi-gcc-ranlib  arm-xilinx-linux-gnueabi-objdump
arm-linux-objdump                   arm-xilinx-linux-gnueabi-elfedit  arm-xilinx-linux-gnueabi-gcov        arm-xilinx-linux-gnueabi-ranlib
arm-xilinx-linux-gnueabi-addr2line  arm-xilinx-linux-gnueabi-g++      arm-xilinx-linux-gnueabi-gdb         arm-xilinx-linux-gnueabi-readelf
arm-xilinx-linux-gnueabi-ar         arm-xilinx-linux-gnueabi-gcc      arm-xilinx-linux-gnueabi-gprof       arm-xilinx-linux-gnueabi-size

alpha@alpha-VirtualBox:~/CodeSourcery/arm-xilinx-linux-gnueabi/bin$ ls
ar  as  c++  g++  gcc  ld  nm  objcopy  objdump  ranlib  README.txt  strip

export PATH=/home/alpha/CodeSourcery/arm-xilinx-linux-gnueabi/bin:/home/alpha/CodeSourcery/bin:$PATH

```
```
alpha@alpha-VirtualBox:~/fpga-zynq/zedboard$ make arm-uboot
# compile board-compatible u-boot
git submodule update --init /home/alpha/fpga-zynq/common/u-boot-xlnx
# copy relevant configuration files
if [ -a soft_config/boards.cfg ] ; \
	then \
	    cp soft_config/boards.cfg /home/alpha/fpga-zynq/common/u-boot-xlnx/ ; \
	fi;
cp soft_config/zynq_zed.h /home/alpha/fpga-zynq/common/u-boot-xlnx/include/configs/
# actually build
cd /home/alpha/fpga-zynq/common/u-boot-xlnx && make CROSS_COMPILE=arm-xilinx-linux-gnueabi- zynq_zed_config
make[1]: Entering directory `/home/alpha/fpga-zynq/common/u-boot-xlnx'
Configuring for zynq_zed board...
make[1]: Leaving directory `/home/alpha/fpga-zynq/common/u-boot-xlnx'
cd /home/alpha/fpga-zynq/common/u-boot-xlnx && make CROSS_COMPILE=arm-xilinx-linux-gnueabi- -j16
make[1]: arm-xilinx-linux-gnueabi-gcc: Command not found
make[1]: Entering directory `/home/alpha/fpga-zynq/common/u-boot-xlnx'
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
dirname: missing operand
Try 'dirname --help' for more information.
  GEN     include/autoconf.mk.dep
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
  GEN     include/autoconf.mk
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  CHK     include/generated/version_autogenerated.h
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
/bin/sh: 1: arm-xilinx-linux-gnueabi-ld: not found
  CC      lib/asm-offsets.s
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
make[2]: *** [lib/asm-offsets.s] Error 127
make[2]: *** Waiting for unfinished jobs....
  CC      arch/arm/lib/asm-offsets.s
/bin/sh: 1: arm-xilinx-linux-gnueabi-gcc: not found
make[2]: *** [arch/arm/lib/asm-offsets.s] Error 127
make[1]: *** [prepare0] Error 2
make[1]: Leaving directory `/home/alpha/fpga-zynq/common/u-boot-xlnx'
make: *** [arm-uboot] Error 2
```
Is something wrong here fpga-zynq/common/u-boot-xlnx. Should the gcc be located here or something ?

davidbiancolin, 2017-08-10:
So you shouldn't have to install it yourself. The vivado provided toolchain is fine. It looks like it's in your path. 

Starting from scratch, and just sourcing the vivado setting, confirm that ```which arm-xilinx-linux-gnueabi-gcc``` succeeds, then check the permissions.

ninode, 2017-08-10:
```
alpha@alpha-VirtualBox:~/Xilinx/Vivado/2016.2$ source settings64.sh
alpha@alpha-VirtualBox:~/Xilinx/Vivado/2016.2$ echo $PATH
/home/alpha/Xilinx/Vivado/2016.2/bin:/home/alpha/Xilinx/Vivado_HLS/2016.2/bin:/home/alpha/Xilinx/DocNav:/home/alpha/Xilinx/SDK/2016.2/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/microblaze/lin/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/arm/lin/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/microblaze/linux_toolchain/lin64_be/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/microblaze/linux_toolchain/lin64_le/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/aarch64/lin/aarch64-linux/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/aarch64/lin/aarch64-none/bin:/home/alpha/Xilinx/SDK/2016.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/home/alpha/Xilinx/SDK/2016.2/tps/lnx64/cmake-3.3.2/bin:/bin:/usr/bin
alpha@alpha-VirtualBox:~/Xilinx/Vivado/2016.2$ which arm-xilinx-linux-gnueabi-gcc
/home/alpha/Xilinx/SDK/2016.2/gnu/arm/lin/bin/arm-xilinx-linux-gnueabi-gcc

```
Yes, which arm-xilinx-linux-gnueabi-gcc succeeds. 

make arm-uboot in /fpga-zynq/zedboard still fails.

davidbiancolin, 2017-08-10:
can you check it's permissions as well. 

``` ls -l `which arm-xilinx-linux-gnueabi-gcc` ```

ninode, 2017-08-10:
```
alpha@alpha-VirtualBox:~/fpga-zynq/zedboard$ ls -l `which arm-xilinx-linux-gnueabi-gcc`
-rwxr-xr-x 1 alpha alpha 735360 juin   3  2016 /home/alpha/Xilinx/SDK/2016.2/gnu/arm/lin/bin/arm-xilinx-linux-gnueabi-gcc
```

ninode, 2017-08-11:
Do you think I should reinstall the Xilinx tools ?

davidbiancolin, 2017-08-11:
If our conversation elsewhere is any indication, you don't need this cross compiler anyways. 

What OS & architecture are you using? The cross compiler is a 32bit binary and you probably trying to run it on a 64 bit machine without the appropriate support. 

What does ```file ls -l `which arm-xilinx-linux-gnueabi-gcc` ``` say?

ninode, 2017-08-12:
file `which arm-xilinx-linux-gnueabi-gcc yields the following:
```
alpha@alpha-VirtualBox:~/fpga-zynq/zedboard$ ls -l `which arm-xilinx-linux-gnueabi-gcc`
-rwxr-xr-x 1 alpha alpha 735360 juin   3  2016 /home/alpha/Xilinx/SDK/2016.2/gnu/arm/lin/bin/arm-xilinx-linux-gnueabi-gcc
```

ninode, 2017-08-12:
> If our conversation elsewhere is any indication, you don't need this cross compiler anyways.

Alright. I can just use the prebuilt devicetree.dtb, uImage and uramdisk.image.gz files, so the make arm-uboot won't be necessary.

Chaning the scala means that the verilog (wrapper.v) will change and consequently the .bit file. So the boot.bin file will need to be rebuilt. Cross compiling isn't necessary for that ? So yes i won't need arm-xilinx-linux-gnueabi-gcc. 

> What OS & architecture are you using? The cross compiler is a 32bit binary and you probably trying to run it on a 64 bit machine without the appropriate support.
I'm on virtual Ubuntu 14.04 64-bit. And yes I am apparently using a 32-compiler with a 64 machine :/ 
Isn't there a 64-bit version of the compiler ?

ninode, 2017-08-25:
Hi @davidbiancolin, I have another issue regarding the gnueabi-gcc error. 
When I open-ramdisk, copy in a binary, then close it. I make fetch-ramdisk. 
When I load the deliver_output files to the SD card, the ramdisk content doesn't change, it doesn't have the binary I added. 

Should I make make arm-uboot after closing and removing the ramdisk ? 

In that case, running make arm-uboot give the gnueabi-gcc error. 

Thank you very much for your help.

closed 2017-09-17 by ninode

KcMeterCEC, 2018-07-18:
@ninode Hi,I think this is the solution ->[you should install ia32-libs](https://www.xilinx.com/support/answers/63561.html)