ucb-bar/fpga-zynq #61: Errors building the FSBL

ninode, 2017-08-01:
Hello, 

When bulding the FSBL, I got errors for make. 

Problems:

```
Description	Resource	Path	Location	Type
make: *** [ps7_cortexa9_0/libsrc/coresightps_dcc_v1_2/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/devcfg_v3_4/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/dmaps_v2_1/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/emacps_v3_2/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/gpiops_v3_1/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/qspips_v3_3/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/scugic_v3_3/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/scutimer_v2_1/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/scuwdt_v2_1/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/sdps_v2_8/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/standalone_v5_5/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/ttcps_v3_1/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/uartps_v3_1/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/usbps_v2_4/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/xadcps_v2_2/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [ps7_cortexa9_0/libsrc/xilffs_v3_3/src/make.libs] Error 2	FSBL_bsp		 	C/C++ Problem
make: *** [src/fsbl_handoff.o] Error 1	FSBL		 	C/C++ Problem
make: Target `all' not remade because of errors.	FSBL_bsp		 	C/C++ Problem
make[1]: *** [cc932.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [diskio.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [ff.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [standalone_libs] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xadcps_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xadcps_intr.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xadcps_selftest.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xadcps_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xadcps.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xcoresightpsdcc.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xdevcfg_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xdevcfg_hw.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xdevcfg_intr.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xdevcfg_selftest.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xdevcfg_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xdevcfg.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xdmaps_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xdmaps_hw.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xdmaps_selftest.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xdmaps_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xdmaps.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xemacps_bdring.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xemacps_control.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xemacps_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xemacps_hw.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xemacps_intr.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xemacps_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xemacps.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xgpiops_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xgpiops_hw.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xgpiops_intr.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xgpiops_selftest.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xgpiops_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xgpiops.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xqspips_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xqspips_hw.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xqspips_options.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xqspips_selftest.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xqspips_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xqspips.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscugic_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscugic_hw.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscugic_intr.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscugic_selftest.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscugic_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscugic.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscutimer_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscutimer_selftest.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscutimer_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscutimer.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscuwdt_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscuwdt_selftest.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscuwdt_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xscuwdt.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xsdps_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xsdps_options.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xsdps_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xsdps.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xttcps_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xttcps_options.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xttcps_selftest.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xttcps_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xttcps.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xuartps_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xuartps_hw.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xuartps_intr.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xuartps_options.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xuartps_selftest.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xuartps_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xuartps.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xusbps_endpoint.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xusbps_g.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xusbps_hw.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xusbps_intr.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xusbps_sinit.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: *** [xusbps.o] Error 1	FSBL_bsp		 	C/C++ Problem
make[1]: Target `libs' not remade because of errors.	FSBL_bsp		 	C/C++ Problem
```
Console:

```
make pre-build main-build 
a9-linaro-pre-build-step
 
Building file: ../src/fsbl_handoff.S
Invoking: ARM v7 gcc compiler
arm-none-eabi-gcc -Wall -O0 -g3 -I"/home/alpha/fpga-zynq/zedboard/zedboard_rocketchip_ZynqConfig/zedboard_rocketchip_ZynqConfig.sdk/rocketchip_wrapper_hw_platform_0" -c -fmessage-length=0 -MT"src/fsbl_handoff.o" -mcpu=cortex-a9 -mfpu=vfpv3 -mfloat-abi=hard -I../../FSBL_bsp/ps7_cortexa9_0/include -MMD -MP -MF"src/fsbl_handoff.d" -MT"src/fsbl_handoff.o" -o "src/fsbl_handoff.o" "../src/fsbl_handoff.S"
arm-none-eabi-gcc: error trying to exec 'cc1': execvp: No such file or directory
make: *** [src/fsbl_handoff.o] Error 1
```
Is this because of CROSS_COMPILE in SDK ? How do I know that I have the right gcc to compile C++ code in SDK ? Or is there another problem ? 

Thank you

davidbiancolin, 2017-08-09:
Why are you trying to rebuild the FSBL? Can you also give me one of the errors from the failing libs build.

ninode, 2017-08-10:
I think this is related to #60. 
Also, I rebuilt the FSBL because I thought I needed it for the LED program in #58.
Since this is done by modifying scala sources, I guess I won't be needing to rebuild the FSBL right ?

davidbiancolin, 2017-08-10:
You shouldn't have to touch the FSBL if this is for the LED program you described in #58. 

But yes, this is the result of your toolchain struggles.

ninode, 2017-08-10:
Okay, just to make sure.

I probably didn't get the following correctly:

> – First stage bootloader
> • Copy the second stage bootloader to DDR RAM
> • Uncached copy (mapping DDR RAM to IO)
> • Re-map DDR RAM to memory address 0
> • Reset Rocket
 
To target GPIO by the rocket core I need to modify the scala sources, so that the bitstream is modified. But to run binaries on the rocket core, I could add these binaries to the ramdisk and run them with pk. So that the uramdisk.image.gz is modified ?

What is the actual RAM space of the rocket chip ? DDR RAM or ramdisk (open, put in binaries, close)Does the ramdisk have access to the rocket ram space ? 

Sorry for the confusion.

davidbiancolin, 2017-08-10:
There is no need to apologize; this is a confusing subject because there are two different computing systems at play here, each with their own memory hierarchies and software stacks. 

The first is Rocket-chip. This is synthesized into the LUTs of the FPGA. We refer to Rocket-chip as the *target*. 

The second is the hardened ARM core on the zynq. It's "hardened" because it's directly implemented in silicon and attached to the reconfigurable fabric (the FPGA-y part of the zynq chip). We call the ARM the *host*. 

These two computers are interconnected with a *tether*.  The *target* relies on the *host* to help it boot. The *host* initializes the *target's* DRAM, by writing directly into it's memory system with the tether. The *host* then instructs the *target* to start reading instructions from its newly initialized DRAM.   

The first-stage bootloader (FSBL) referred to there is the FSBL of rocketchip -- the target's FSBL. While the FSBL you were trying to build above, is the FSBL for the hardened ARM core on the FPGA -- the host's FSBL.

There are also two ramdisks. One belongs to the *host*,  that is the uramdisk.image.gz. The ARM core mounts that image when it boots. The second, target ramdisk, is included in the BBL when you want to run linux. 

When you "run" the proxy kernel (PK) and a RISC-V application, both the PK and the binary live on  on the *host's* filesystem, which was initialized by uramdrisk.image.gz. The fesvr-zynq program does the real magic that manages the tether i described above, and copies both these files into the *target's* memory system. 

The only *host* system thing you need to touch is the uramdrisk.image.gz, and only to include 
the binary -- let's call it <blink-some-leds.riscv> -- which fesvr-zynq will copy to your modified rocket-chip with PK. You should not have to touch PK.  

This blink-some-leds.riscv program will achieve it's goal by writing to *target* memory mapped registers of the GPIO device you added to the *target*. In fact, when blink-some-leds.riscv executes it actually has no idea the *host* even exists, the only memory system it can interact with is the *target's* memory system. This blink-some-leds.riscv program should feel exactly like writing a blinky program for any other microcontroller. 

The real challenge is understanding how to modify the scala to include this GPIO hardware in the *target*.

ninode, 2017-08-11:
Thank you so much !!

> This blink-some-leds.riscv program should feel exactly like writing a blinky program for any other microcontroller.

Can I use the Xilinx [gpio example](https://github.com/Micro-Studios/Xilinx-GPIO-Interrupt/blob/master/GPIO_Interrupt.sdk/GPIO_Interrupt/src/helloworld.c) as blink-some-leds.riscv ?

closed 2017-08-22 by ninode