ucb-bar/fpga-zynq #62: make ramdisk-close ERROR mkimage not found

ninode, 2017-08-09:
I copied a binary in ramdisk and this happened when closing the ramdisk. Would you know why ? 
Thank you !
```
root@alpha-VirtualBox:/home/alpha/fpga-zynq/zedboard# make ramdisk-close
sh -c 'cd ramdisk/ && sudo find . | sudo cpio -H newc -o' | gzip -9 > uramdisk.cpio.gz
37049 blocks
mkimage -A arm -O linux -T ramdisk -d uramdisk.cpio.gz fpga-images-zedboard/uramdisk.image.gz
/bin/bash: mkimage: command not found
make: *** [ramdisk-close] Error 127
```

davidbiancolin, 2017-08-10:
Do you have mkimage installed?

ninode, 2017-08-10:
sudo apt-get install u-boot-tools solves it. I thought it had something to do with the arm-uboot problems in the other issue.

closed 2017-08-10 by ninode