ucb-bar/fpga-zynq #66: Newer version of Vivado not supported

rocketchiphelppls, 2017-12-07:
hi

i tried your rocketchip generator, but i cannot create a vivado project with the provided tcl scripts as my version is newer (2017 version is not working).

could you please hekp me out how to manually create the project? i know how to create IPs, but have no idea what parameters to use and how the connection has to be set up.

thank you very much in advance!

qant-um, 2017-12-08:
Hi,
Generally in tcl file, you can find this code:

################################################################
# Check if script is running in correct Vivado version.
################################################################
set scripts_vivado_version 2016.2
set current_vivado_version [version -short]

if { [string first $scripts_vivado_version $current_vivado_version] == -1 } {
   puts ""
   catch {common::send_msg_id "BD_TCL-109" "ERROR" "This script was generated using Vivado <$scripts_vivado_version> and is being run in <$current_vivado_version> of Vivado. Please run the script in Vivado <$scripts_vivado_version> then open the design in Vivado <$current_vivado_version>. Upgrade the design by running \"Tools => Report => Report IP Status...\", then run write_bd_tcl to create an updated script."}

   return 1
}

So you can change, the version number (here 2016.2) in your Vivado version.
I hope this can help you.

davidbiancolin, 2018-01-12:
See [this script](common/scripts/upgrade_version.sh).

closed 2018-01-12 by davidbiancolin