ucb-bar/fpga-zynq #69: priv-1.10 images

heshamelmatary, 2018-02-27:
Would be great if this repo can be updated with images that are synced with priv-1.10 version of the Rocket Chip and riscv-tools (e.g. https://github.com/freechipsproject/rocket-chip/commit/86c10b3cef66e676d76a4b7fd8eada20f83ad2b8)

chrisguikema, 2018-03-02:
We have programs running with the 1.10 version using spike and QEMU. Using the latest version of these images on the zedboard gives the following errors:

```
root@zynq:~# ./fesvr-zynq sel4test-driver-image-riscv-spike.bbl
../machine/mtrap.c:16: machine mode: unhandlable trap 2 @ 0x0000000080002330
```

Having the 1.10 synced version of this repository would help.

heshamelmatary, 2018-03-03:
@chrisguikema Right, that's because the current pre-built images (most importantly boot.bin that has the rocket-chip bit file) in this repo are priv-1.9 i.e. won't run priv-1.10 binaries. sel4test image works fine on QEMU and Spike given that both support priv-1.10.