ucb-bar/fpga-zynq #71: make project failed

clx7913, 2018-03-20:
I followed the 3.1) Project Setup of the README.md.
After typing "make project" in the directory of /fpga-zynq/zedboard,I encountered one error.
The log information is here:

root@vm:/home/chang/fpga-zynq/zedboard# make project
make -C /home/chang/fpga-zynq/rocket-chip/firrtl SBT="java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/chang/fpga-zynq/rocket-chip/sbt-launch.jar" root_dir=/home/chang/fpga-zynq/rocket-chip/firrtl build-scala
make[1]: Entering directory `/home/chang/fpga-zynq/rocket-chip/firrtl'
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/chang/fpga-zynq/rocket-chip/sbt-launch.jar "assembly"
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
Getting org.scala-sbt sbt 0.13.12 ...

:: problems summary ::
:::: WARNINGS
		module not found: org.scala-sbt#sbt;0.13.12

	==== local: tried

	  /root/.ivy2/local/org.scala-sbt/sbt/0.13.12/ivys/ivy.xml

	  -- artifact org.scala-sbt#sbt;0.13.12!sbt.jar:

	  /root/.ivy2/local/org.scala-sbt/sbt/0.13.12/jars/sbt.jar

	==== Maven Central: tried

	  https://repo1.maven.org/maven2/org/scala-sbt/sbt/0.13.12/sbt-0.13.12.pom

	  -- artifact org.scala-sbt#sbt;0.13.12!sbt.jar:

	  https://repo1.maven.org/maven2/org/scala-sbt/sbt/0.13.12/sbt-0.13.12.jar

	==== typesafe-ivy-releases: tried

	  https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/sbt/0.13.12/ivys/ivy.xml

	==== sbt-ivy-snapshots: tried

	  https://repo.scala-sbt.org/scalasbt/ivy-snapshots/org.scala-sbt/sbt/0.13.12/ivys/ivy.xml

		::::::::::::::::::::::::::::::::::::::::::::::

		::          UNRESOLVED DEPENDENCIES         ::

		::::::::::::::::::::::::::::::::::::::::::::::

		:: org.scala-sbt#sbt;0.13.12: not found

		::::::::::::::::::::::::::::::::::::::::::::::

:::: ERRORS
	Server access Error: Connection refused (Connection refused) url=https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/sbt/0.13.12/ivys/ivy.xml

:: USE VERBOSE OR DEBUG MESSAGE LEVEL FOR MORE DETAILS
unresolved dependency: org.scala-sbt#sbt;0.13.12: not found
Error during sbt execution: Error retrieving required libraries
  (see /root/.sbt/boot/update.log for complete log)
Error: Could not retrieve sbt 0.13.12
make[1]: *** [/home/chang/fpga-zynq/rocket-chip/firrtl/utils/bin/firrtl.jar] Error 1
make[1]: Leaving directory `/home/chang/fpga-zynq/rocket-chip/firrtl'
make: *** [/home/chang/fpga-zynq/rocket-chip/firrtl/utils/bin/firrtl.jar] Error 2
root@vm:/home/chang/fpga-zynq/zedboard# 

Anyone knows what's wrong with it ?

davidbiancolin, 2018-03-20:
This repo uses an older version of SBT. You'll need 0.13.12.

clx7913, 2018-03-21:
I haven't found the resource of sbt 0.13.12.

It seems that the official website has already stopped support.

Do you have the resource?

clx7913, 2018-03-21:
Ok,I think I'm really a noob.

Now I've installed the sbt 0.13.12 by "apt-get".

clx7913, 2018-03-21:
I type "make project" again and get another error 

Error wrapping InputStream in GZIPInputStream: java.util.zip.ZipException: Not in GZIP format
	at sbt.ErrorHandling$.translate(ErrorHandling.scala:10)
	at sbt.WrapUsing.open(Using.scala:34)
	at sbt.Using.apply(Using.scala:23)
	at sbt.IO$$anonfun$gzipFileIn$1.apply(IO.scala:863)
	at sbt.IO$$anonfun$gzipFileIn$1.apply(IO.scala:862)
	at sbt.Using.apply(Using.scala:24)
	at sbt.IO$.gzipFileIn(IO.scala:862)
	at sbt.Sync$.readUncaught(Sync.scala:88)
	at sbt.Sync$.readInfo(Sync.scala:84)
	at sbt.Sync$$anonfun$apply$1.apply(Sync.scala:28)
	at sbt.Sync$$anonfun$apply$1.apply(Sync.scala:22)
	at sbt.Defaults$$anonfun$copyResourcesTask$1.apply(Defaults.scala:922)
	at sbt.Defaults$$anonfun$copyResourcesTask$1.apply(Defaults.scala:918)
	at scala.Function4$$anonfun$tupled$1.apply(Function4.scala:35)
	at scala.Function4$$anonfun$tupled$1.apply(Function4.scala:34)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:237)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.util.zip.ZipException: Not in GZIP format
	at java.util.zip.GZIPInputStream.readHeader(GZIPInputStream.java:165)
	at java.util.zip.GZIPInputStream.<init>(GZIPInputStream.java:79)
	at sbt.Using$$anonfun$gzipInputStream$1.apply(Using.scala:84)
	at sbt.Using$$anonfun$gzipInputStream$1.apply(Using.scala:84)
	at sbt.Using$$anon$1.openImpl(Using.scala:51)
	at sbt.WrapUsing$$anonfun$open$2.apply(Using.scala:34)
	at sbt.ErrorHandling$.translate(ErrorHandling.scala:10)
	at sbt.WrapUsing.open(Using.scala:34)
	at sbt.Using.apply(Using.scala:23)
	at sbt.IO$$anonfun$gzipFileIn$1.apply(IO.scala:863)
	at sbt.IO$$anonfun$gzipFileIn$1.apply(IO.scala:862)
	at sbt.Using.apply(Using.scala:24)
	at sbt.IO$.gzipFileIn(IO.scala:862)
	at sbt.Sync$.readUncaught(Sync.scala:88)
	at sbt.Sync$.readInfo(Sync.scala:84)
	at sbt.Sync$$anonfun$apply$1.apply(Sync.scala:28)
	at sbt.Sync$$anonfun$apply$1.apply(Sync.scala:22)
	at sbt.Defaults$$anonfun$copyResourcesTask$1.apply(Defaults.scala:922)
	at sbt.Defaults$$anonfun$copyResourcesTask$1.apply(Defaults.scala:918)
	at scala.Function4$$anonfun$tupled$1.apply(Function4.scala:35)
	at scala.Function4$$anonfun$tupled$1.apply(Function4.scala:34)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:237)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
[error] (compile:copyResources) Error wrapping InputStream in GZIPInputStream: java.util.zip.ZipException: Not in GZIP format
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? 

Have I omitted anying?

Need I specify some settings?

davidbiancolin, 2018-03-21:
Now that you have the right version of SBT i'd try re-cloning this repository and starting from scratch.

clx7913, 2018-03-21:
OK.
Let me try.

clx7913, 2018-03-22:
I have already installed the sbt 0.13.12 by "apt-get install sbt=0.13.12",but when I tried "make project",It still said "Getting org.scala-sbt sbt 0.13.12 ...".

Is it normal?

clx7913, 2018-03-22:
I realized that one website "http://maven.oschina.net/" has closed down,so I couldn't access it.

Then I change it by "http://mirrors.ibiblio.org/maven2/".

Came the only error:

[error] (compile:copyResources) Error wrapping InputStream in GZIPInputStream: java.util.zip.ZipException: Not in GZIP format

clx7913, 2018-03-23:
Do you mean trying this command from the root of system（/）?

clx7913, 2018-03-23:
I encountered the same error again after I typed the command at /

root@vm:/home/chang/fpga-zynq/zedboard# cd /
root@vm:/# find . -name target -exec rm -r "{}" \;
find: `./lib/modules/4.4.0-31-generic/kernel/drivers/target': No such file or directory
find: `./path/to/install/riscv/toolchain/share/openocd/scripts/target': No such file or directory
find: `./usr/src/linux-headers-4.4.0-31-generic/include/config/ip6/nf/target': No such file or directory
find: `./usr/src/linux-headers-4.4.0-31-generic/include/config/target': No such file or directory
find: `./usr/src/linux-headers-4.4.0-31-generic/include/config/netfilter/xt/target': No such file or directory
find: `./usr/src/linux-headers-4.4.0-31-generic/include/config/ip/nf/target': No such file or directory
find: `./usr/src/linux-headers-4.4.0-31/include/target': No such file or directory
find: `./usr/src/linux-headers-4.4.0-31/drivers/target': No such file or directory
root@vm:/# cd /home/chang/fpga-zynq/zedboard#
bash: cd: /home/chang/fpga-zynq/zedboard#: No such file or directory
root@vm:/# cd /home/chang/fpga-zynq/zedboard
root@vm:/home/chang/fpga-zynq/zedboard# sbt sbt-version
[info] Set current project to zedboard (in build file:/home/chang/fpga-zynq/zedboard/)
[info] 0.13.12
root@vm:/home/chang/fpga-zynq/zedboard# make project
make -C /home/chang/fpga-zynq/rocket-chip/firrtl SBT="java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/chang/fpga-zynq/rocket-chip/sbt-launch.jar" root_dir=/home/chang/fpga-zynq/rocket-chip/firrtl build-scala
make[1]: Entering directory `/home/chang/fpga-zynq/rocket-chip/firrtl'
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/chang/fpga-zynq/rocket-chip/sbt-launch.jar "assembly"
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading project definition from /home/chang/fpga-zynq/rocket-chip/firrtl/project
Error wrapping InputStream in GZIPInputStream: java.util.zip.ZipException: Not in GZIP format
	at sbt.ErrorHandling$.translate(ErrorHandling.scala:10)
	at sbt.WrapUsing.open(Using.scala:34)
	at sbt.Using.apply(Using.scala:23)
	at sbt.IO$$anonfun$gzipFileIn$1.apply(IO.scala:863)
	at sbt.IO$$anonfun$gzipFileIn$1.apply(IO.scala:862)
	at sbt.Using.apply(Using.scala:24)
	at sbt.IO$.gzipFileIn(IO.scala:862)
	at sbt.Sync$.readUncaught(Sync.scala:88)
	at sbt.Sync$.readInfo(Sync.scala:84)
	at sbt.Sync$$anonfun$apply$1.apply(Sync.scala:28)
	at sbt.Sync$$anonfun$apply$1.apply(Sync.scala:22)
	at sbt.Defaults$$anonfun$copyResourcesTask$1.apply(Defaults.scala:922)
	at sbt.Defaults$$anonfun$copyResourcesTask$1.apply(Defaults.scala:918)
	at scala.Function4$$anonfun$tupled$1.apply(Function4.scala:35)
	at scala.Function4$$anonfun$tupled$1.apply(Function4.scala:34)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:237)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.util.zip.ZipException: Not in GZIP format
	at java.util.zip.GZIPInputStream.readHeader(GZIPInputStream.java:165)
	at java.util.zip.GZIPInputStream.<init>(GZIPInputStream.java:79)
	at sbt.Using$$anonfun$gzipInputStream$1.apply(Using.scala:84)
	at sbt.Using$$anonfun$gzipInputStream$1.apply(Using.scala:84)
	at sbt.Using$$anon$1.openImpl(Using.scala:51)
	at sbt.WrapUsing$$anonfun$open$2.apply(Using.scala:34)
	at sbt.ErrorHandling$.translate(ErrorHandling.scala:10)
	at sbt.WrapUsing.open(Using.scala:34)
	at sbt.Using.apply(Using.scala:23)
	at sbt.IO$$anonfun$gzipFileIn$1.apply(IO.scala:863)
	at sbt.IO$$anonfun$gzipFileIn$1.apply(IO.scala:862)
	at sbt.Using.apply(Using.scala:24)
[... 3 lines cut ...]
	at sbt.Sync$$anonfun$apply$1.apply(Sync.scala:28)
	at sbt.Sync$$anonfun$apply$1.apply(Sync.scala:22)
	at sbt.Defaults$$anonfun$copyResourcesTask$1.apply(Defaults.scala:922)
	at sbt.Defaults$$anonfun$copyResourcesTask$1.apply(Defaults.scala:918)
	at scala.Function4$$anonfun$tupled$1.apply(Function4.scala:35)
	at scala.Function4$$anonfun$tupled$1.apply(Function4.scala:34)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:237)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
[... cut ...]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
[error] (compile:copyResources) Error wrapping InputStream in GZIPInputStream: java.util.zip.ZipException: Not in GZIP format
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore?

davidbiancolin, 2018-03-23:
I meant from the root of the repository. Sorry :(

clx7913, 2018-03-23:
Never mind. I have snapshots. (#^.^#)

I just have a try at the root of the repo.Now the former error has disappeared but there is another error:

root@vm:/home/chang/fpga-zynq/zedboard# make project
cd /home/chang/fpga-zynq/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/chang/fpga-zynq/rocket-chip/sbt-launch.jar pack
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading project definition from /home/chang/fpga-zynq/rocket-chip/project
[info] Updating {file:/home/chang/fpga-zynq/rocket-chip/project/}rocket-chip-build...
[info] Resolving com.typesafe.sbt#sbt-ghpages;0.5.3 ...
[warn] problem while downloading module descriptor: https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.typesafe.sbt/sbt-ghpages/scala_2.10/sbt_0.13/0.5.3/ivys/ivy.xml: Connection refused (Connection refused) (21017ms)
[warn] problem while downloading module descriptor: http://mirrors.ibiblio.org/maven2/com/typesafe/sbt/sbt-ghpages_2.10_0.13/0.5.3/sbt-ghpages-0.5.3.pom: invalid sha1: expected=<html>
[warn] <head><title>index computed=b113767b47336dcc165c5dd2222b5df4cb86b7ce (7204ms)
[warn] 	module not found: com.typesafe.sbt#sbt-ghpages;0.5.3
[warn] ==== typesafe-ivy-releases: tried
[warn]   https://repo.typesafe.com/typesafe/ivy-releases/com.typesafe.sbt/sbt-ghpages/scala_2.10/sbt_0.13/0.5.3/ivys/ivy.xml
[warn] ==== sbt-plugin-releases: tried
[warn]   https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.typesafe.sbt/sbt-ghpages/scala_2.10/sbt_0.13/0.5.3/ivys/ivy.xml
[warn] ==== local: tried
[warn]   /root/.ivy2/local/com.typesafe.sbt/sbt-ghpages/scala_2.10/sbt_0.13/0.5.3/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/com/typesafe/sbt/sbt-ghpages_2.10_0.13/0.5.3/sbt-ghpages-0.5.3.pom
[warn] ==== ibiblio: tried
[warn]   http://mirrors.ibiblio.org/maven2/com/typesafe/sbt/sbt-ghpages_2.10_0.13/0.5.3/sbt-ghpages-0.5.3.pom
[warn] ==== jcenter: tried
[warn]   http://jcenter.bintray.com/com/typesafe/sbt/sbt-ghpages_2.10_0.13/0.5.3/sbt-ghpages-0.5.3.pom
[warn] ==== maven-central: tried
[warn]   http://repo1.maven.org/maven2/com/typesafe/sbt/sbt-ghpages_2.10_0.13/0.5.3/sbt-ghpages-0.5.3.pom
[warn] ==== jgit-repo: tried
[warn]   http://download.eclipse.org/jgit/maven/com/typesafe/sbt/sbt-ghpages_2.10_0.13/0.5.3/sbt-ghpages-0.5.3.pom
[info] Resolving com.typesafe.sbt#sbt-site;0.8.1 ...
[error] Server access Error: Connection refused (Connection refused) url=https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.typesafe.sbt/sbt-site/scala_2.10/sbt_0.13/0.8.1/ivys/ivy.xml
[warn] problem while downloading module descriptor: http://mirrors.ibiblio.org/maven2/com/typesafe/sbt/sbt-site_2.10_0.13/0.8.1/sbt-site-0.8.1.pom: invalid sha1: expected=<html>
[warn] <head><title>index computed=b113767b47336dcc165c5dd2222b5df4cb86b7ce (7003ms)
[warn] 	module not found: com.typesafe.sbt#sbt-site;0.8.1
[warn] ==== typesafe-ivy-releases: tried
[warn]   https://repo.typesafe.com/typesafe/ivy-releases/com.typesafe.sbt/sbt-site/scala_2.10/sbt_0.13/0.8.1/ivys/ivy.xml
[warn] ==== sbt-plugin-releases: tried
[warn]   https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.typesafe.sbt/sbt-site/scala_2.10/sbt_0.13/0.8.1/ivys/ivy.xml
[warn] ==== local: tried
[warn]   /root/.ivy2/local/com.typesafe.sbt/sbt-site/scala_2.10/sbt_0.13/0.8.1/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/com/typesafe/sbt/sbt-site_2.10_0.13/0.8.1/sbt-site-0.8.1.pom
[warn] ==== ibiblio: tried
[warn]   http://mirrors.ibiblio.org/maven2/com/typesafe/sbt/sbt-site_2.10_0.13/0.8.1/sbt-site-0.8.1.pom
[warn] ==== jcenter: tried
[warn]   http://jcenter.bintray.com/com/typesafe/sbt/sbt-site_2.10_0.13/0.8.1/sbt-site-0.8.1.pom
[warn] ==== maven-central: tried
[warn]   http://repo1.maven.org/maven2/com/typesafe/sbt/sbt-site_2.10_0.13/0.8.1/sbt-site-0.8.1.pom
[warn] ==== jgit-repo: tried
[warn]   http://download.eclipse.org/jgit/maven/com/typesafe/sbt/sbt-site_2.10_0.13/0.8.1/sbt-site-0.8.1.pom
[info] Resolving org.scala-sbt.ivy#ivy;2.3.0-sbt-2cc8d2761242b072cedb0a04cb39435[info] Resolving org.fusesource.jansi#jansi;1.4 ...
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: com.typesafe.sbt#sbt-ghpages;0.5.3: not found
[warn] 	:: com.typesafe.sbt#sbt-site;0.8.1: not found
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn] 	Note: Some unresolved dependencies have extra attributes.  Check that these dependencies exist with the requested attributes.
[warn] 		com.typesafe.sbt:sbt-ghpages:0.5.3 (scalaVersion=2.10, sbtVersion=0.13)
[warn] 		com.typesafe.sbt:sbt-site:0.8.1 (scalaVersion=2.10, sbtVersion=0.13)
[warn] 
[warn] 	Note: Unresolved dependencies path:
[warn] 		com.typesafe.sbt:sbt-ghpages:0.5.3 (scalaVersion=2.10, sbtVersion=0.13) (/home/chang/fpga-zynq/rocket-chip/project/plugins.sbt#L3-4)
[warn] 		  +- default:rocket-chip-build:0.1-SNAPSHOT (scalaVersion=2.10, sbtVersion=0.13)
[warn] 		com.typesafe.sbt:sbt-site:0.8.1 (scalaVersion=2.10, sbtVersion=0.13) (/home/chang/fpga-zynq/rocket-chip/project/plugins.sbt#L5-6)
[warn] 		  +- default:rocket-chip-build:0.1-SNAPSHOT (scalaVersion=2.10, sbtVersion=0.13)
sbt.ResolveException: unresolved dependency: com.typesafe.sbt#sbt-ghpages;0.5.3: not found
unresolved dependency: com.typesafe.sbt#sbt-site;0.8.1: not found
[... 31 lines cut ...]
[... cut ...]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
[error] (*:update) sbt.ResolveException: unresolved dependency: com.typesafe.sbt#sbt-ghpages;0.5.3: not found
[error] unresolved dependency: com.typesafe.sbt#sbt-site;0.8.1: not found
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? 

Is it the reason of this error that I don't possess valid web sources？
Or that's beacause the official websites  have stopped surpporting some old packages?

clx7913, 2018-03-23:
I add "resolvers += Resolver.url("bintray-sbt-plugins", url("https://dl.bintray.com/sbt/sbt-plugin-releases/"))(Resolver.ivyStylePatterns) " to /fpga-zynq/rocket-chip/project/plugins.sbt. 

The terminal has already ran for a long time,and I haven't encountered the former errors now!

It seems that it will take a I think it will succeed this time!

clx7913, 2018-03-23:
Wonderful! I've completed it without errors.

Thank for your patience and help! Good job,Toothless!

Hope this communication will help other people,too.

closed 2018-03-23 by clx7913