ucb-bar/fpga-zynq #72: [Branch new-devices] Make icenet public?

lucashmorais, 2018-03-23:
Hi there,

I would like to use the code from branch new-devices to program a ZC706 board, but as others have noticed, the icenet submodule is private. Would it be possible to make it public?

I just managed to build a SMP-enabled Linux image with the buildroot configuration found in the sifive/freedom-u-sdk repo and would like to generate a priv-1.10 compliant Rocket Chip bitstream with fpga-zynq.

Thanks

davidbiancolin, 2018-03-29:
See my comment at the end of https://github.com/ucb-bar/fpga-zynq/issues/65. This repo will be deprecated at the end of this summer, once we've opened up [FireSim](https://fires.im/).

As part of the deprecation i may contribute a shell for ZC706 and ZedBoard back to https://github.com/sifive/fpga-shells. 

In the meantime, if you want to use a non-master branch you're on your own.  I think you'll find that you should be able to remove IceNet without too much difficulty.

davidbiancolin, 2018-04-23:
We've bumped RC for a final time and merged in a stripped down version of the new devices branch. I hope that, while late, that'll be of some use.

closed 2018-04-23 by davidbiancolin