ucb-bar/fpga-zynq #74: make arm-linux error

clx7913, 2018-03-30:
Following the README.md,I encountered some errors when I input "make arm-linux".The log is below:

root@c:/home/chang/fpga-zynq/zedboard# make arm-linux
# compile board-compatible u-boot
git submodule update --init /home/chang/fpga-zynq/common/u-boot-xlnx
# copy relevant configuration files
if [ -a soft_config/boards.cfg ] ; \
	then \
	    cp soft_config/boards.cfg /home/chang/fpga-zynq/common/u-boot-xlnx/ ; \
	fi;
cp soft_config/zynq_zed.h /home/chang/fpga-zynq/common/u-boot-xlnx/include/configs/
# actually build
cd /home/chang/fpga-zynq/common/u-boot-xlnx && make CROSS_COMPILE=arm-xilinx-linux-gnueabi- zynq_zed_config
make[1]: Entering directory `/home/chang/fpga-zynq/common/u-boot-xlnx'
Configuring for zynq_zed board...
make[1]: Leaving directory `/home/chang/fpga-zynq/common/u-boot-xlnx'
cd /home/chang/fpga-zynq/common/u-boot-xlnx && make CROSS_COMPILE=arm-xilinx-linux-gnueabi- -j16
make[1]: Entering directory `/home/chang/fpga-zynq/common/u-boot-xlnx'
  GEN     include/autoconf.mk.dep
  GEN     include/autoconf.mk
make[1]: Leaving directory `/home/chang/fpga-zynq/common/u-boot-xlnx'
make[1]: Entering directory `/home/chang/fpga-zynq/common/u-boot-xlnx'
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  CHK     include/generated/version_autogenerated.h
  CC      lib/asm-offsets.s
  CC      arch/arm/lib/asm-offsets.s
  GEN     include/generated/asm-offsets.h
  GEN     include/generated/generic-asm-offsets.h
  HOSTCC  tools/common/env_embedded.o
  HOSTCC  tools/dumpimage.o
  HOSTCC  tools/envcrc.o
  HOSTCC  tools/image-host.o
  HOSTCC  tools/mkenvimage.o
  HOSTCC  tools/mkimage.o
  HOSTLD  tools/envcrc
  HOSTLD  tools/mkenvimage
  HOSTLD  tools/dumpimage
  HOSTLD  tools/mkimage
  CC      drivers/input/input.o
  CC      drivers/i2c/i2c_core.o
  CC      drivers/net/zynq_gem.o
  CC      drivers/pcmcia/tqm8xx_pcmcia.o
  CC      board/xilinx/zynq/board.o
  CC      disk/part.o
  CC      drivers/net/phy/phy.o
  CC      drivers/dfu/dfu.o
  CC      drivers/fpga/fpga.o
  CC      arch/arm/cpu/armv7/cache_v7.o
  CC      drivers/mmc/mmc.o
  LD      drivers/pcmcia/built-in.o
  CC      drivers/i2c/zynq_i2c.o
  CC      api/api.o
  CC      drivers/mtd/spi/sf.o
  CC      arch/arm/cpu/armv7/zynq/timer.o
  CC      drivers/dfu/dfu_mmc.o
  CC      board/xilinx/zynq/thor.o
  AS      arch/arm/lib/crt0.o
  LD      drivers/input/built-in.o
  CC      disk/part_dos.o
  CC      arch/arm/lib/bootm-fdt.o
  LD      drivers/i2c/built-in.o
  CC      drivers/mtd/spi/sf_params.o
  LD      board/xilinx/zynq/built-in.o
  CC      drivers/mmc/sdhci.o
  CC      drivers/dfu/dfu_ram.o
  CC      arch/arm/cpu/armv7/cpu.o
  CC      arch/arm/cpu/armv7/zynq/cpu.o
  CC      drivers/fpga/zynqpl.o
[... 268 lines cut ...]
  LD      spl/u-boot-spl
  OBJCOPY spl/u-boot-spl.bin
/home/chang/fpga-zynq/common/u-boot-xlnx/tools/zynq-boot-bin.py -o boot.bin -u spl/u-boot-spl.bin
  OBJCOPY u-boot.srec
  OBJCOPY u-boot.bin
  MKIMAGE u-boot.img
Input file is: spl/u-boot-spl.bin
Output file is: boot.bin
Using /home/chang/fpga-zynq/common/u-boot-xlnx/spl/u-boot-spl.bin to get image length - it is 43712 (0xaac0) bytes
After checksum waddr= 0x13  byte addr= 0x4c
Number of registers to initialize 0
Generating binary output  /home/chang/fpga-zynq/common/u-boot-xlnx/boot.bin
make[1]: Leaving directory `/home/chang/fpga-zynq/common/u-boot-xlnx'
mkdir -p soft_build
cp /home/chang/fpga-zynq/common/u-boot-xlnx/u-boot soft_build/u-boot.elf
# compile kernel
git submodule update --init /home/chang/fpga-zynq/common/linux-xlnx
# no make clean included here since one copy of linux should work on all boards
cd /home/chang/fpga-zynq/common/linux-xlnx && make ARCH=arm CROSS_COMPILE=arm-xilinx-linux-gnueabi- xilinx_zynq_defconfig 
make[1]: Entering directory `/home/chang/fpga-zynq/common/linux-xlnx'
drivers/Kconfig:31: can't open file "drivers/target/Kconfig"
make[2]: *** [xilinx_zynq_defconfig] Error 1
make[1]: *** [xilinx_zynq_defconfig] Error 2
make[1]: Leaving directory `/home/chang/fpga-zynq/common/linux-xlnx'
make: *** [arm-linux] Error 2

clx7913, 2018-03-30:
I know what's going on.
I used "find . -name target -exec rm -r "{}" \;" in the root of "fpga-zynq" repo,so it deleted some disrelated "target" directories,such as "common/linux-xlnx/drivers/target","common/linux-xlnx/include/target".

closed 2018-03-30 by clx7913

davidbiancolin, 2018-03-30:
Oh geez, good catch.