ucb-bar/fpga-zynq #78: Zedboard Issues

thedevonwhite, 2018-04-03:
We have been able to run ./fesvr-zynq pk hello using the included prebuilt image
We have successfully followed the entire tutorial to create our own image. We did not change any of the verilog yet. This would be as a test.
However, when running ./fesvr-zynq pk hello again on our image, the terminal freezes and is unresponsive to anything. We have repeated the steps to build the image and it yielded the same results.

clx7913, 2018-04-07:
I've solved it.

Take a look at the bottom of [https://github.com/ucb-bar/fpga-zynq/issues/77](url).

NovoDavid, 2018-04-17:
I'm having problems booting your prebuilt image on the Zedboard. I'm able to boot the Zedboard with its Out-of-Box Demo (https://reference.digilentinc.com/_media/zedboard/zedboard_oob_design.zip) so I believe that there is nothing wrong with my board configuration. While I am also able to boot your other prebuilt image on the Zybo board, I get a panic kernel when doing the same for the Zedboard:

> mmc0: no vqmmc regulator found
> mmc0: no vmmc regulator found
> mmc0: Invalid maximum block size, assuming 512 bytes
> mmc0: SDHCI controller on e0100000.sdhci [e0100000.sdhci] using ADMA
> usbcore: registered new interface driver usbhid
> usbhid: USB HID core driver
> TCP: cubic registered
> NET: Registered protocol family 17
> Registering SWP/SWPB emulation handler
> drivers/rtc/hctosys.c: unable to open rtc device (rtc0)
> ALSA device list:
>   No soundcards found.
> Waiting for root device /dev/mmcblk0p1...
> mmc0: new high speed SDHC card at address 59b4
> mmcblk0: mmc0:59b4 USD   14.9 GiB 
>  mmcblk0: p1 p2
> VFS: Mounted root (vfat filesystem) on device 179:1.
> devtmpfs: error mounting -2
> Freeing unused kernel memory: 196K (c0615000 - c0646000)
> Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.

NovoDavid, 2018-04-17:
I could solve the problem by changing the **sdboot** uboot env variable from:
>sdboot=echo Copying Linux from SD to RAM... && mmcinfo && fatload mmc 0 0x3000000 ${kernel_image} && fatload mmc 0 0x2A00000 ${devicetree_image} && bootm 0x3000000 - 0x2A00000

to:
>sdboot=echo Copying Linux from SD to RAM... && mmcinfo && fatload mmc 0 0x3000000 ${kernel_image} && fatload mmc 0 0x2A00000 ${devicetree_image} && fatload mmc 0 0x2000000 ${ramdisk_image} && bootm 0x3000000 0x2000000 0x2A00000

How is it possible that nobody else had a problem with that before? The images are +2 years old...

NovoDavid, 2018-04-18:
Yes, from the boot.bin file in the pre-images downloaded after running:
>$ make fetch-images