ucb-bar/fpga-zynq #79: make rocket CONFIG=DualCoreConfig error in zedboard directory

clx7913, 2018-04-08:
root@c:/home/chang/fpga-zynq/zedboard# make rocket CONFIG=DualCoreConfig
mkdir -p /home/chang/fpga-zynq/common/build
cd /home/chang/fpga-zynq/common && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/chang/fpga-zynq/rocket-chip/sbt-launch.jar "run /home/chang/fpga-zynq/common/build zynq Top zynq DualCoreConfig"
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading project definition from /home/chang/fpga-zynq/common/project
[info] Set current project to zynq (in build file:/home/chang/fpga-zynq/common/)
[info] Running zynq.Generator /home/chang/fpga-zynq/common/build zynq Top zynq DualCoreConfig
[error] (run-main-0) java.lang.Exception: Unable to find part "zynq.DualCoreConfig" from "DualCoreConfig", did you misspell it?
java.lang.Exception: Unable to find part "zynq.DualCoreConfig" from "DualCoreConfig", did you misspell it?
	at Chisel.package$throwException$.apply(compatibility.scala:208)
	at util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:32)
	at util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:27)
	at scala.collection.IndexedSeqOptimized$class.foldr(IndexedSeqOptimized.scala:62)
	at scala.collection.IndexedSeqOptimized$class.foldRight(IndexedSeqOptimized.scala:70)
	at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:48)
	at util.HasGeneratorUtilities$class.getConfig(GeneratorUtils.scala:27)
	at zynq.Generator$.getConfig(Generator.scala:3)
	at util.GeneratorApp$class.config(GeneratorUtils.scala:79)
	at zynq.Generator$.config$lzycompute(Generator.scala:3)
	at zynq.Generator$.config(Generator.scala:3)
	at util.GeneratorApp$class.world(GeneratorUtils.scala:80)
	at zynq.Generator$.world$lzycompute(Generator.scala:3)
	at zynq.Generator$.world(Generator.scala:3)
Caused by: java.lang.ClassNotFoundException: zynq.DualCoreConfig
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:264)
	at util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:29)
	at util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:27)
	at scala.collection.IndexedSeqOptimized$class.foldr(IndexedSeqOptimized.scala:62)
	at scala.collection.IndexedSeqOptimized$class.foldRight(IndexedSeqOptimized.scala:70)
	at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:48)
	at util.HasGeneratorUtilities$class.getConfig(GeneratorUtils.scala:27)
	at zynq.Generator$.getConfig(Generator.scala:3)
	at util.GeneratorApp$class.config(GeneratorUtils.scala:79)
	at zynq.Generator$.config$lzycompute(Generator.scala:3)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:2081)
	at sbt.Defaults$.toError(Defaults.scala:39)
	at sbt.Defaults$$anonfun$runTask$1$$anonfun$apply$38$$anonfun$apply$39.apply(Defaults.scala:750)
	at sbt.Defaults$$anonfun$runTask$1$$anonfun$apply$38$$anonfun$apply$39.apply(Defaults.scala:748)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
[error] (compile:run) Nonzero exit code: 1
[error] Total time: 0 s, completed Apr 8, 2018 8:40:46 PM
make: *** [/home/chang/fpga-zynq/common/build/Top.DualCoreConfig.fir] Error 1
root@c:/home/chang/fpga-zynq/zedboard#

clx7913, 2018-04-08:
make project & make vivado by default CONFIG work fine.The default CONFIG in common/Makefrag is `CONFIG ?= ZynqConfig`.

BenjaminLpn, 2018-04-08:
It's clear that there is no <code>zynq.DualCoreConfig</code>. The fpga-zynq repository does not include this configuration by default. You may add your own configuration in file: fpga-zynq/common/src/main/scala/Configs.scala.

clx7913, 2018-04-09:
I found the `DualCoreConfig` in the /fpga-zynq/rocket-chip/src/main/scala/rocketchip/configs.scala

The fpga-zynq/common/src/main/scala imports many classes:

`package zynq

import cde.{Parameters, Config, CDEMatchError}
import rocketchip._
import rocket._
import coreplex.BuildTiles
import uncore.devices.NTiles
import uncore.tilelink.TLId
import uncore.agents.NAcquireTransactors
import testchipip._
import chisel3._`

So I thought fpga-zynq already included the `DualCoreConfig`.Was I wrong?

clx7913, 2018-04-10:
I was wrong indeed.

Now I konw what to do,thank you~

closed 2018-04-10 by clx7913