ucb-bar/fpga-zynq #83: ../machine/mtrap.c:15:machine mode: unhandlable trap 3 @ 0x0000000080000270

clx7913, 2018-04-18:
I implemented dual-core rocket on zedboard and I encountered`../machine/mtrap.c:15:machine mode: unhandlable trap 3 @ 0x0000000080000270` when I ran`./fesvr-zynq pk hello`.

When I just implemented one-core rocket on zedboard,there wasn't this problem and I saw "Hello world!" successfully.

I used the same `hello` program in both cases.

clx7913, 2018-04-18:
The version of rocket-chip and riscv-tools are `https://github.com/freechipsproject/rocket-chip/tree/f3299ae91d3f01d0349eb4746886e303e8fb1b41`,`https://github.com/riscv/riscv-tools/tree/6136df9cba56afe12c2fa321ef497090917c15aa`,respectively.

davidbiancolin, 2018-04-23:
I'd suggest trying again with master.

clx7913, 2018-04-23:
I started updating the fpga-zynq repo including all submodules.

Thank you for you suggestion.

But I encountered one strange problem when I ran "./build.sh":
`root@c:/home/chang/fpga-zynq/rocket-chip/riscv-tools# ./build.shStarting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
application-specific initialization failed: Can't find a usable init.tcl in the following directories: 
    /tmp/pabuild/tcl8.5.14/lib/tcl8.5 /opt/Xilinx/SDK/2016.2/bin/unwrapped/lib/tcl8.5 /opt/Xilinx/SDK/2016.2/bin/lib/tcl8.5 /opt/Xilinx/SDK/2016.2/bin/unwrapped/library /opt/Xilinx/SDK/2016.2/bin/library /opt/Xilinx/SDK/2016.2/bin/tcl8.5.14/library /opt/Xilinx/SDK/2016.2/tcl8.5.14/library

This probably means that Tcl wasn't installed properly.

invalid command name "clock"
    while executing
"clock millis"
    invoked from within
"set autosetup(start) [clock millis]"
    (file "/home/chang/fpga-zynq/rocket-chip/riscv-tools/riscv-openocd/build/jimtcl/../../jimtcl/autosetup/autosetup" line 937)
configure: error: ../../jimtcl/configure.gnu failed for jimtcl
`
Have you ever seen this error?

clx7913, 2018-04-23:
Oh,I know the answer now.

According to my understanding,`source /opt/Xilinx/Vivado/2016.2/settings64.sh` will "occupy" some functions of tcl.When you run other programs which will use the same functions,the conflict occurs.

So I cancel the sourcing when I build the riscv-tools,and now "./build.sh" works!

closed 2018-04-23 by clx7913

clx7913, 2018-04-23:
The sbt version is 1.1.1,and verilator test has passed.