ucb-bar/fpga-zynq #85: latest fpga-zynq ./fesvr-zynq pk hello  ERROR: No cores found

clx7913, 2018-04-25:
Whether I use `boot.bin devicetree.dtb uImage uramdisk.image.gz` received by `make fetch-ramdisk` or use those files made by myself,`./fesvr-zynq pk hello` always failed.

 _____       _           _      _
| ___ \     | |         | |    (_)
| |_/ / ___ | |_   __ _ | |     _  _ __   _   _ __  __
|  __/ / _ \| __| / _` || |    | || '_ \ | | | |\ \/ /
| |   |  __/| |_ | (_| || |____| || | | || |_| | >  <
\_|    \___| \__| \__,_|\_____/|_||_| |_| \__,_|/_/\_\

PetaLinux v2013.10 (Yocto 1.4) zynq ttyPS0

zynq login: root 
Password: 
login[818]: root login  on `ttyPS0'

root@zynq:~# ./fesvr-zynq pk hello 
ERROR: No cores found

clx7913, 2018-04-25:
Under the last version of fpga-zynq,I `./fesvr-zynq pk hello` works fine using `class ZynqConfig extends Config(new WithZynqAdapter ++ new DefaultFPGAConfig)`.

So I doubt that this is because the rocket-chip source files have bugs.

clx7913, 2018-05-12:
I still could not solve this problem,and I just want to back to the previous version.

I don't know how to thoroughly back to the previous version.

clx7913, 2018-05-27:
I had a misunderstanding when I tried to change the version.

But fortunately now I've changed my version of repository into the previous one.

closed 2018-05-27 by clx7913

alexanderFell, 2018-06-12:
Hey, @clx7913 . Just a quick question: Which old version are you using? The latest one that works for me, is commit f03982e from last year.

JeremieMelo, 2018-10-17:
me too. I am using zybo-z7-20. After booting the linux from ramdisk, this "fesvr-zynq pk hello" will output no cores found. I don't know why.

josecm, 2019-01-27:
I am also getting this on a zybo board with the lastest commit

gongqingfeng, 2019-06-03:
> me too. I am using zybo-z7-20. After booting the linux from ramdisk, this "fesvr-zynq pk hello" will output no cores found. I don't know why.

I also meet the same problem: I use "make fetch-images" to get files, and run my zybo. However, "ERROR: No cores found" occurred when I run "./fesvr-zynq pk hello" or "./fesvr-zynq bbl".

josecm, 2019-06-03:
@gongqingfeng  try https://github.com/josecm/fpga-images-zybo

gongqingfeng, 2019-06-03:
> @gongqingfeng try https://github.com/josecm/fpga-images-zybo

Thank you! I have followed your advice, and I run "./fesvr-zynq pk hello" successfully. However, when I run "./fesvr-zynq pk hello" , I get some new errors like this:
"
terminate called after throwing an instance of 'std::runtime_error'
what(): could not open bb
l".
Have you ever met this problem. This is my issue: https://github.com/ucb-bar/fpga-zynq/issues/107