ucb-bar/fpga-zynq #87: I have encountered an error in prebuild image.

freefreer, 2018-06-06:
When I copy the prebuild file to the sd card, I can start linux on the zedboard, but when I run the hello example, I get the following error:

Root@zynq:~# ./fesvr-zynq pk hello
ERROR: No cores found

alexanderFell, 2018-06-12:
Hey,
I got the same issue. The prebuilt images are not working for me. Even if I build the image by myself for the Zedboard, I get the same error message. The last version, that works for me, was commit #f03982e

I am not sure, if someone figured it out, how to fix it, since this repository is deprecated now. Maybe it is also related to #77, #78, and #85

zhemao, 2018-07-09:
Sorry, I forgot to update all the binaries in the uramdisk.image.gz.

Could you please try the files in the master branch of fpga-images-zedboard? I've tested them on my board and they seem to work.

https://github.com/ucb-bar/fpga-images-zedboard

msyksphinz, 2018-07-10:
Hi,
After updating fpga-images-zedboard, I encountered boot problem on U-boot.
```
zynq-uboot> fatload mmc 0 0x3000000 uImage
zynq-uboot> fatload mmc 0 0x2A00000 devicetree.dtb
zynq-uboot> fatload mmc 0 0x2000000 uramdisk.image.gz
zynq-uboot> bootm 0x3000000 0x2000000 0x2A00000
```
and error messages are shown:
```
   Data Size:    11010692 Bytes = 10.5 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
```
I use commit:e85e4b0 for fpga-image-zedboard, and commit:f55bd8e for fpga-zynq, this pair is right?
My booting command is wrong?

Shuwen03, 2018-07-23:
Hi, 
I met the same problem. My board is ZC706 and the prebuilt image cannot work on the board. I got the same error:
Root@zynq:~# ./fesvr-zynq pk hello
ERROR: No cores found

zhemao, 2018-08-01:
@msyksphinz You shouldn't need to manually type uboot commands. The uImage should just boot automatically.

@Shuwen03 Are you using the commit of fpga-images-zc706 that is submoduled into fpga-zynq, or are you using the master branch of fpga-images-zc706? You should try out the latter and see if it works.

heaton16, 2018-10-02:
Hopefully this is the most recent thread; however, I am using a zc706 and get the ERROR: No cores found issue as well. Did we ever resolve this?

davidbiancolin, 2018-10-27:
Resolved by #98.

closed 2018-10-27 by davidbiancolin

tylerheaton16, 2018-11-02:
I know you have marked this as resolved; however, it looks like the zedboard file was only updated when everyone is commenting on the zc706 as well.

Goshik92, 2019-04-12:
I also get `ERROR: No cores found` on zc706. Have no idea what causes it.

heaton16, 2019-04-12:
Hey, I was also never able to get this work. It looks like it has something to do with maybe the device tree etc, but I am not sure. I would go ahead and take a look at the Sifive Freedom github page which has the same flow. It requires a zc707, however.

Goshik92, 2019-04-28:
bdd16b4 commit from https://github.com/riscv-boom/fpga-zynq worked for me. Though, it is boom core, not rocket. The author also issued a newer commit with label "verified" a few days ago. I did not test it, but maybe it also works.