ucb-bar/fpga-zynq #89: 'make project'  does not work successfully

girijabhagwat, 2018-07-06:
I am trying to implement rocket chip on zedboard. I have given rocket chip and testchipip's paths in ..common/Makefrag. Also, I have initialized the submodules. When I run the command 'make project' in my board's directory I am getting following errors. When I looked into files for errors I learned that there are some definitions missing or there are different definitions. I am not getting how to solve these issues. 

[info] Compiling 7 Scala sources to /home/hardwaresec/fpga-zynq/common/target/scala-2.11/classes...
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Configs.scala:20: value copy is not a member of Option[freechips.rocketchip.subsystem.MasterPortParams]
[error]   case ExtMem => up(ExtMem, site).copy(idBits = 6)
[error]                                   ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Configs.scala:20: not found: value idBits
[error]   case ExtMem => up(ExtMem, site).copy(idBits = 6)
[error]                                        ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Configs.scala:21: value copy is not a member of Option[freechips.rocketchip.subsystem.SlavePortParams]
[error]   case ExtIn => up(ExtIn, site).copy(beatBytes = 4, idBits = 12)
[error]                                 ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Configs.scala:21: not found: value beatBytes
[error]   case ExtIn => up(ExtIn, site).copy(beatBytes = 4, idBits = 12)
[error]                                      ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Configs.scala:21: not found: value idBits
[error]   case ExtIn => up(ExtIn, site).copy(beatBytes = 4, idBits = 12)
[error]                                                     ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/TestHarness.scala:38: value beatBytes is not a member of Option[freechips.rocketchip.subsystem.SlavePortParams]
[error]   val zynq = LazyModule(new ZynqAdapterCore(base, config.beatBytes))
[error]                                                          ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Top.scala:38: not found: type HasMasterAXI4MemPort
[error]     with HasMasterAXI4MemPort
[error]          ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Top.scala:50: not found: type HasMasterAXI4MemPortModuleImp
[error]     with HasMasterAXI4MemPortModuleImp
[error]          ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/TestHarness.scala:26: value connectSimAXIMem is not a member of zynq.FPGAZynqTopModule
[error]   dut.connectSimAXIMem()
[error]       ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Top.scala:17: type mismatch;
[error]  found   : Option[freechips.rocketchip.subsystem.SlavePortParams]
[error]  required: freechips.rocketchip.subsystem.SlavePortParams
[error]   val adapter = Module(LazyModule(new ZynqAdapter(address, config)).module)
[error]                                                            ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Top.scala:19: value mem_axi4 is not a member of zynq.FPGAZynqTopModule
[error]   require(target.mem_axi4.size == 1)
[error]                  ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Top.scala:23: value mem_axi4 is not a member of zynq.FPGAZynqTopModule
[error]     val mem_axi = target.mem_axi4.head.cloneType
[error]                          ^
[error] 12 errors found
[error] (compile:compileIncremental) Compilation failed
[error] Total time: 10 s, completed Jul 6, 2018 3:25:26 PM
../common/Makefrag:91: recipe for target '/home/hardwaresec/fpga-zynq/common/build/Top.ZynqFPGAConfig.fir' failed
make: *** [/home/hardwaresec/fpga-zynq/common/build/Top.ZynqFPGAConfig.fir] Error 1

Is repository not updated or I am following some wrong instructions?
Thanks.

zhemao, 2018-07-09:
Which branch of rocket-chip are you using? You seem to be on a more recent branch than the one submoduled by fpga-zynq. There were backwards-breaking changes introduced in the latest rocket-chip master, so that's why you're seeing these issues.

girijabhagwat, 2018-07-09:
Thanks. Currently, I have installed rocket chip master branch. As per mentioned on this page, I can understand this repository does not support rocket chip master. So, can you please suggest which branch should I install?

zhemao, 2018-07-09:
Just use the commit pointed to by the submodule. You can create a new branch from it and then add your changes.
    
    cd fpga-zynq
    git submodule update rocket-chip
    git checkout -b hardwaresec

girijabhagwat, 2018-07-09:
I am still getting same error. I Followed the suggested steps and created the new branch. But I am not sure which changes to add.

hoangt, 2018-07-24:
I made some changes
```
In /common/src/main/scala/Configs.scala
case ExtMem => MasterPortParams(base=0x80000000L, size=0x10000000L, beatBytes=8, idBits=6)
case ExtIn => SlavePortParams(beatBytes=8, idBits=8, sourceBits=2)

In /common/src/main/scala/Top.scala
    //with HasMasterAXI4MemPort
   with CanHaveMasterAXI4MemPort
  ...
    with HasRTCModuleImp
    //with HasMasterAXI4MemPortModuleImp
```  
Still facing these errors which are very odd
```
[error] val zynq = LazyModule(new ZynqAdapterCore(base, config.beatBytes))
....
[error] val adapter = Module(LazyModule(new ZynqAdapter(address, config)).module)

```

hoangt, 2018-07-24:
Fixed errors remained
In /common/src/main/scala/TestHarness.scala, add follow line (after  `val config = p(ExtIn) `)
`  val param   = config.get`

girijabhagwat, 2018-07-26:
Thank you so much. I followed the instructions and it worked for that part. But, I am getting another set of errors in further execution.

[info] Compiling 7 Scala sources to /home/hardwaresec/fpga-zynq/common/target/scala-2.11/classes...
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/TestHarness.scala:26: value connectSimAXIMem is not a member of zynq.FPGAZynqTopModule
[error]   dut.connectSimAXIMem()
[error]       ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Top.scala:17: type mismatch;
[error]  found   : Option[freechips.rocketchip.subsystem.SlavePortParams]
[error]  required: freechips.rocketchip.subsystem.SlavePortParams
[error]   val adapter = Module(LazyModule(new ZynqAdapter(address, config)).module)
[error]                                                            ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Top.scala:19: value mem_axi4 is not a member of zynq.FPGAZynqTopModule
[error]   require(target.mem_axi4.size == 1)
[error]                  ^
[error] /home/hardwaresec/fpga-zynq/common/src/main/scala/Top.scala:23: value mem_axi4 is not a member of zynq.FPGAZynqTopModule
[error]     val mem_axi = target.mem_axi4.head.cloneType
[error]                          ^
[error] four errors found
[error] (compile:compileIncremental) Compilation failed
[error] Total time: 9 s, completed Jul 26, 2018 1:15:17 PM
../common/Makefrag:91: recipe for target '/home/hardwaresec/fpga-zynq/common/build/Top.ZynqFPGAConfig.fir' failed
make: *** [/home/hardwaresec/fpga-zynq/common/build/Top.ZynqFPGAConfig.fir] Error 1

Can you please help?
Thanks.

hmxf, 2019-06-12:
Maybe we just troubled by the same problem? I used default configurations except rocket-chip, but the rocket-chip I hold is successfully built and simulated, so I don't think it's a problem from rocket-chip.
Here's what I met. 

>risc-v@Zaphkiel:\~/RISC-V/fpga-zynq/zybo$ make rocket
>rm -f /home/risc-v/RISC-V/fpga-zynq/common/../testchipip/lib
>ln -s /home/risc-v/RISC-V/fpga-zynq/common/lib /home/risc-v/RISC-V/fpga-zynq/common/../testchipip/lib
>cd /home/risc-v/RISC-V/fpga-zynq/common/../testchipip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/risc-v/RISC-V/rocket-chip/sbt-launch.jar ++2.11.12 package
>OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
>[info] Loading project definition from /home/risc-v/RISC-V/fpga-zynq/testchipip/project
>[info] Loading settings from build.sbt ...
>[info] Set current project to testchipip (in build file:/home/risc-v/RISC-V/fpga-zynq/testchipip/)
>[info] Setting Scala version to 2.11.12 on 1 projects.
>[info] Reapplying settings...
>[info] Set current project to testchipip (in build file:/home/risc-v/RISC-V/fpga-zynq/testchipip/)
>[info] Compiling 11 Scala sources to /home/risc-v/RISC-V/fpga-zynq/testchipip/target/scala-2.11/classes ...
>[error] /home/risc-v/RISC-V/fpga-zynq/testchipip/src/main/scala/NoDebug.scala:19:11: value dmi is not a member of Option[freechips.rocketchip.devices.debug.ClockedDMIIO]
>[error]   debugIO.dmi.req.valid := false.B
>[error]           ^
>[error] /home/risc-v/RISC-V/fpga-zynq/testchipip/src/main/scala/NoDebug.scala:20:11: value dmi is not a member of Option[freechips.rocketchip.devices.debug.ClockedDMIIO]
>[error]   debugIO.dmi.resp.ready := false.B
>[error]           ^
>[error] /home/risc-v/RISC-V/fpga-zynq/testchipip/src/main/scala/NoDebug.scala:21:11: value dmiClock is not a member of Option[freechips.rocketchip.devices.debug.ClockedDMIIO]
>[error]   debugIO.dmiClock := clock
>[error]           ^
>[error] /home/risc-v/RISC-V/fpga-zynq/testchipip/src/main/scala/NoDebug.scala:22:11: value dmiReset is not a member of Option[freechips.rocketchip.devices.debug.ClockedDMIIO]
>[error]   debugIO.dmiReset := reset.toBool
>[error]           ^
>[error] /home/risc-v/RISC-V/fpga-zynq/testchipip/src/main/scala/Serdes.scala:293:25: value error is not a member of freechips.rocketchip.tilelink.TLBundleC
>[error]     merged.union   := c.error
>[error]                         ^
>[error] /home/risc-v/RISC-V/fpga-zynq/testchipip/src/main/scala/Serdes.scala:307:37: value error is not a member of freechips.rocketchip.tilelink.TLBundleD
>[error]     merged.union   := Cat(d.sink, d.error)
>[error]                                     ^
>[error] /home/risc-v/RISC-V/fpga-zynq/testchipip/src/main/scala/Serdes.scala:389:7: value error is not a member of freechips.rocketchip.tilelink.TLBundleC
>[error]     c.error   := chan.union(0)
>[error]       ^
>[error] /home/risc-v/RISC-V/fpga-zynq/testchipip/src/main/scala/Serdes.scala:409:7: value error is not a member of freechips.rocketchip.tilelink.TLBundleD
>[error]     d.error   := chan.union(0)
>[error]       ^
>[error] /home/risc-v/RISC-V/fpga-zynq/testchipip/src/main/scala/Switcher.scala:112:38: value UNCACHEABLE is not a member of object freechips.rocketchip.diplomacy.RegionType
>[error]                      else RegionType.UNCACHEABLE,
>[error]                                      ^
>[error] /home/risc-v/RISC-V/fpga-zynq/testchipip/src/main/scala/Unittests.scala:7:8: object ErrorParams is not a member of package freechips.rocketchip.devices.tilelink
>[error] import freechips.rocketchip.devices.tilelink.{TLTestRAM, TLROM, TLError, ErrorParams}
>[error]        ^
>[error] /home/risc-v/RISC-V/fpga-zynq/testchipip/src/main/scala/Unittests.scala:320:38: not found: value ErrorParams
>[error]   val error = LazyModule(new TLError(ErrorParams(
>[error]                                      ^
>[error] 11 errors found
>[error] (Compile / compileIncremental) Compilation failed
>[error] Total time: 5 s, completed Jun 12, 2019, 5:14:08 PM
>/home/risc-v/RISC-V/fpga-zynq/common/Makefrag.pkgs:2: recipe for target '/home/risc-v/RISC-V/fpga-zynq/common/lib/testchipip.stamp' failed
>make: *** [/home/risc-v/RISC-V/fpga-zynq/common/lib/testchipip.stamp] Error 1
>risc-v@Zaphkiel:\~/RISC-V/fpga-zynq/zybo$