ucb-bar/fpga-zynq #92: Zybo z7

mehr74, 2018-07-23:
I want to run RISC-V rocket-chip on Zynq FPGA Board (Zybo Z7), what are the constraints I have to add the project?

yuliu2015, 2018-08-31:
I'm also trying to run the rocket core on Zybo Z7 (part no: XC7Z010-1CLG400C) without success using the pre-built. Seems we need to re-build from scratch with the new Zybo Z7 board constrain file (provided by Digilent) in vivado.

mehr74, 2018-08-31:
I tried to build from scratch as well but  I could not be able to run any program on the FPGA. Please let me know if you did that.

JeremieMelo, 2018-10-17:
> I'm also trying to run the rocket core on Zybo Z7 (part no: XC7Z010-1CLG400C) without success using the pre-built. Seems we need to re-build from scratch with the new Zybo Z7 board constrain file (provided by Digilent) in vivado.

I am using zybo-z7-20, I can boot linux from ramdisk, but cannot use fesvr-zynq to communicate with rocket core. How about your project?

gongqingfeng, 2019-06-03:
I run fetch-images and get somefiles. I run them on my zybo succeed.

gongqingfeng, 2019-06-03:
> I am using zybo-z7-20, I can boot linux from ramdisk, but cannot use fesvr-zynq to communicate with rocket core. How about your project?

I meet the same problem with you.  I am using zybo, I can boot linux from Ramdisk, however, I can't run "./fesvr-zynq pk hello" successfully: I don't get any response from the command line, that is, arm cores can't communicate with rocket core.
Do you solve this problem? Thank you very much!