ucb-bar/fpga-zynq #94: VM Fault when SFENCE.VMA is before satp write

natestuder, 2018-09-13:
We've been using the zedboard Rocket Chip for seL4 RISC-V development and noticed that we would get random vm faults when running on this platform.  

The 1.10 RISC-V Privileged Specification states:

> If the new address space’s page tables have been modified, it maybe necessary to execute an SFENCE.VMA instruction *prior* to writing satp.

Which is how the seL4 context switch was implemented, but we discovered that using the opposite order:  SFENCE.VMA after the satp write resolved the random vm faults.

Since we don't see similar faults on the HiFive Unleashed board from SiFive regardless of the order of these two instructions, it may just be a quirk of the Rocket-Chip or the zedboard implementation.

aswaterman, 2018-09-13:
As I mentioned inhttps://github.com/riscv/riscv-isa-manual/issues/226, you need to do an SFENCE after reusing an ASID.  Since rocket-chip doesn't provide ASIDs at this time, every SATP change is an ASID reuse, so you need an SFENCE after the SATP change.

Naturally, I have no idea what's actually going on here, and so there's little chance my suggestion will fix the problem.  Maybe try replicating it in RTL simulation?  Please report back if you figure it out.

natestuder, 2018-09-13:
What you've described makes sense with the behavior we were seeing and putting the SFENCE after the SATP change does fix it in our case, so we're planning on upstreaming that change to seL4.

closed 2018-09-13 by natestuder

aswaterman, 2018-09-13:
Cool!