ucb-bar/fpga-zynq #100: Module ‘system’ not found

yonghaoXuGit, 2018-11-10:
When I use vivado to synthesis the generated verilog，the command report an error:"module 'system' not found",the error is located at line 139 in rocketchip_wrapper.v.Is there any wrong?Or it lacks some files?

dengwxn, 2019-08-05:
Hello, did you solve it? I'm having the same problem. @yonghaoXuGit

zslwyuan, 2021-05-05:
@dengwxn @yonghaoXuGit I suspect that your system is not built successfully. There might be a Tcl error complaining about your version of Vivado like this one:

ERROR: [BD_TCL-109] This script was generated using Vivado <2016.2> and is being run in <2020.1> of Vivado. Please run the script in Vivado <2016.2> then open the design in Vivado <2020.1>. Upgrade the design by running "Tools => Report => Report IP Status...", then run write_bd_tcl to create an updated script.

change line 23 in "fpga-zynq/zedboard/src/tcl/zedboard_bd.tcl" into "set scripts_vivado_version 20xx.x" according to your version of Vivado.