ucb-bar/fpga-zynq #101: Problems with fesvr-zynq on zc702

ipgul, 2018-11-16:
Greetings,

I've recently cloned the fpga-zynq repository and followed section 3 for "building everything from scratch" and I've reached the point where linux is booting on the ARM core. However when I run ./fesvr-zynq pk hello I have no response.

`root@zynq:~# ./fesvr-zynq pk hello`

**I read all the issues related to this particular problem** and similar ones but none of the presented solutions worked for me. I didn't know whether I should respond to one of the other issues or open a new one.

At first I tried with a modified ZynqSmallConfig (with the ZynqAdapter) where I inserted a new memory mapped module. Then I tried again just with the simple small config and the result was the same.

Also, I tried using the files in the ramdisk that I got from fetch-ramdisk and then as another attempt I've rebuilt the pk (both for 64-bit and 32-bit), recompiled hello (also 64-bit and 32-bit), tried using different combinations of compilers for the FSBL (the standard arm-none and with arm-xilinx - the latter gave me errors with VFP so in the end it did not work) and for fesvr-zynq (the standard arm-xilinx-linux and with the gnueabihf one), among other attempts (namely those suggested in other issues), nothing worked.

I tried building the riscv tools with the --isa=rv64ima option, same result.

I can run spike pk hello with no problem.

I did not forget to copy the libfesvr.so to the proper location after rebuilding it.

I'm using Vivado/SDK 2016.2 as suggested.

Since I'm using a different board than those supported by the repository (zc702), I've modified a couple of files (namely the constraints).

The note below in the readme leads me to believe I should remove the "with HasNoDebug" option. Is that correct? If so, what do I put in this line in fpga-zynq/common/src/main/scala/Top.scala: target.debug := DontCare ?

> "Note: If you are seeking to test your own modifications to rocket chip (RC) using this repository, it must be derived from RC commit fb476d1 or later, as this project expects both a debug module to be present, and links directly against RC to build a top-level project directly in chisel. Otherwise, you should use an older version of this repository bf6d00c2 or earlier, which has support for HTIF-based rocket chip instances."

At the moment I don't have a JTAG cable for debugging.

Any ideas on what can be wrong?
Thanks in advance,

zxhero, 2019-01-07:
Hi, how about your project? 

When I run `root@zynq:~# ./fesvr-zynq pk hello`, I got a message `error: no core found`.

I tried both pre-build boot.bin and building from scratch, but I got same error message.

I can run `spike pk hello` with no problem.

I'm using zc706 board.

ipgul, 2019-01-07:
Hi @zxhero,

In one of my attempts to run the code I got to that error message as well.

In the meantime I tried to mess with the fesvr-zynq code and the registers created in zynq_driver.cc for a more direct attempt at trying to figure out what's wrong. I realized that any changes I wrote to them were not persistent. Meaning I wrote to them with no error, but reading again would prove the value had not changed.

I also found out it gets stuck on a loop in the poll function in zynq_driver.cc and that's why I get no response.

I asked someone with a zedboard to try the code and it worked with an old branch of the zedboard image (e85e4b06de).

So far I think the problem comes from fesvr, since I can run everything with spike and since that branch works with zedboard. I don't think it's an issue of the programmable logic but I'm not 100% sure. But I haven't been able to solve the problem and continue the project.

zxhero, 2019-01-12:
Hi @ipgul ,

I also think it is a problem of fesvr-zynq. 

In my case, it may because mismatch between how fesvr-zynq and debug module internal to rocket-chip communicate (some errors in poll function). 

When I tried to use newest fesvr-zynq and an old branch of the fpga-zynq (f03982e), I can get message `hello!`. But I can not boot riscv-linux.

Then when I tried to use fesvr-zynq compiled from the fpga-zynq (f03982e), eveything worked fine for me.

mars20, 2019-02-03:
Hi @zxhero,

I ran into similar problems as you, when I was trying to set up rocket core on ZC706 FPGA. I was using the pre-build fpga-image files. 

`root@zynq:~# ./fesvr-zynq pk hello`
`error: no core found`

I followed the instructions to make the fesvr-zynq and updated the fesvr-zynq, libfesvr.so and pk. 

That fixed the error! 
Hope it helps you

ipgul, 2019-02-27:
Hi all,

@zxhero, thanks for pointing out [f03982e](https://github.com/ucb-bar/fpga-zynq/commit/f03982e6a155d333b3b3708185898745df5d4004).

The fpga-zynq [f03982e](https://github.com/ucb-bar/fpga-zynq/commit/f03982e6a155d333b3b3708185898745df5d4004) branch worked for me. I made rocket with the riscv-tools version tracked by that branch and used a newer ramdisk image of the ZC706 with a fesvr-zynq from [f03982e](https://github.com/ucb-bar/fpga-zynq/commit/f03982e6a155d333b3b3708185898745df5d4004). The ramdisk from the ZC706 image repository version at the fpga-zynq [f03982e](https://github.com/ucb-bar/fpga-zynq/commit/f03982e6a155d333b3b3708185898745df5d4004) commit was not working (the ARM core did not boot), that is why I used a newer one.

I got the "hello!" message and was able to book riscv-linux. Then I used the example from the project-template repository (the PWM MMIO peripheral) and was able to add it to the ZynqSmallConfig design.

For anyone using the ZC702, I only had to modify 3 files. The ZC706 Makefile, the zc706_bd.tcl and the base.xdc constraints file.

In the Makefile I chaged to this:

```
BOARD = zc702
UBOOT_CONFIG = zc70x
BOARD_MODEL = xilinx.com:zc702:part0:1.0
PART = xc7z020clg484-1
CONFIG = ZynqFPGAConfig
```

In the contraints file I used:

```
set_property PACKAGE_PIN D18 [get_ports SYSCLK_P]
set_property` IOSTANDARD LVDS_25 [get_ports SYSCLK_P]

set_property PACKAGE_PIN C19 [get_ports SYSCLK_N]
set_property IOSTANDARD LVDS_25 [get_ports SYSCLK_N]

create_clock -add -name SYSCLK_P -period 5.00 -waveform {0 2.5} [get_ports SYSCLK_P]
```

And in the tcl I mostly changed the board and device type. I compared the rest with another design I had for the ZC702 and realized the values did not change much, so I left the rest untouched.

Hope this helps!

closed 2019-02-27 by ipgul